iiab/iiab #1532: RPi3 visiting http://box/kolibri causes browser to infinitely loop, preventing login

holta, 2019-02-27:
On one particular RPi 3 (running IIAB 7.0/master on Raspbian Lite) it loops constantly when I try to visit http://192.168.0.131/kolibri &mdash; causing the browser to get stuck in an infinite loop, redirecting a couple times per second, back & forth between these 2 pages:

- http://192.168.0.131/kolibri/user/#/signin
- http://192.168.0.131/kolibri/user/

Rebooting does _not_ change the situation FYI.  Interestingly however, this problem #2 does not occur when you visit Kolibri with the WAN/OpenVPN address http://10.8.0.50/kolibri !  Making me wonder if this could possibly be related to Apache issues or possibly networking issues like #1527 ?

Tangentially related to: #1531

holta, 2019-02-27:
CLARIFS:
- running ./iiab-network did not fix this problem
- this RPi 3 was installed using an Ethernet cable (not WiFi)
- iiab-hotspot-on & iiab-hotspot-off were never run, as the hotspot is already on as it should be

holta, 2019-02-27:
This infinite loop behavior has now appeared *one* another machine &mdash; an Ubuntu 18.04 NUC being the one (and only one) of several NUC PC's &mdash; that is rapidly redirecting / oscillating between these two URL's:
- http://192.168.0.121/kolibri/user/#/signin
- http://192.168.0.121/kolibri/user/

Just like the afflicted RPi 3, the infinite loop does not happen when browsers visit it from the WAN side using http://10.8.0.34/kolibri

fyi _all_ these machines (not just the 2 afflicted) are running the latest IIAB 7.0/master.  As the 2 machines are very different in HW & OS e& networking topology, I do not have a clue what might be causing it at this time :/

tim-moody, 2019-02-27:
is captive portal on?

holta, 2019-02-27:
> is captive portal on?

All 5 machines are BIG-sized so Captive Portal is set to True/True &mdash; including the afflicted 2 machines (RPi & NUC) &mdash; and the 3 afflictecd machines (also RPi & NUC).

Further Oddities:
- http://box/kolibri has a *different* login screen when the infinite redirects happen (showing BOTH username & password textfields).  Whereas all other machines show JUST THE USERNAME textfield on that initial screen.
- infinite redirects don't happen when using alternative URL http://box:8009/kolibri
- infinite redirects don't happen on a mobile interface, even when the mobile browser requests the "Desktop Site"

holta, 2019-02-27:
@arky if you recognize any of these baffling patterns above (#1533) plz LMK!

holta, 2019-03-06:
Kolibri 0.12.0 appears to have fixed (or papered over?!) the situation on both machines.

(Let's keep this tkt open for another month or so, and close it in April 2019 if the problem does not recur...)

floydianslips, 2019-03-19:
Running into this issue on daily build of Debian 10 running on VM from March 18, 2019 as well.

holta, 2019-03-19:
Holt: Rebooting does not help.  Upgrading from 0.11.1 to 0.12.x does (temporarily) solve the problem...

Tim: I believe it is the conf file not captive portal

The browser should catch infinite redirection

Holt: For some reason browser doesn't put an end to this particular one

It seems to be cycling fast, back+forth betw 2 diff pages

Tim: Site does a redirect and conf sends back

Holt: Very weird how http://box/kolibri works initially, then suddenly/permanently starts failing at a later date.

holta, 2019-03-19:
Firefox is a workaround when Chrome/Chromium definitely fail according to @floydianslips 

"Also works on duck duck go browser on mobile"

holta, 2019-03-19:
CLARIF: this syndrome definitely occurs with both Kolibri 0.11.1 and 0.12.1

floydianslips, 2019-03-19:
This seems to be effecting Chrome. I have tested other browsers including Safari and Edge without the error occuring.

holta, 2019-03-19:
Chrome browser on other machines sometimes works, while Chrome browser continues to fail with http://box/kolibri on other machines.

We have seen Chrome fail on Ubuntu 18.04 and Windows both.

This might possibly be an issue of Kolibri storing incorrect cookies from 1969??  But clearing cookies completely does NOT solve the problem, so that theory is highly uncertain.

holta, 2019-03-19:
FYI disabling all Chrome extensions & deleting all cookies does not solve the problem.

And yet a Chrome incognito window does work around the problem.

Extremely weird.

floydianslips, 2019-03-19:
A complete history wipe (for me it was 24 hours) on Chrome seems to have solved the issue. There were 2 machines running both with a loop at the login screen, after clearing history both are functioning normally.

holta, 2019-03-19:
> A complete history wipe (for me it was 24 hours) on Chrome seems to have solved the issue. There were 2 machines running both with a loop at the login screen, after clearing history both are functioning normally.

@benjaoming

This strongly suggests a flaw in Kolibri itself, with regards to how it stores cookies or prior state, rather than a flaw with the Chrome browser.

Is there someone that can help us understand how we're supposed to diagnose this, with what log files or otherwise?

holta, 2019-03-19:
_RECAP from @floydianslips who correctly summarizes the problem as it appears anyway:_

> It seemed like Kolibri was trying to log in with crap credentials and then redirecting back to the same page [continuously, in an infinite loop, both yesterday and earlier today]

holta, 2019-03-21:
Many thanks to @jredrejo from Learning Equality, who recommends these diagnostic approaches so we're ready the next time:

> ...tried several [things], but haven't found a way to reproduce it nor can think of how it could happen.
To diagnose kolibri there are several tools available:
> - Logs: https://kolibri.readthedocs.io/en/latest/manage.html#locate-kolibri-log-files in your case server.log and kolibri.log would be the ones that might have more information.
> - Profile and benchmark commands: https://kolibri.readthedocs.io/en/latest/manage.html?highlight=profile#test-kolibri-server-performance
>
> But in your case, a dump of the requests producing the loop from  network tab from Chrome (or FIrefox) dev tools would be the most useful part, as it seems the problem is not in the backend of the application but in the frontend:

[image]

> Having the server logs plus this dump would be really useful to reproduce this problem or detect what's happening.

KW4NP, 2019-04-29:
I'm having this issue as well, but it is only affecting Firefox, not Chromium or Safari.  Using the specific port allows login but there is still a different issue that could be related (#1626 ).

KW4NP, 2019-04-29:
Here are all of the diagnostic files.  debug.log was empty.
[Archive 19-04-29 17-19-06.har.zip](https://github.com/iiab/iiab/files/3128271/Archive.19-04-29.17-19-06.har.zip)

[kolibri.log](https://github.com/iiab/iiab/files/3128259/kolibri.log)
[server.log](https://github.com/iiab/iiab/files/3128260/server.log)

KW4NP, 2019-04-29:
@jredrejo, here's what the apache access log looks like.
`192.168.5.37 - - [29/Apr/2019:15:53:01 +0000] "GET /kolibri/user/ HTTP/1.1" 200 14501 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.
0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/api/auth/session/current/?active=false HTTP/1.1" 403 534 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X1
1; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.400.woff HTTP/1.1" 200 198240 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.700.woff HTTP/1.1" 200 163489 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:02 +0000] "GET /kolibri/user/ HTTP/1.1" 200 14501 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.
0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/api/auth/session/current/?active=false HTTP/1.1" 403 534 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X1
1; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.700.woff HTTP/1.1" 200 259057 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.400.woff HTTP/1.1" 200 118600 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:03 +0000] "GET /kolibri/user/ HTTP/1.1" 200 14501 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.
0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:04 +0000] "GET /kolibri/api/auth/session/current/?active=false HTTP/1.1" 403 534 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X1
1; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:04 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.700.woff HTTP/1.1" 200 205481 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:04 +0000] "GET /kolibri/static/assets/fonts/noto-full.NotoSans.400.woff HTTP/1.1" 200 166384 "http://192.168.5.24/kolibri/static/asse
ts/fonts/noto-full.en.modern.css?v=0.12.2" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:04 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
192.168.5.37 - - [29/Apr/2019:15:53:04 +0000] "GET /kolibri/redirectuser/ HTTP/1.1" 302 296 "http://192.168.5.24/kolibri/user/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; 
rv:65.0) Gecko/20100101 Firefox/65.0"
`

holta, 2019-04-29:
Many thanks @KW4NP

Others indicate this MIGHT NOT just be an issue on Firefox?  But rather something triggered by Kolibri's own cookies or similar?

In any case, the bug appears to go away when Firefox is restarted after wiping its cache and cookies (etc) COMPLETELY clean!

KW4NP, 2019-04-29:
I believe I tried a full clean of cache and cookies earlier in conjunction with a reboot of the pi but it did not seem to help.  Can't try again because of the workaround for #1626.  Either way all other browsers seemed to be fine.

holta, 2019-04-29:
> I believe I tried a full clean of cache and cookies earlier in conjunction with a reboot of the pi but it did not seem to help.

Not quite enough:

What I meant to say earlier: deleting Chrome's browser history was a (temporary) workaround for @floydianslips

Yes this is confounding.

This bug definitely affects Firefox and Chrome both.

benjaoming, 2019-04-29:
@KW4NP thanks for the .har

KW4NP, 2019-04-30:
@benjaoming, here is the new .har file.  I found that I had selected the JS tab last time, so we only got the .js files.  This should be everything.  Thanks
[Archive 19-04-30 06-56-52.har.zip](https://github.com/iiab/iiab/files/3130063/Archive.19-04-30.06-56-52.har.zip)

benjaoming, 2019-04-30:
Thanks. It seems that it's a redirect caused in the session keep alive request that immediately returns 403 after the login.

The issue doesn't seem to be that the request is made to different hosts, so I would really suspect any captive portals and such to be playing part.

Ping @rtibbles - maybe you could look at the traceback from @KW4NP and guess a reason?

Request:

```
Host: 192.168.5.24
User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0
Accept: application/json, application/json;q=0.8, text/plain;q=0.5, */*;q=0.2
Accept-Language: en-US,fr-FR;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Referer: http://192.168.5.24/kolibri/user/
Content-Type: application/json
X-CSRFToken: mxx8uHw2Wd2ERqweQSKSf1b1ubk49dPQ6WL3gSyknZY2EqtZSHivepjKlXcokcN1
Authorization: Basic aWlhYi1hZG1pbjpEMHduMW5nLTE5
Connection: keep-alive
Cookie: csrftoken=mxx8uHw2Wd2ERqweQSKSf1b1ubk49dPQ6WL3gSyknZY2EqtZSHivepjKlXcokcN1; kolibri=6stn1oovdjjvnjo2fssjzyynea1auwel
```

Response:

```
HTTP/1.1 403 Forbidden
Date: Tue, 30 Apr 2019 05:56:36 GMT
Server: 0.0.0.0
Content-Length: 71
Content-Language: en
Vary: Accept,Cookie,Accept-Language
Allow: GET, DELETE, HEAD, OPTIONS
X-Frame-Options: SAMEORIGIN
Content-Type: application/json
Set-Cookie: csrftoken=mxx8uHw2Wd2ERqweQSKSf1b1ubk49dPQ6WL3gSyknZY2EqtZSHivepjKlXcokcN1; expires=Tue, 28-Apr-2020 05:56:36 GMT; Max-Age=31449600; Path=/
Keep-Alive: timeout=1, max=8
Connection: Keep-Alive
```

holta, 2019-04-30:
> Thanks. It seems that it's a redirect caused in the session keep alive request that immediately returns 403 after the login.
> 
> The issue doesn't seem to be that the request is made to different hosts, so I would really suspect any captive portals and such to be playing part.

@KW4NP can you clarify (just in case this is related) if you installed { MIN-sized, MEDIUM-sized or BIG-sized } [local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) when you originally installed IIAB?

As the table shows in http://FAQ.IIAB.IO Item 9 ("What services (IIAB apps) are suggested during installation?") IIAB's Captive Portal is turned OFF within MIN-sized.

Note that IIAB's Captive Portal is also OFF by default in general:
https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L123-L129

But just to be sure, @KW4NP can you clarify the current state of these 2 variables in your /etc/iiab/local_vars.yml ?
```
captive_portal_install: False
captive_portal_enabled: False
```

KW4NP, 2019-04-30:
@holta , I think I started with the MIN-sized local_vars.yml but made some changes later.  I just now confirmed that captive portal is definitely off - both variables set to false.  Also, this problem is occurring even on the WAN side of the box.

rtibbles, 2019-04-30:
The only thing that could be causing a 403 on the session endpoint in this way would be if the CSRF token was missing or invalid: https://github.com/learningequality/kolibri/blob/release-v0.12.x/kolibri/core/auth/api.py#L316

There is no other authentication or validation on the endpoint.

The csrftoken cookie name is generic, so if you have any other services running on the same domain that also set a csrftoken with the default django name of csrftoken, then this could well cause an issue. Given that wiping the cookies completely fixes the issue, this sounds like the most likely cause.

Is anything else running that would set this cookie?

holta, 2019-04-30:
> captive portal is definitely off

Thanks @KW4NP

> problem is occurring even on the WAN side of the box

Is it definitely occuring on the LAN side i.e. http://172.18.96.1/kolibri via the RPi's WiFi AP?

Is it also occuring within the box itself i.e. http://localhost/kolibri if you have access from within?

FWIW good to keep track...in context with our initial bug reports from February (at the top of this ticket!)

Background: https://github.com/iiab/iiab/wiki/IIAB-Networking

KW4NP, 2019-05-01:
It seems to be occurring on the LAN side when using the WAN IP address, but the LAN IP address and hostname don't seem to give any errors.  I don't have a browser set up inside the box so have not checked localhost.

benjaoming, 2019-05-01:
@rtibbles I checked the `.har` traceback and the `csrftoken` cookie doesn't change value.

holta, 2019-05-01:
> Given that wiping the cookies completely fixes the issue

CLARIFICATION: That's not what @floydianslips experienced, when he spent many hours trying to diagnose this in December/January/February.

Deleting cookies was **not** sufficient to work around this problem, as he reconfirmed several times.

A complete reinstall of the browser (deleting browser history & all client-side state) **did** however work (as we've seen with both Chrome or Firefox).

rtibbles, 2019-05-01:
> @rtibbles I checked the .har traceback and the csrftoken cookie doesn't change value.

Right, my assumption was that it hadn't changed, when it should have - for Kolibri. The only other option for the 403 is if the session cookie is not getting preserved?

holta, 2019-05-01:
FWIW: a Chrome incognito window **does** work around this problem.

rtibbles, 2019-05-01:
> FWIW: a Chrome incognito window does work around this problem.

Right - that is what is making me think it is a clash of cookies causing chaos. Just not sure what the other source of cookies on the same domain might be.

holta, 2019-05-01:
> Right - that is what is making me think it is a clash of cookies causing chaos. Just not sure what the other source of cookies on the same domain might be.

Deleting Chrome cookies and restarting Chrome did not improve anything for @floydianslips 

Deleting Chrome's web browsing history did work for him, once again allowing him to login to http://box/kolibri

Confounding!

KW4NP, 2019-05-09:
This has started happening on the LAN side at the 172.18.96.1 address of my box.  However, it only started happening after I started doing some work on the box (admin console, KA Lite, Kiwix, etc.) from the LAN side, leading me to believe that the offending cookie (or something) is happening when I visit a different site on the box.

holta, 2019-05-09:
@KW4NP your ongoing reportage from the heart of Africa continues to be priceless in driving us+all towards an eventual solution, Thank You!

rtibbles, 2019-05-09:
> This has started happening on the LAN side at the 172.18.96.1 address of my box. However, it only started happening after I started doing some work on the box (admin console, KA Lite, Kiwix, etc.) [...]

This will be minimally reproduceable by:
a) Visiting KA Lite
b) Visiting Kolibri

They both use Django, and so are both setting a `csrftoken` cookie on the domain from which they are visited. If you are seeing differential behaviour between visiting over WAN vs over LAN, it will almost certainly be because of a different history of interaction for the domain via which you are accessing it.

When Kolibri tries to authenticate, it then uses the KA Lite csrftoken cookie, rather than the Kolibri one, and gives the error.

holta, 2019-05-10:
> When Kolibri tries to authenticate, it then uses the KA Lite csrftoken cookie, rather than the Kolibri one, and gives the error.

@KW4NP can you reproduce-or-corroborate the malignancy by using both Learning Equality apps, as @rtibbles has laid out?

KW4NP, 2019-05-10:
Tested on Chromium but cannot seem to reproduce the expected behavior.  Will try again on Firefox using a different address when I get a chance.

rtibbles, 2019-05-11:
This is being tracked on the Kolibri issue above, and a fix for this (and other issues that might arise for similar reasons) has been opened here: https://github.com/learningequality/kolibri/pull/5480

KW4NP, 2019-05-11:
@rtibbles , I can't seem to reproduce the behavior that you expected.  It seems reasonable and is also what I expected, but it must be a different site on the IIAB that is causing the problem as I have not been able to reproduce the problem by visiting KA-Lite on Firefox or Chromium.

holta, 2019-05-18:
Closeable now that Kolibri 0.12.3 (#1608) is part of IIAB?!

benjaoming, 2019-05-18:
Would be nice with a final confirmation from @KW4NP since the above comment seemed to indicate something not being reproduced that @rtibbles indicated should be reproducible in order for said cookie name theory to fit the case?

KW4NP, 2019-05-20:
Sorry, I can't confirm or deny that this fixes the problem as the box has been deployed and I probably won't see it again until sometime in July for updates.  I can check it at that point but others mention that upgrading Kolibri seems to temporarily fix the problem, so it might not come up again right away.

benjaoming, 2019-05-20:
Sorry to hear that @KW4NP - but deployments must be deployed! :)

holta, 2019-05-29:
Bad News :/

Even with Kolibri 0.12.3 I have in front of me an http://box/kolibri that continually redirects between these two:

http://10.8.0.22/kolibri/user/#/signin
http://10.8.0.22/kolibri/user/

FYI 10.8.0.22 is a pre-release of IIAB 7.0 installed on Ubuntu 16.04 on NUC hardware, where Kolibri was installed on top of 0.12.2 (is there some special procedure that needs to be run to completely remove prior versions of Kolibri?)  Or course if I do that, I will lose this current of the the failure mode (is there any log file data I need to gather to help diagnose?)

benjaoming, 2019-05-29:
> is there some special procedure that needs to be run to completely remove prior versions of Kolibri?

I take it that both were installed via `pip install` into the same environment? In this case, no. If they are different installation methods, then yes, that would go as "unsupported" and perhaps a bit "unpredictable" :) You can consult the version number displaying when you navigate to the Device => Info.

[image]

holta, 2019-05-29:
> > is there some special procedure that needs to be run to completely remove prior versions of Kolibri?
> 
> I take it that both were installed via `pip install` into the same environment? In this case, no.

Here Kolibri is always installed using pip (since this is an Internet-in-a-Box install or reinstall) as can be seen here:
https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L23-L30

> You can consult the version number displaying when you navigate to the Device => Info.
> 
> [image]

Remember I'm being blocked from logging into Kolibri, so I don't know how I'd get to the above screen...

So I use this instead:
```
root@box:~# /usr/local/kolibri/bin/kolibri --version
0.12.3
```
_(Is any other debugging / diagnosis info needed before I wipe Kolibri here?)_

benjaoming, 2019-05-29:
> Remember I'm being blocked from logging into Kolibri, so I don't know how I'd get to the above screen...

LOL, sorry.. award for worst support tip goes to me :1st_place_medal: I have, though, read your Ansible configurations before responding :)

@rtibbles what should we proceed with? Some more dumps from the Firefox web developer toolbar?

rtibbles, 2019-05-29:
A complete dump of the cookies on that domain would be good.

Also any server logs for the redirect, and anything from the console.log when the redirect is happening.

rtibbles, 2019-05-29:
By chance.. I have just managed to replicate a similar redirect loop locally during development.

rtibbles, 2019-05-29:
My guess now is that the server has not been provisioned properly, which is causing a redirect on the backend when it checks the session endpoint. Investigating further locally.

To confirm my suspicion, it would be good to see if you are getting something like this in the network requests in the browser:
```
"GET /static/assets/fonts/noto-full.NotoSans.400.woff HTTP/1.1" 200 432828
"GET /static/assets/fonts/noto-full.NotoSans.700.woff HTTP/1.1" 200 433488
"GET /api/auth/session/current/?active=false HTTP/1.1" 200 158
"GET /en/redirectuser/ HTTP/1.1" 302 0
"GET /en/user/ HTTP/1.1" 200 58677
```
(this is from my local Django console, but the URLs should be a similar sequence).

holta, 2019-05-29:
> To confirm my suspicion, it would be good to see if you are getting something like this in the network requests in the browser:
>
> ```
> "GET /static/assets/fonts/noto-full.NotoSans.400.woff HTTP/1.1" [...]

How and where do look at "network requests in the browser" in Chrome or Firefox to get you what you need?

rtibbles, 2019-05-29:
Under the dev tools -> Network tab will give a list of requests.

rtibbles, 2019-05-29:
Also, to see if my hypothesis of the cause is correct, what happens if you stop and then restart the Kolibri server, does the issue persist?

holta, 2019-05-29:
> Also, to see if my hypothesis of the cause is correct, what happens if you stop and then restart the Kolibri server, does the issue persist?

Rebooting did not solve the problem in the past (February 2019).

> Under the dev tools -> Network tab will give a list of requests.

Damn I screwed up and accidentally closed my completely overloaded browser (too many tabs) which was set to delete all cookies, and hence I can no longer reproduce the problem at this time &mdash; http://10.8.0.22/kolibri/user/#/signin works allowing logins at this time.

Apologies!

rtibbles, 2019-05-29:
OK, no problem - as it still seems to be related to the cookie persistence, it is probable that the way that I replicated it is not the underlying cause of what you are seeing.

[66 more replies]

closed 2020-06-03 by holta