iiab/iiab #1533: How do we make Calibre-Web logo+description auto-appear at http://box during IIAB installs?

holta, 2019-02-27:
@georgejhunt or @tim-moody can you help make Calibre-Web appear automatically on http://box.lan for all new IIAB installs where Calibre-Web is installed?

1. Just fyi MEDIUM-sized and BIG-sized installs of IIAB both include Calibre-Web, as confirmed by the table George seeded at http://FAQ.IIAB.IO Number 9.

2. Just fyi I cleaned up [js-menu/files/menu-files/menu-defs/en-calibreweb.json](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-calibreweb.json) this morning as part of https://github.com/iiab/iiab-admin-console/pull/155, so Calibre-Web's description is now more useful to educators etc.

tim-moody, 2019-02-27:
iiab_update_menus.py sets menu def to en-calibre-web, but it should be en-calibreweb

item 2 is irrelevant to the issue.

holta, 2019-02-27:
Solved by iiab/iiab-admin-console#156

closed 2019-02-27 by holta