iiab/iiab #1534: Wikipedia portals and project pages

aidan-fitz, 2019-02-27:
Certain Wikipedia project pages should be included in every offline distribution of Wikipedia, such as:

- [Wikipedia:About](https://en.wikipedia.org/wiki/Wikipedia:About)
- A subset of the [FAQ](https://en.wikipedia.org/wiki/Wikipedia:FAQ/Index), especially:
  - [Wikipedia:FAQ/Readers](https://en.wikipedia.org/wiki/Wikipedia:FAQ/Readers)
  - [Wikipedia:FAQ/Schools](https://en.wikipedia.org/wiki/Wikipedia:FAQ/Schools)
- Disclaimers
  - [Wikipedia:General disclaimer](https://en.wikipedia.org/wiki/Wikipedia:General_disclaimer), [Wikipedia:Content disclaimer](https://en.wikipedia.org/wiki/Wikipedia:Content_disclaimer), and [Wikipedia:Risk disclaimer](https://en.wikipedia.org/wiki/Wikipedia:Risk_disclaimer) for all distributions
  - [Wikipedia:Medical disclaimer](https://en.wikipedia.org/wiki/Wikipedia:Medical_disclaimer) for use in medical institutions
  - [Wikipedia:Legal disclaimer](https://en.wikipedia.org/wiki/Wikipedia:Legal_disclaimer) and [Wikipedia:Survey disclaimer](https://en.wikipedia.org/wiki/Wikipedia:Survey_disclaimer) are probably less important

Additionally, certain portals might be useful to offline readers, depending on the target audience:
- [Portal:Contents](https://en.wikipedia.org/wiki/Portal:Contents) - an index of portals on Wikipedia
  - Various [geography-related](https://en.wikipedia.org/wiki/Portal:Contents/Geography_and_places) portals
  - Various [health-related](https://en.wikipedia.org/wiki/Portal:Contents/Health_and_fitness) portals
  - Various [science-](https://en.wikipedia.org/wiki/Portal:Contents/Natural_and_physical_sciences) and [math-related](https://en.wikipedia.org/wiki/Portal:Contents/Mathematics_and_logic) portals
  - Various [philosophy-](https://en.wikipedia.org/wiki/Portal:Contents/Philosophy_and_thinking) and [religion-related](https://en.wikipedia.org/wiki/Portal:Contents/Religion_and_belief_systems) portals
- Specific portals:
  - [Portal:Agriculture](https://en.wikipedia.org/wiki/Portal:Agriculture)
  - [Portal:Business and economics](https://en.wikipedia.org/wiki/Portal:Business_and_economics)
  - [Portal:Civil engineering](https://en.wikipedia.org/wiki/Portal:Civil_engineering)
  - [Portal:Computing](https://en.wikipedia.org/wiki/Portal:Computing)
  - [Portal:Engineering](https://en.wikipedia.org/wiki/Portal:Engineering)
  - [Portal:Medicine](https://en.wikipedia.org/wiki/Portal:Medicine)

Obviously, these lists would have to be localized to whichever language is being used.

I actually edit the English Wikipedia semi-regularly as [Qzekrom](https://en.wikipedia.org/wiki/User:Qzekrom), so I have a sense of how the wiki and community work. I welcome any feedback you guys have.

holta, 2019-02-28:
Most Wikipedia "portals" and "project pages" have unfortunately not made it into production ZIM files over the years &mdash; for subtle reasons that @kelson42 and @tim-moody can hopefully explain?

> A distribution of Wiktionary (which is multi-lingual), Wikibooks, and Wikiversity could also be included.

These are already available as part of many Internet-in-a-Box installations.  Here's how you can take advantage:

- Explore http://box.lan/admin (typically with username/password iiab-admin/g0adm1n as per http://FAQ.IIAB.IO) and then click on the "Install Content" tab.

- This allows you to download/unpack/install the ZIM files at http://download.kiwix.org/zim/ as well as OER2Go/RACHEL modules at  http://oer2go.org &mdash; and then automatically creates a logo & basic description for your community's IIAB home page (http://box or http://box.lan or http://172.18.96.1 typically) ...that you can then drag+drop into your favorite position!

- Read http://FAQ.IIAB.IO #15 ("[How do I customize my Internet-in-a-Box home page?](http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_customize_my_Internet-in-a-Box_home_page.3F)")

holta, 2019-03-04:
@aidan-fitz there are limitations within the current scraping processes to create ZIM files from MediaWik, that might prevent graping portals?

In any case, see for example this excerpt from https://wiki.openzim.org/wiki/Build_your_ZIM_file#MediaWiki_Collection_Extension to try to confirm the facts:

> **MediaWiki Collection Extension**
>
> _Unfortunately this feature was removed. Please subscribe and show your interest on [this ticket](https://phabricator.wikimedia.org/T73660) if you want to see it back]._
>
> <strike>You can create on Wikipedia and all other Wikimedia projects, ZIM files of article selections. For example on http://en.wikipedia.org. This is based on the MediaWiki Collection Extension. The Collection extension for MediaWiki provides the ability to select articles and export them in different formats, such as PDF, ODF and ZIM. The main purpose of Collection extension is to create printed books - instead of export you can also choose to have the selected articles printed on demand as a real book.
>
> The Collection extension can be easily added to any MediaWiki installation:
>
> - [Get the extension](http://www.mediawiki.org/wiki/Extension:Collection)
> - [Details on Collection and openZIM](http://www.mediawiki.org/wiki/Extension:Collection/openZIM)</strike>

aidan-fitz, 2019-03-04:
According to that page, the Wikipedia Book Creator ([Special:Book]) can also export books in ZIM format, but I don't see it. Downloading (even as PDF) is currently disabled.

[Special:Book]: https://en.wikipedia.org/wiki/Special:Book

kelson42, 2019-03-04:
@holta @aidan-fitz If something is not OK or can be improve in a ZIM, this should be reported here https://github.com/openzim/zim-requests. If that has already been done, could you please put the links as reference to that discussion? I think this would be helpful then.

ericnitschke, 2019-03-04:
I use this portal every day: https://en.wikipedia.org/wiki/Portal:Current_events

Would love to eventually have a daily "News" tab on certain IIAB splash pages, scraped from this portal.

Potentially relevant links:

Zim Library: https://github.com/openzim/libzim

**Zim Scrapers/Converters**
HTTP Sites: https://github.com/openzim/zimit/
Youtube: https://github.com/openzim/youtube
TED: https://github.com/openzim/ted
ZIP to ZIM: https://github.com/openzim/zip2zim
MediaWiki: https://github.com/openzim/mwoffliner
OpenEdX: https://github.com/openzim/openedx
Stack Overflow: https://github.com/openzim/sotoki
Gutenberg: https://github.com/openzim/gutenberg
Phet: https://github.com/openzim/phet

http://www.openzim.org/wiki/Build_your_ZIM_file

aidan-fitz, 2019-03-09:
@kelson42 @ericnitschke The openZIM website is very slow; I actually find it faster to browse [on the Internet Archive]. I suggest moving it to a public cloud platform if it isn't already, and hosting it on a [container service] to reduce resource costs.

Suggestion aside, do you know why the website is slow?

[on the Internet Archive]: https://web.archive.org/web/20190218125802/https://wiki.openzim.org/wiki/Main_Page
[container service]: https://searchitoperations.techtarget.com/definition/Containers-as-a-Service-CaaS