iiab/iiab #1536: IIAB installer should make hotspot & networking choices far more understandable

holta, 2019-02-28:
Excellent design ideas from @georgejhunt & @jvonau during our weekly call this morning!

Apologies if I did not capture it all, but this is a start &mdash; see meeting minutes item 1. b. under Feb 28, 2019 here: http://minutes.iiab.io

holta, 2019-03-28:
#1582 & #1583 are big steps forward in making http://d.iiab.io tangible to newcomers.

Now things like #1575 (more automated reporting of networking & similar issues) are needed to keep grassroots communities' trials/tribulations circulating, so this kind of "fieldback" keeps organically growing...

jvonau, 2021-08-17:
Given this was tagged for 8.0 in 2020 why the sudden interest in networking a year later when this is a 8.0 objective?

jvonau, 2022-05-11:
there years later, is it good enough?

holta, 2022-05-11:
> there years later, is it good enough?

It's a lot better!

Things like these below should also stand to greatly help community onboarding &mdash; as almost everyone finds networking overwhelming &mdash; educators especially, who core focus is learning/content/usability:

- PR #3070
- PR #3173
- PR #3179 for #3097

holta, 2022-07-20:
FYI latest progress is here:

- PR #3314
  - cf2b5a409755c567ca263b35e7cad6fa529de219
- PR #3313

jvonau, 2025-01-27:
status?

holta, 2025-01-27:
There's a lot of progress since 2019!  So let's close this ticket and follow up elsewhere where introductory doc/UX polish is needed, via other tickets like:

- #1520
- #2863
- #3385
- #3403

closed 2025-01-27 by holta