iiab/iiab #1537: IIAB blocked from installing on Raspbian Desktop due to LXDE-pi popup

holta, 2019-02-28:
Emily @ UArkansas is blocked from installing MIN-sized IIAB 7.0/master on Raspbian Desktop due to:

> TASK [iiab-admin : Put line in /home/pi/.config/lxsession/LXDE-pi/autostart to run the above (raspbian)] ****************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /home/pi/.config/lxsession/LXDE-pi/autostart does not exist !", "rc": 257}
	to retry, use: --limit @/opt/iiab/iiab/iiab-support.retry

Rebooting and re-running "sudo iiab" does not help.

It sounds like Raspbian's new OOB install wizard (popup that asks you to set the language, keyboard, TZ etc) may be colliding with IIAB's attempt to create /home/pi/.config/lxsession/LXDE-pi/autostart

_...in some but not all conditions?_  It appears IIAB _might be_ installable on Raspbian Desktop if you do complete the OSs install wizard?  (Or perhaps it's if you _don't_ complete Raspbian Desktop's install wizard??)  Sorry the facts are not yet 100% clear.

@georgejhunt may understand this but I do not yet.  Exact pattern of what causes this serious install failure needs to identified.  I'll talk to Emily again within 24h, hoping the root cause becomes more clear...

holta, 2019-02-28:
@georgejhunt should we force the creation of `/home/pi/.config/lxsession/LXDE-pi/autostart` in those cases where it no longer exists?

What template file should be used if so?

(Or is simply "touching" this file safe/acceptable, to create an empty such `autostart` file?)

holta, 2019-03-01:
Let's try to make this code more resilient/robust as we fix it &mdash; e.g. what are good ways to survive changes that Raspbian Desktop might make within /home/pi/.config/lxsession/LXDE-pi/* over the course of time?

holta, 2019-03-01:
A detailed diagnosis is not something UArkansas can do for us.

In the end one of us will need to poke at (observe) Raspbian Desktop's own `/home/pi/.config/lxsession/LXDE-pi/autostart` while running and/or bypassing the OS's out-of-the-box wizard, on a fresh install of the Raspbian Desktop OS.

And then it should be easy work to clean up our code here appropriately: (so IIAB can install reliably onto Raspbian Desktop)
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L29-L44

georgejhunt, 2019-03-01:
The last time I looked at this, the missing file/directory was part of the
raspbian image. I concluded that the only way the file/directory could be
missing was if the image copy had been defective -- or perhaps a defective
SD card. So a verify after copy is the way to go.
So when I look at win32 diskimager, it looks like there is a verify
function.

holta, 2019-03-02:
> The last time I looked at this, the missing file/directory was part of the raspbian image. I concluded that the only way the file/directory could be missing was if the image copy had been defective [...]

I'm assuming the above is impossible, given Emily @ Univ of Arkansas is using https://Etcher.io which by default does 2-passes, including a full verification of the burned/flashed OS image.

So my assumption (for now) is that IIAB's failure to install depends on what exactly you do with Raspbian Desktop after you've flashed it onto microSD:
- Do you ssh in or do you use the OS's "mandatory" graphical install wizard?
- Do you complete the OS's "mandatory" graphical install wizard or click skip some steps?

My (guess) is that one of the above patterns (or similar!) causes /home/pi/.config/lxsession/LXDE-pi/autostart to disappear.

holta, 2019-03-07:
@georgejhunt I've confirmed UArkansas' report that `/home/pi/.config/lxsession/LXDE-pi/autostart` generally does not exist in Raspbian Desktop after the OS's "mandatory" install wizard has completed (to set Time Zone, language, etc) typically run during 1st visual/graphical boot:
```
root@box:~# ls -l /home/pi/.config/lxsession/LXDE-pi/autostart
ls: cannot access '/home/pi/.config/lxsession/LXDE-pi/autostart': No such file or directory
```
As such can you help revise [roles/iiab-admin/tasks/main.yml Lines 29-44](https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L29-L44) so that IIAB is quite resiliently installable on Raspbian Desktop &mdash; regardless whether the wizard has been run or not?

_(FWIW a growing number of IIAB teachers and implementers have begun requiring Raspbian Desktop in recent months...not all for the same reasons...but in any case, this is definitely worth fixing !)_

holta, 2019-03-07:
/home/pi/.config/lxsession (and autostart therein) no longer exist in Raspbian 2018-11-13:
https://www.raspberrypi.org/forums/viewtopic.php?t=228532

@georgejhunt hopefully you can help navigate Raspbian's new arrangement?  Perhaps these extracts help &mdash; fyi these were run on Raspbian Desktop prior to completing the OS's out-of-box wizard:
```
root@raspberrypi:~# locate autostart
/etc/xdg/autostart
/etc/xdg/autostart/gsettings-data-convert.desktop
/etc/xdg/autostart/lxpolkit.desktop
/etc/xdg/autostart/piwiz.desktop
/etc/xdg/autostart/pprompt.desktop
/etc/xdg/lxsession/LXDE-pi/autostart
/etc/xdg/lxsession/LXDE/autostart
/etc/xdg/openbox/autostart
/usr/bin/lxsession-xdg-autostart
/usr/lib/arm-linux-gnueabihf/openbox-autostart
/usr/lib/arm-linux-gnueabihf/openbox-xdg-autostart
/usr/share/man/man1/lxsession-xdg-autostart.1.gz
```
```
root@raspberrypi:~# locate LXDE-pi
/etc/xdg/lxpanel/LXDE-pi
/etc/xdg/lxpanel/LXDE-pi/config
/etc/xdg/lxpanel/LXDE-pi/panels
/etc/xdg/lxpanel/LXDE-pi/panels/panel
/etc/xdg/lxsession/LXDE-pi
/etc/xdg/lxsession/LXDE-pi/autostart
/etc/xdg/lxsession/LXDE-pi/desktop.conf
/etc/xdg/lxsession/LXDE-pi/sshpwd.sh
/etc/xdg/pcmanfm/LXDE-pi
/etc/xdg/pcmanfm/LXDE-pi/desktop-items-0.conf
/etc/xdg/pcmanfm/LXDE-pi/pcmanfm.conf
/home/pi/.cache/lxsession/LXDE-pi
/home/pi/.cache/lxsession/LXDE-pi/run.log
/home/pi/.config/lxpanel/LXDE-pi
/home/pi/.config/lxpanel/LXDE-pi/panels
```

holta, 2019-03-07:
@georgejhunt several people say that Raspbian 2018-11-13 no longer includes LXDE, and to try editing `/etc/lightdm/lightdm.conf` instead of /home/pi/.config/lxsession/LXDE-pi/autostart:

- https://github.com/MichMich/MagicMirror/issues/1473
- https://raspberrypi.stackexchange.com/questions/90749/cant-find-home-pi-config-lxsession-lxde-pi-autostart/90985

holta, 2019-03-07:
I'm assigning this to @georgejhunt & myself to help keep organized.  But others please feel free to help too!

holta, 2019-03-07:
_Not our concern, but just FYI as a side note:_

Raspbian Desktop's "Welcome to Raspberry pi" wizard pops up rather immediately on new installs &mdash; and then if you don't touch anything its "Warning" popup about your weak/unchanged 'pi' password lands right smack on top of that (a minute-or-few-minutes later).

i.e. it's an awkward GUI design that can be very off-putting to newcomers...but as an introductory GUI at least it largely works and brings more ppl to explore the DIY/RPi/Linux possibilities...and will prob be made cleaner in coming years (-:

holta, 2019-03-07:
@georgejhunt does we really need/want `/etc/lightdm/lightdm.conf` in the end??

Certainly the 2 underlying / auto-triggered commands below (`sudo piwiz` and `sh /etc/xdg/lxsession/LXDE-pi/sshpwd.sh`) reveal possible paths forward:
```
root@raspberrypi:~# more /etc/xdg/autostart/piwiz.desktop
[Desktop Entry]
Type=Application
Name=Raspberry Pi First-Run Wizard
Name[en_GB]=Raspberry Pi First-Run Wizard
GenericName=Raspberry Pi First-Run Wizard
GenericName[en_GB]=Raspberry Pi First-Run Wizard
Comment=Configure Raspberry Pi system on first run
Comment[en_GB]=Configure Raspberry Pi system on first run
Icon=rpi
Exec=sudo piwiz
NotShowIn=GNOME;XFCE;
StartupNotify=true
Categories=Settings;DesktopSettings;GTK;X-LXDE-Settings;
```
```
root@raspberrypi:~# more /etc/xdg/autostart/pprompt.desktop
[Desktop Entry]
Type=Application
Name=pprompt
Comment=Prompt to change default password if ssh enabled
NoDisplay=true
Exec=sh /etc/xdg/lxsession/LXDE-pi/sshpwd.sh
NotShowIn=GNOME;KDE;XFCE;
```

holta, 2019-03-07:
FYI here are the 2 new password-warning scripts for (1) graphical/desktop and (2) shell &mdash; as they ship within Raspbian Desktop 2018-11-13:
```
root@raspberrypi:~# cat /etc/xdg/lxsession/LXDE-pi/sshpwd.sh
#!/bin/bash

export TEXTDOMAIN=pprompt

. gettext.sh

if [ -e /run/sshwarn ] ; then
        zenity --warning --width=400 --text="$(gettext "SSH is enabled and the default password for the 'pi' user has not been changed.\n\nThis is a security risk - please login as the 'pi' user and run Raspberry Pi Configuration to set a new password.")"
fi
```
```
root@raspberrypi:~# cat /etc/profile.d/sshpwd.sh
export TEXTDOMAIN=Linux-PAM

. gettext.sh

if [ -e /run/sshwarn ] ; then
    echo
        echo $(/usr/bin/gettext "SSH is enabled and the default password for the 'pi' user has not been changed.")
        echo $(/usr/bin/gettext "This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.")
        echo
fi
```

PR #1551 (2019-03-08): Fix iiab-admin published password warnings for Raspbian Desktop 2018-11-13 especially
Fixes #1537, the visual warning (popup) especially.

Also cleans up the text warning on shell and ssh login.

Testing ongoing.
closed 2019-03-08 by holta