iiab/iiab #1538: How does a teacher install a ZIM file from Windows to IIAB? (using a FAT/FAT32 USB stick if poss?)

holta, 2019-03-01:
The folks at Univ of Arkansas tried repeatedly but could not get the ZIM file installed.

Maybe we just need to explain this better?  These 2 docs certainly aren't enough for now:
https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content ?
http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_add_ZIM_files.2C_like_Wikipedia.3F

@tim-moody just clarified that:
- The USB stick must be laid out the same as IIAB (but I don't understand what that means on FAT, FAT32, exFAT and NTFS)
- It seems a Linux-like filesystem (e.g. ext4) is recommended if possible, but might not be required?
- It seem it might work on FAT/FAT32 if these 2 files are created?
  - E:\library\zims\content\RANDOM.zim
  - E:\library\zims\library.xml
- Can Admin Console -> Install Content -> Manage Content install the needed directories onto a FAT-or-similar USB stick?

Finally when this is well-understood, we should write up an actual/foolproof step-by-step sequence that less-technical teachers can follow to make this happen-

holta, 2019-03-01:
It appears likely exFAT is being shipped on many/most new larger USB memory sticks (32+ GB) going forward.  As we also support for http://box/usb (per FAQ.IIAB.IO Question 4 = "[Can teachers display their own content?](http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F)")

So if we can work with locals/teachers existing strong expectations around exFAT/FAT/FAT32 sticks (to move around large volumes of content, and back it up on pre-existing Windows HDD's) that'll be very powerful.

Conversely if it's not realistic to use USB sticks in the most popular/predominant format here, that's unfortunate but we'll just have to warn folks and provide alternatives~

holta, 2019-03-02:
The biggest opportunity here is to make aspects of Admin Console's -> Install Content -> [Manage Content](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content) work for "non geeks".

Specifically IIAB's remote teachers and operators **crave** Sneakernet-of-Alexandria pragmatism that "just works" with their pre-existing FAT32-like sticks.

Like any mix-tape or family photo collection or podcasts that most all cultures already carry around in their USB memory sticks, IIAB needs to work within existing cultural norms as much as possible, to be taken seriously.

Documenting precise recipes as to what exactly works for 2019 is the place to start.  Then @m-anish will help us turn these into very short videos, as these practical/evolving usability facts become ever more clear.

m-anish, 2019-03-02:
> The biggest opportunity here is to make aspects of Admin Console's -> Install Content -> [Manage Content](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content) work for "non geeks".
> 
> Specifically IIAB's remote teachers and operators **crave** Sneakernet-of-Alexandria pragmatism that "just works" with their pre-existing FAT32-like sticks.
> 
> Like any mix-tape or family photo collection or podcasts that most all cultures already carry around in their USB memory sticks and pockets, IIAB needs to work within existing cultural norms as much as possible, to be taken seriously.
> 
> Documenting precise recipes as to what exactly works for 2019 is the place to start. Then @m-anish will help us turn these into very short videos, as these practical/evolving usability facts become ever more clear.

Infact, this can be a good smoke test. Can you produce a sequence of steps - clearly defined, that I can follow to do what is being asked in this issue. I can produce a video following that sequence.

holta, 2019-03-02:
> Infact, this can be a good smoke test. Can you produce a sequence of steps - clearly defined, that I can follow to do what is being asked in this issue. I can produce a video following that sequence.

Perfect.  As we do this with Univ of Arkansas let's also keep in mind essential UX/usability pieces like https://github.com/iiab/iiab/issues/1530 ("Calibre-Web: fix cover/title thumbnails for PDF's and other docs/books", and related WordPress pragmatism for "non geeks") that @tim-moody and others are working on.

Offline Social Media (OSM!) is in fact the very fabric that democracy is built from...we obviously didn't invent the grassroots forces that make sneakernets-of-Alexandria possible...but we need to unleash the "Offline YouTubes" of the planet where millions of people want more intelligent choices than just TV reruns/radio payola...

holta, 2019-03-02:
This or a similar video can also address common grassroots curation ("Sneakernet of Alexandria") questions like:

https://github.com/iiab/iiab/issues/828#issuecomment-468933366 Copy Content Packs to/from USB sticks/storage
#857 Auto-display of ZIM files when Teachers insert USB sticks/drives
https://github.com/iiab/iiab/issues/1002#issuecomment-462183401 Field editing of Content Pack descriptions/logos by non-technical operators
#1530 Calibre-Web: fix cover/title thumbnails for PDF's and other docs/books

holta, 2019-03-05:
@arky @tim-moody  @m-anish

Kolibri empowers "offline" civic replication...i.e. some of the raw ingredients we need to cultivate Sneakernet-of-Alexandria community dissemination action...with 2 mechanisms to export Content Pack (channels):

1. ...to USB sticks/drives, including everyday media like FAT/FAT32/exFAT/NTFS:
https://kolibri.readthedocs.io/en/latest/manage.html#export-from-kolibri-to-local-drive
The resulting USB sticks appear a bit messy (Content Packs are not quite self-contained in cleanly-named directories?) but it does work:
- `__KOLIBRI__DATA__` placed at the root of the USB device
- content storage not human understandable, it's based on hash of contents, e.g. `/content/storage/9/8/98113r161732763726376.mp4`
- channel metadata stored in the databases/{{channel_id}}.sqlite3

2. ...to another Kolibri instance on the (e.g. local) network:
https://kolibri.readthedocs.io/en/latest/manage.html#import-content-from-a-local-network-or-internet

3. Kolibri unfortunately does NOT export Content Pack (channels) to Android at this time. 
 Possibly some students would "try to learn too much" mass-downloading content to their phones even if Raspberry Pi-like servers are slowed down for other students?  Needs some design refining, so "hoarders" helping export/import to nearby villages are welcome — _but at the same time do not damage the learning opportunities of students present & learning in school at that time!_

llovemixo, 2020-01-14:
A suggestion of a simple way for new users to add zims:

Copy an existing zim (test zim ?) to usb using Admin Console.  All the needed directory structure will then be present.
Add the desired new zim  to same directory as copied zim.
Install the new zim from the usb using Admin Console.

For the menu item, clone an existing zim (zim test ?) menu item and change fields in the same format.
This avoids confusion of names in the field for content and menu code.
The menu code starts with "en-" for modules and osm maps, but the zim values do NOT start with "en-" .  For modules and osm maps, the content field also starts with "en-", but for zims,  content does NOT start with "en-".

I have not tested this with zim, but have tested with modules.
The error message (from /library/www/html/undefined) for invalid names  describes usage for module, but not zim.

tim-moody, 2020-01-14:
Does cloning test.zim work? It is very minimal. You could choose something closer to the target zim, especially to get special fields in the footnote. Also, you should make sure the menu def does not already exist before creating by cloning another.

btw, if you download a zim it automatically gets added to the menu and a new menu def is generated if it does not exist.