iiab/iiab #1539: Digital agriculture

aidan-fitz, 2019-03-02:
I'm thinking that smallholders could use the IIAB as part of a digital agriculture system, e.g. [FarmBeats]. I just had the idea for this (I'm at an [ag hackathon] right now), but I have nothing specific in mind.

Some questions:
- What software apps for small farmers could we ship with IIAB?
- How could IIAB be integrated into a digital agriculture system consisting of both hardware and software?

[FarmBeats]: https://www.microsoft.com/en-us/research/project/farmbeats-iot-agriculture/
[ag hackathon]: https://dahackathon.splashthat.com/

holta, 2019-03-02:
Very challenging & important question!

@aidan-fitz are you familiar with http://www.digitalgreen.org which does similar work, apparently having quite some success with small rural skillsharing workshops using projectors?

m-anish, 2019-03-02:
[Related issue](https://github.com/eka-foundation/home/issues/29)

aidan-fitz, 2019-03-02:
I haven't heard of it. Thank you for the pointer!

aidan-fitz, 2019-03-03:
*Comment:* My team at the hackathon realized that agricultural management software would benefit farms of all sizes because farmers have to make a lot of careful, potentially high-stakes management decisions every day. We're designing a desktop/web app that would integrate data from a farm management system (e.g. [DairyComp]) with commodity prices from e.g. the [USDA] or the farm's co-op. However, our design assumes that the customer has a reliable Internet connection and that machine-readable market data for their region are available.

[DairyComp]: https://web.vas.com/dairycomp/
[USDA]: https://mymarketnews.ams.usda.gov/

aidan-fitz, 2019-03-03:
- [farmOS](https://farmos.org/) - GPL, web (Drupal), mobile (iOS/Android)

More at [awesome-agriculture](https://github.com/beaorn/awesome-agriculture)

mikkokotila, 2019-03-04:
[Farmism](http://farmism.org/) is also related with this. It's basically a simulator for micro-sized farm planning based on an economic model that was built over 4 years collecting data from farmers on the ground.

holta, 2019-04-28:
Patrick O'Connor recommends:

https://soilandhealth.org