iiab/iiab #1540: DESIGN, PROPOSAL | Rethink network, avahi, zeroconf, link-local for decentralized and multiple IIAB's on a network

m-anish, 2019-03-03:
This is an umbrella ticket meant to track design discussions and work related to supporting multiple IIAB's over a network. 

The premise is this:
In a small, medium sized mesh network with many users, having one IIAB poses a few challenges:
1. It becomes a single point of failure vis-a-vis DHCP, DNS, Gateway, Content serving.
2. IIAB is difficult to update, other than the content part, atleast while in-field. This slows down progress introducing new features. 

It would be better if we could support multiple IIAB's over a network in a decentralized fashion. Some ideas (and this is not a well vetted, or exhaustive list, so would love to discuss this) would be:
1. Use avahi more smartly. Create a structure, wherein, all IIAB services installed in an instance are advertised with machine readable details. This might include things like:
    1. Service access URL, port
    2. Service description
    3. Any sample credentials
2. Think about supporting a link-local configuration. Challenges here are:
    1. How would DNS work in such a situation?
    2. Would some services complain if working over a link-local network, rather than a typical LAN subnet.
    3. Any other unforeseen problems
    4. Alternatively, think about decentralized dhcp through a shared leases file.
3. Use mDNS, DNS-SD to aid in service discovery
4. One could also think of a web-client kind of front end. For example, think of a box.lan homepage, that is populated by everything that is advertised (and verified using keys if necessary) on the network. Such a page would be dynamic, and would support multiple IIAB's on a network, while making it transparent to the user. 

These enhancements are only slightly interrelated and can be done bit by bit. For example, we could begin with making IIAB services announce themselves on avahi, for a start.

Thoughts?

Happy to discuss on a call if people are willing to participate in this discussion.

m-anish, 2019-03-03:
tagging @pdey and @mikkokotila

m-anish, 2019-03-03:
##Links: 

* [Service Name and Transport Protocol Port Number Registry](https://www.iana.org/assignments/service-names-port-numbers/service-names-port-numbers.xhtml)
* [Python zeroconf project](https://pypi.org/project/zeroconf/) (there may be other better ones)
* [A js library](https://github.com/watson/bonjour) (there may be other better ones)

holta, 2019-03-03:
Tangentially related:
#1205 Signal, Briar, Manyverse, Serval, Riot.im on Matrix, Rocket.Chat, F-Droid client &/or server?

mikkokotila, 2019-03-03:
Shall we get on to a call to discuss this? It might be a good basis for a meaningful thread.

holta, 2019-03-03:
Internet-in-a-Box should also consider endorsing "the best" Android filesharing apps among things like:

SHAREit, Xender, Zapya, AirDroid, Superbeam, Portal/PushBullet, Send Anywhere (etc!)

Sneakernet-of-Alexandria refs:
#828 #857 #1002 #1530 #1538

m-anish, 2019-03-12:
I've started this experimentation with #1561 which may be merged without issues as far as I can tell.