iiab/iiab #1541: Can IIAB recommend/include an Android filesharing app for offline communities?

holta, 2019-03-03:
Which are most recommended among things like SHAREit, Xender, Zapya, AirDroid, Superbeam, Portal/PushBullet, Send Anywhere ?

Sneakernet-of-Alexandria issues &mdash; directly or indirectly related:
#828 #857 #1002 #1530 #1538 #1540

holta, 2019-03-04:
Also consider Manyverse which has added Bluetooth sync?

https://github.com/iiab/iiab/issues/1205#issuecomment-469239835

holta, 2019-03-24:
This difficult question was discussed during Thursday's call (http://minutes.iiab.io) in the context of local community mapmaking like:

#1568 Magrit: Cartography Teaching+Learning as part of IIAB

tim-moody, 2019-03-24:
nextcloud

adbenitez, 2019-08-07:
TrebleShot is a free software app for files and apk sharing (similar to Zapya)
there also exist NitroShare which is also free software, you can find them on F-Droid
(I recommed TrebleShot)