iiab/iiab #1542: what do you think about mastodon?

mikkokotila, 2019-03-04:
To avoid doubt, I'm not sure about this myself in terms of anything other than discussing it, but Mastodon does seem potentially interesting: https://github.com/tootsuite/mastodon

holta, 2019-03-04:
Thanks @mikkokotila !

Offline Social Media urgently needs an answer, that will be respected by global grassroots.

We need to evaluate pragmatic citizen-centric motorcycle-to-motorcycle solutions that lead IIAB/etc to interoperate with the Android and/or FAT32 sticks that regular folks already have ingrained in their informational/educational lives.  To exchange personal photos/education jewels/touching podcasts/etc that purposefully involve people &mdash; possibly using approaches like:

- https://github.com/iiab/iiab/issues/828 Copy Content Packs to/from USB sticks/storage
- https://github.com/iiab/iiab/issues/1205#issuecomment-469239835 Signal, Briar, Manyverse, Serval, Riot.im on Matrix, Rocket.Chat, F-Droid client &/or server?
- https://github.com/iiab/iiab/issues/1530 Calibre-Web (personal/community media cache, nurturing local curation of PDF's & e-books etc, with instant/automatic thumbnails to make it appealing?)
- https://github.com/iiab/iiab/issues/1538 How teachers install offline content like ZIM files, using normal USB sticks?
- https://github.com/iiab/iiab/issues/1541 Can IIAB recommend/include an Android filesharing app for offline communities?

holta, 2019-03-05:
fwiw Mastodon is used for "legalized child pornography" in Japan...which is (in fact!) the very last thing IIAB schools want to be involved with:
http://www.ethanzuckerman.com/blog/2017/08/18/mastodon-is-big-in-japan-the-reason-why-is-uncomfortable/

Ref: #1546