iiab/iiab #1543: Use different branch of iiab/iiab while installing

aidan-fitz, 2019-03-04:
I'm setting up a staging VM for #1242. Is there a simple way to use a different branch of the iiab/iiab repository while installing IIAB (either through the installer or from scratch)?

holta, 2019-03-04:
> Is there a simple way to use a different branch of the iiab/iiab repository while installing IIAB (either through the installer or from scratch)?

Yes:
1. Run `curl d.iiab.io/install.txt | sudo bash`
2. When it asks you "Edit /etc/iiab/local_vars.yml to customize your Internet-in-a-Box? [Y/n]", hit [Enter] to take the default (YES).
3. Edit Line 183 of [/usr/sbin/iiab](https://github.com/iiab/iiab-factory/blob/master/iiab#L183) so that it it specifies a specific branch or commit that you want to clone.  An example Line 183 (from http://d.iiab.io/6.7/iiab) is here:
```
git clone https://github.com/iiab/iiab -b release-6.7 --depth 1
```
4. Run `sudo iiab` to continue installing IIAB, as per usual.

aidan-fitz, 2019-03-04:
Probably shouldn't be `sudo bash` as this runs the installer as root, resulting in all repositories being owned by root.

holta, 2019-03-04:
> Probably shouldn't be `sudo bash` as this runs the installer as root, resulting in all repositories being owned by root.

What are you recommending instead?  User `iiab-admin` possibly?

Note that `iiab-admin` also has sudoer / root privs [*] as a result of:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/admin-user.yml

[*] in order to check for weak passwords.

PS http://FAQ.IIAB.IO Question 28 ("What security tips exist?") & http://wiki.laptop.org/go/IIAB/Security should be updated with the latest security recommendations regardless!

aidan-fitz, 2019-03-04:
Does running as the current user work?

holta, 2019-03-04:
> Does running as the current user work?

Installing IIAB requires root (sudoer) access at many different levels.  So if I'm understanding your question correctly (?) the answer is a definite no.

aidan-fitz, 2019-03-05:
I see. Well, the side effect of this is that the entire `/opt/iiab` directory tree is owned by `root:root`, which results in having to explicitly `sudo` in order to do anything in there. Maybe we can set the `setuid` or `setgid` bit for some scripts?

holta, 2019-03-05:
> the entire `/opt/iiab` directory tree is owned by `root:root`

That' would seem to have been your choice as the author of the Gitea playbook, if you created it as such, without using Ansible parameters such as `owner`, `group` and `mode`.

If you want other permissions after the fact, why not do as all other playbook authors do, using Ansible's `file` module which is one one way for you to recursively fix ownership/permissions, as mentioned at https://github.com/iiab/iiab/pull/1242#issuecomment-469485982 ?

aidan-fitz, 2019-03-05:
To clarify, `/opt/iiab` is where all of the IIAB-related repositories are, whereas `/library/iiab` is where programs get installed. The file permissions within `/opt/iiab` are therefore not controlled by Ansible.

holta, 2019-03-05:
> /library/iiab is where programs get installed.

In many cases yes.

In other case, programs are installed within /usr (and occasionally others places too).

> The file permissions within /opt/iiab are therefore not controlled by Ansible.

Why not?  Ansible can change essentially any file's ownership & permissions.

holta, 2019-03-05:
Clarification: there are other things aside from repos in `/opt/iiab`, for example :

- /opt/iiab/downloads
- /opt/iiab/kiwix

FWIW these have various permissions/ownership.

aidan-fitz, 2019-03-05:
I see.

On my system, they are all owned by root:
```
$ ls -l
total 28
drwxr-xr-x 2 root root 4096 Mar  4 19:12 downloads
drwxr-xr-x 8 root root 4096 Mar  4 20:12 iiab
drwxr-xr-x 8 root root 4096 Mar  4 19:14 iiab-admin-console
drwxr-xr-x 7 root root 4096 Mar  4 19:05 iiab-factory
drwxr-xr-x 3 root root 4096 Mar  4 19:12 kiwix
drwxr-xr-x 2 root root 4096 Mar  4 19:14 pip-packages
drwxr-xr-x 2 root root 4096 Mar  4 19:07 yum-packages
```

I think they don't need to be. `/opt/iiab` especially should not be owned by root as this gives the installers too many privileges and prevents git from being run on it except through sudo. We can add a task that `chown`s this entire directory to `iiab-admin` during the first run.

holta, 2019-03-18:
Clarif: user iiab-admin has sudoer (root) privileges, so that it can police against published passwords (and provide remainders at several crucit junctures).

Details here:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/admin-user.yml
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L24-L54
https://github.com/iiab/iiab-factory/blob/master/iiab#L52-L86

holta, 2019-03-18:
fyi @floydianslips too, set up his own IIAB dev environment as above (https://github.com/iiab/iiab/issues/1543#issuecomment-469376017)

Should this recipe be added to https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide ?

(Or can we bring this tkt to a close in another way?)

floydianslips, 2019-03-18:
> IIAB dev environment as above (#1543 (comment)

I'm not sure I set it up like that. I just pointed the install script to
clone my fork of iiab/iiab.

closed 2019-03-19 by aidan-fitz