iiab/iiab #1544: Containerized IIAB

aidan-fitz, 2019-03-04:
What are the exact obstacles to running IIAB in a container? I know Ansible is part of the issue. I think setup time would be much faster if this were possible.

deldesir, 2019-03-18:
Interesting question. To dig deeper, let's assume you need to install and run only what you need, would running IIAB as a container give you the "freedom" to do that? With all those services and apps, even as a container, IIAB is in fact _**platforms**_-in-a-plaform. IMHO Containerization would benefit IIAB for performance purpose if applied directly to apps not as a whole system.

holta, 2019-03-18:
Please consult @darkenvy & @tim-moody who have (different!) opinions here if/when progress is possible here.

My own assumption (for now) is that we need to wait for the details of Raspberry Pi 4 in 2020, before genuinely pragmatic Docker-like strategies emerge...

tim-moody, 2019-03-18:
Wouldn't it be more important to ask what the benefits are before asking what the obstacles are. The fastest way to get a development environment is to clone a vm, beats both vagrant and a docker container. The fastest way to get a deployment is an image.

A dockerized app is estimated to run at 97% of the speed of the app itself. How does adding a bunch of such components result in an overall performance increase?

Most docker containers I've seen are just another means of automation in that they aren't really prebuilt images, just a list of steps to build the container, so I'm not clear how that makes set up faster. it's true that ansible isn't very performant, but is docker more so?

To my mind the biggest obstacle is that containers are not computers. They expect to run applications in enterprise environments where spinning up an instance on some arbitrary machine and moving it elsewhere is a desirable characteristic. They take the network configuration to be a given, something IIAB tries to configure.

Docker works great for something like mwoffliner, an app that reads some stuff, does some work, and writes some stuff. It is a nightmare to install and configure and docker gets rid of all of that. But I'm not convinced that docker has enough benefit over ansible to justify the rewrite for IIAB.

Sadly, I know, docker is just too cool to resist.

aidan-fitz, 2019-03-18:
@tim-moody To be clear, I'm talking about using a container instead of a VM to test, not about build automation with Docker instead of Ansible.