iiab/iiab #1545: Kolbri Big Scrub for IIAB 7.0

holta, 2019-03-05:
@arky let's track all these small Kolibri issues that have emerged recently, so everything comes together smoothly for [IIAB 7.0](https://github.com/iiab/iiab/milestone/5):

- [ ] #977 Kolibri 0.10.2+ Release Polish (let's close out this former checklist in coming days!)
- [ ] #1094 Kolibri did not start within 2MIN timeout during BIG install on RPi 3
- [ ] #1314 Refine Kolibri 0.11 to remove red/pink warnings during (re)install
- [ ] #1387 Kolibri not quite working on Debian 10 "Buster" ?
- [ ] #1428 Validate Kolibri 0.11.1 -> 0.11.2 on release, around February 2019?
- [ ] #1448 Validate Kolibri 0.11.2 -> 0.12.0 on release, around March 2019
- [ ] #1489 Kolibri service is timing out on boot on Ubuntu 18.04 (and possibly other OS's too?)
- [ ] #1531 kolibri-logo.svg isn't loading on http://box/kolibri main page!
- [ ] #1532 Some RPi3 & Ubuntu 18.04 machines visiting http://box/kolibri causes browser to infinitely loop, preventing login
- [ ] #1538 How does a teacher install a ZIM file or other Content Pack from Windows to IIAB? (using a FAT/FAT32 USB stick if poss?)
  - [ ] Record Sneakernet-of-Alexandria video with @m-anish, showing how regular teachers can exchange Content Packs using FAT32 USB sticks (and in future Android phones?)
- [ ] Identify best docs/videos showing teachers how to import THEIR own content, PDF's etc into Kolibri!
- [ ] Understand why Kolibri on RPi can be slow and/or use a high amount of RAM?

tstauch, 2019-06-17:
Kolibri login page continuously reloads without user interaction.

Completely new fresh install on both Nuc (Ubuntu 18.0.4) and RPi (2019-04-08-raspbian-stretch-lite), 

iiab: Full System (3).

On both environments, opening Kolibri from the main menu (/kolibri) results in this indefinite page reload.

I managed to stall the refresh and enter credentials (guessed Admin/changeme) and clicked Login. I then momentarily saw a black bar message saying "You were automatically logged out due to inactivity".

Console log shows some errors:

http://192.168.0.34/kolibri/api/auth/facilityusername/?facility&search=Admin&1560796325304=1560796325304 
Response: 403 Forbidden

holta, 2019-06-17:
> Kolibri login page continuously reloads without user interaction.
>
> Completely new fresh install on both Nuc (Ubuntu 18.0.4) and RPi (2019-04-08-raspbian-stretch-lite),
>
> iiab: Full System (3).
>
> On [...]

Thank you @tstauch !

1) Can you run confirm you're running Kolibri 0.12.3 by running the following?
```
/usr/local/kolibri/bin/kolibri --version
```
2) Can you read thru known issue #1532 to see if we can finally crack this in cooperation with Kolibri's upstream staff @rtibbles @jredrejo and @benjaoming ?

rtibbles, 2019-06-17:
The 403 looks indicative of a CSRF issue still, with the CSRF token from the cookie still being set to something from a different service.

holta, 2020-11-13:
FYI Kolibri 0.14.4 is supposed to be released today:

- https://github.com/learningequality/kolibri/releases
- https://github.com/learningequality/kolibri/milestone/60
- https://github.com/learningequality/kolibri/commits/release-v0.14.x

_(Please try it if they do release!)_

holta, 2020-11-19:
Kolibri 0.14.4's release process has begun...
https://github.com/learningequality/kolibri/releases/tag/v0.14.4

Changelog excerpt: _"File downloads now run concurrently, taking better advantage of a device's bandwidth and reducing the time needed to import resources from Kolibri Studio or other content sources"_

.deb files (and pip .pex files, and other installation packages) can take a few days to appear here: (but let's try it when that happens!)
https://learningequality.org/download/

PS .deb files etc should also appear under "Assets" at https://github.com/learningequality/kolibri/releases/tag/v0.14.4 in coming days.

jvonau, 2021-08-29:
#2739