iiab/iiab #1546: Podcasting platform for IIAB

m-anish, 2019-03-05:
I am interested to explore whether it is possible to have a nicely done open source podcasting platform on the IIAB. It should ideally:
1. Come with some sort of app/web-frontend that lets users search, discover, subscribe, listen
2. Allow podcasters to create and publish the podcast on the app/web-frontend itself

m-anish, 2019-03-05:
[This](http://www.podcastgenerator.net/) looks interesting.

holta, 2019-03-05:
@mikkokotila should we align IIAB with any of the emerging "Offline YouTube" platforms like [PeerTube](https://en.wikipedia.org/wiki/PeerTube), and if so which one?

Some of these appear to be gutters for viral hate speech:
https://www.polygon.com/2018/3/7/17087668/steemit-dtube-bitchute-youtube-purge

Likewise Mastodon is used for "legalized child pornography" in Japan...which is (in fact!) the very last thing IIAB schools want to be involved with:
http://www.ethanzuckerman.com/blog/2017/08/18/mastodon-is-big-in-japan-the-reason-why-is-uncomfortable/

Refs: #1530 #1538 #1540 #1541 #1542

m-anish, 2019-03-05:
Another open source solution - [Podcast Server](http://davinkevin.github.io/Podcast-Server/)

holta, 2019-03-05:
@ericnitschke recommends YouPHPTube which mostly just to "offline" i.e. rehost/stream YouTube videos — but that's a start:

https://github.com/DanielnetoDotCom/YouPHPTube
https://www.youphptube.com
https://demo.youphptube.com
https://youphptube.readthedocs.io

ericnitschke, 2019-03-06:
> @ericnitschke recommends YouPHPTube which is mostly just to "offline" i.e. rehost/stream YouTube videos — but that's a start:

Users can upload videos/podcasts, and record their own right within the service.  We use it during workshops because participants can DL others' videos immediately to their phones.

Not a perfect solution but it's working well so far. Will keep you all updated.

m-anish, 2019-03-07:
YouPHPtube looks interesting. As a test, I was able to broadcast a livestream on OBS, and watch it at the same time in a youphptube browser window (albiet with some lags, and the buffering dialog popping up in between sometimes). But promising that it "just worked"

Initial thoughts:
1. It might be slightly bandwidth heavy, but it remains to be seen
2. It might be compute resource heavy, it seems to be loaded with features. Also remains to be seen if it can be tweaked.
3. I havent tested the install process, so also that remains to be seen how much of a pain that is.
4. I havent seen if it can just do audio streaming. That would be nice to have. 
5. There seem to be some features to play around with advertising, and I see a dashboard with dollar signs and account balances which isnt relevant to us. Perhaps it can be turned off, but that remains to be seen. 
6. All in all, worth exploring deeper!

m-anish, 2019-03-07:
[Related issue](https://github.com/eka-foundation/home/issues/31)

m-anish, 2019-03-07:
[This](https://medium.com/@yenthanh/setup-a-rtmp-livestream-server-in-15-minutes-with-srs-1b0046c77267) might be an interesting options to explore too

m-anish, 2019-03-17:
Made [some progress](https://github.com/eka-foundation/home/issues/31#issuecomment-473524562) on this.

mikkokotila, 2019-03-17:
I think that this is identical to [#31 in EKA](https://github.com/iiab/iiab/issues/1546#issuecomment-473618943) in the sense that there is no difference between "podcasting" and "live streaming" from the standpoint of being able to broadcast one-to-many and having a recording for past broadcasts.

holta, 2019-03-17:
This has nothing to do with IIAB or India per se, but we should be aware of movements like the white supremacy movement which uses viral decentralized video to cultivate hatred:

> Facebook removed 1.5 million videos of the Christchurch attacks within 24 hours — and there were still many more.
https://www.washingtonpost.com/world/facebook-removed-15-million-videos-of-the-christchurch-attacks-within-24-hours--and-there-were-still-many-more/2019/03/17/fe3124b2-4898-11e9-b871-978e5c757325_story.html

mikkokotila, 2019-03-17:
That's right. Right has definitely been a frontrunner in leveraging digital technology in the recent years. 

Political extremism is part of the sub-text in [this](https://github.com/eka-foundation/home/issues/30) issue where I think you were cc'd as well. I think licensing gets a disproportionate share of attention in OSS given how critically important the topic is. I think it can be "radically" rethought in a way that makes more sense for the future of the world. The first challenge would be to address "unwanted use". Shall we continue this on the thread?

holta, 2019-03-17:
> Shall we continue this on the thread?

I do not understand licensing and have a lot to learn if this is related, if you can recommend any thread yes, ideally supplemented by any book(s) folks can recommend on these topics?  Certainly "memes" drive humanity (dangerous & healthy memes, viral online & viral offline memes) and we should try to understand journalistic/sociological realities as best we can.

Interesting that [Discourse.org](https://www.discourse.org/) uses "Civilized Discussion" as its slogan, as if software alone can achieve civility.  Which it cannot...but nor can we get there without software at this point (!)  e.g. I do think IIAB and related communities should investigate Discourse's imperfect-but-greatly-maturing community support/team toolset and what associated "ethics" can nurture purposeful solidarity ethos...in a federated/global, listening-more-than-preaching way.

Journalist: What do you think of Western civilization?
Gandhi: I think it would be a good idea. [unconfirmed that he actually said this]

holta, 2019-03-18:
FWIW another data point:

> YouTube executive offers behind-the-scenes look at how site tried to block grisly video of New Zealand massacre
>
> New videos were uploaded to the streaming site as quickly as one per second in the hours after the attacks on two mosques, YouTube’s chief product officer, Neal Mohan, told The Post.
>
> Inside YouTube’s struggles to shut down video of the New Zealand shooting — and the humans who outsmarted its systems
https://www.washingtonpost.com/technology/2019/03/18/inside-youtubes-struggles-shut-down-video-new-zealand-shooting-humans-who-outsmarted-its-systems/

mikkokotila, 2019-03-18:
@holta this current system is designed to work just like this. There is no way where Youtube keeps creators happy while still effectively blocks "unwanted" content. Whenever something goes wrong, Google sends their pr team to cozy advertiser associations, like ISBA, ANA, or WFA behind closed doors, and sometimes (rarely) makes a public statement as a desperate notion of things having gone out of hand worse than a simple half-day cozying session would mitigate. I know this game all too well.

mikkokotila, 2019-03-18:
Related with the previous point/s, I think there are a couple of interesting aspects here. On one hand there is the "god mode" question, to which extent it could/should be exercised if any. On the network side this might seem more clear, but on the content side it's a big topic. A really hot topic for a good reason. 

Then the other is system-regulation. So basically technology where morals and intended use is inseparable. An unfortunate example would be an oppressive regime providing network access to its population, but using it to oppress people by limiting use and otherwise. A more fortunate example is a machine intelligence algorithm that had the means to optimize against morality of outcomes. We have lot of examples of poor implementations of the system-regulation, but not much good ones. I think it is a critical realm of research and development. But here are pretty important and obvious questions that can't be escaped. How does the system know when to regulate? It is based on the rules that people introduce to those systems and then monitoring the indicators related with those rules. So people create simple rules, which often end up in machines that somehow regulate or influence other people. Because we don't have much in terms of success stories for the human-regulated approach, the implication seems to be that we don't really know how to do that. We don't basically know the rules. Assuming that there is an unsupervised approach (where we don't need to input rules and ground truths), then we end up still with the question of monitoring. How much monitoring is needed in order to reach a given moderation goal?

mikkokotila, 2019-03-18:
So ok, so that this would not become just depressing, there is the concept of "zero cost identity". Most of these problems can be reduced to "identity cost". How much does it cost to me to take a certain action? If I post this article now, what is the price I pay if it's clickbait spam? How about if it's angry towards some group of users or some individual? How if it's just counter productive? There is a price point where nobody will do nothing because it's too expensive, and there is a price point where everyone will do whatever because there is no cost...somewhere in between, there are areas where...well, things could work. This theory explains clearly why in a small local community online extensions of those communities might work really well. The cost is inbuilt as everyone is known in the community outside of the online extension.

holta, 2019-03-18:
Scary Important Stuff, thanks @mikkokotila

And thanks all for pushing forward community/privacy models that can be very meaningful in the "kinda offline" developing world especially.  Quite relevant is this discussion @m-anish kicked off 2.5 weeks ago:

Offline Networking, Offline Chat, Offline Filesharing (Internet-in-a-Box, Android)
http://lists.laptop.org/pipermail/server-devel/2019-March/008303.html

holta, 2019-03-20:
Excerpts from Charlie Warzel's "We’re Asking the Wrong Questions of YouTube and Facebook After New Zealand: It’s time for a real conversation about the infrastructure and incentives that Big Tech provides far-right extremists."

> if platform companies are going to provide the broadcast tools for sharing hateful ideologies, they are going to share the blame for normalizing them...

> On one hand, the stats could be interpreted as a rare bit of transparency on behalf of the companies...

> Most troubling, it’s also a bid to reframe the conversation toward content moderation rather than addressing the role the platforms play in fostering and emboldening online extremism...

> The horror of the New Zealand massacre should be a wake-up call for Big Tech and an occasion to interrogate the architecture of social networks that incentivize and reward the creation of extremist communities and content.  Focusing only on moderation means that Facebook, YouTube and other platforms, such as Reddit, don’t have to answer for the ways in which their platforms are meticulously engineered to encourage the creation of incendiary content, rewarding it with eyeballs, likes and, in some cases, ad dollars. Or how that reward system creates a feedback loop that slowly pushes unsuspecting users further down a rabbit hole toward extremist ideas and communities...

> Stewards of our broken online ecosystem need to accept responsibility — not just for moderating the content but for the cultures and behaviors they can foster. Accepting that responsibility will require a series of hard conversations on behalf of the tech industry’s most powerful companies. It’ll involve big questions about the morality of the business models that turned these start-ups into money-printing behemoths. And even tougher questions about whether connectivity at scale is a universal good...

https://www.nytimes.com/2019/03/19/opinion/facebook-youtube-new-zealand.html

mikkokotila, 2019-03-20:
I understand these articles seem to have some weight, but this same thing has been playing out all along. For example, ISIS demonstrably used these same platforms to recruit attacker and instigate attacks. That's years ago, but no substantial changes have been made. Well ok, back then it was really easy to get an AK-47 from local black market through Facebook, I think some sharp edges like that might have been smoothen in the past few years

Generally speaking commentaries on this topic fall into one of many pitfalls, such as here saying:

> Stewards of our broken online ecosystem need to accept responsibility

The platforms are not stewards, nor will they be stewards, because it is against their own interest.  A global corporate will never care of a community like a community will care for a community. It's not even their responsibility, a person must be deeply misguided to think it is. Their responsibility is to their shareholders, and that is actually a written law. It is incredibly rare to hear anyone suggest the actual solution, which is personal change. Don't like what Youtube is doing for the world? Stop going to youtube. Don't like what advertising dollars are doing the internet, stop supporting it, stop doing anything that generates ad revenue for online services. Don't like what the internet turned out to be? Well, [BEEP] it. Go and create a meaningful alternative for it. 

But people like to point fingers away from themselves.

This narrative has a lot do with why when I first met Anish and he setup school server and mesh in my house, I said within 30 minutes, I'm going to have to become part of this somehow.

holta, 2019-03-20:
@Mikko plz see this somewhat related ongoing work to bring some of the Internet Archive's "Decentralized Web" infra into IIAB: #1576

mitra42, 2019-03-20:
Thanks for the ref, and I took a look at the thread ... from a policy perspective decentralization makes it harder - both for 'good' censorship (hate speech etc) and for 'bad' censorship (e.g. political supression, human rights).

At the basic level - what I'm trying to do is make it possible to share content that is in Internet Archive collections (which anyone can upload to) via IIAB, so that a progam can supplement the KA and OER material with books, video, audio, playable through an interface that is not customized to the content, but is easier to use than browsing directories. 

On the decentralized level the server can support several of the decentralized platforms (IPFS, WEBTORRENT, GUN at present). Most of that support comes for free since the libraries we are using were developed initially for the Archive's decentralized version (dweb.archive.org). In particular we are hoping that seeding to webtorrent on the LAN can reduce load on the RPI (since the browsers can use webtorrent to share with each other) and that webtorrent/ipfs/gun on the WAN can help the system work behind censorwalls where the box may not have direct access to the Archive's servers (we are blocked in many repressive regimes and usually not for the content that IIAB users would be looking for).

holta, 2019-03-27:
"Coming [June 18-21]: The First DWeb Camp"
http://blog.archive.org/2019/03/24/coming-this-summer-the-first-dweb-camp/

mitra42, 2019-03-27:
Yes - and I'm hoping to have IIAB - with the Archive server integrated - on at least one RPi on the meshnet there !

ericnitschke, 2019-03-27:
> "Coming [June 18-21]: The First DWeb Camp"
> http://blog.archive.org/2019/03/24/coming-this-summer-the-first-dweb-camp/

Looks like a great event.  Wish I could make it. 

Will any of you be at [BattleMesh ](https://battlemesh.org/BattleMeshV12)this year?

m-anish, 2019-03-27:
@mitra42 you might also be interested in some progress we're making on having decentralized p2p video streaming over mesh. https://github.com/eka-foundation/home/issues/31

Would love to hear your thoughts on it.

mitra42, 2019-03-27:
Comments posted there.

holta, 2019-05-28:
@m-anish can you explain how https://www.azuracast.com = https://github.com/AzuraCast/AzuraCast relates, i.e. what you're building for North India and/or IIAB?

holta, 2019-05-30:
During this year's North India deployments, @m-anish will try to deploy:
1) YouPHPTube "comes with its own Ansible playbook that runs on Debian 9" (he'll try to make it work on Ubuntu 18.04)
2) A Docker containing (audio-only) [AzuraCast](https://www.azuracast.com/)

holta, 2019-06-13:
See @m-anish's

#1733 Initial checkin of AzuraCast role [for online radio stations]

holta, 2019-06-14:
@m-anish please check out "simple and inexpensive video streaming" that @tim-moody found at https://github.com/jaffamd/streambox just in case there's overlap with your approach?

m-anish, 2019-07-19:
This might be an interesting options as well - https://github.com/arslancb/clipbucket

holta, 2019-11-20:
@m-anish I forgot to mention when we talked yesterday that basically every field operation I talk to is looking for some kind of "offline YouTube" (community dropbox for DIY vids + important local community vids + transient fun/viral vids + of borrowed films of various kind...with basic curation so Local Hero / sys-operator can highlight the best vids...and yes remove overly salacious/hateful/illegal stuff when these inev arise later!)

Now that basically every single country/deployment group has asked us for some version of this (even "Offline TikTok" has been mentioned, i.e. basic remixing/karaoke-like tools if later possible, above and beyond Khan Academy + TED Talk + kids science shows like "Crash Course" + "C'est Pas Sorcier" + "Beakman's World" potentially in Spanish) we should attempt to consolidaye and winnow down the best of these ideas...

i.e. There's a certain urgency to now consolidate all these grassroots mediamaking / grassroots media library design recommendations/desires bringing sense to PeerTube + YouPHPTube + Cham (etc) inspirations...possibly also including pragmatic photo-sharing along the lines of "Offline Flickr" or "Offline Google Photos" using the best plugins for Nextcloud or WordPress?  Certainly with the flood of Android smartphones' vids/photos being an unavoidable design question (!)

Conclusion: several of us should try to winnow down community requirements to those ideas shared by the most number of schools/countries around the world, so that you and Mitra Ardron (and any/all others work closest with the world's "Offline 4 Billion") can Actively Defend/Deploy the most practical community workflows!

mitra42, 2019-11-20:
@holta - I've been hearing something slightly different - which is the need for community generated content to be uploadable, curatable, and selectively uploadable (for sharing with other communities).  There is probably a lot of overlap with what you've written above though.

holta, 2019-11-20:
> @holta - I've been hearing something slightly different - which is the need for community generated content to be uploadable, curatable, and selectively uploadable (for sharing with other communities). There is probably a lot of overlap with what you've written above though.

Yes!

BOTH are needed...
- "vertical" upstreaming/publishing of the emergeant / generative / peer-produced / grassroots / local storytelling vids...
- lateral re-distribution of popular/important vids (e.g. "Sneakernet-of-Alexandria") no matter their source...

(Some communities emphasize one distribution model more than the other...and [cultural appropriation](https://en.wikipedia.org › wiki › Cultural_appropriation) questions of course enter the picture...but both are certainly needed in most cases!)

mitra42, 2019-11-20:
Yes - the "cultural appropriation: question was why I said "selectively uploadable"

holta, 2022-08-16:
Just FYI good progress appears possible now 3 years later &mdash; e.g. to stream MP3 files and similar &mdash; thanks to [AzuraCast improvements](https://github.com/AzuraCast/AzuraCast/blob/main/CHANGELOG.md)[1] that @jvonau has looked into.

If you're interested, please see these 2 tickets:

- PR #2946
- #2991

[1] Example: AzuraCast consolidated its ~7 Docker containers to about ~4 containers &mdash; and has now further consolidated down to 1 single Docker container.  This should streamline + simplify ongoing maintenance and improve security too.

[2] See also:

- https://docs.azuracast.com
- https://docs.azuracast.com/en/getting-started/updates

jvonau, 2022-08-30:
https://docs.azuracast.com/en/getting-started/installation/post-installation-steps
https://docs.azuracast.com/en/user-guide/streaming-software

holta, 2022-08-31:
FYI we might not be 100% there yet &mdash; as broad community testing is now beginning (https://twitter.com/internet_in_box/status/1564986581664014342) &mdash; but there's good progress with these 3 PR's merged earlier today:

- PR #2946 
- PR #3353 
- PR #3355

jvonau, 2022-08-31:
["Internet-in-a-Box allows kids to "instantly" run their own radio newscasts, and music programs:"](https://twitter.com/internet_in_box/status/1564986581664014342) is very misleading.... "own ""internet"" radio newscasts" otherwise it sounds like an over-the-air broadcast ability which might be regulated...

[image]

holta, 2022-08-31:
Those who use AzuraCast in North India always call it "radio."

But for those who are not used to that lingo, perhaps we should call it "web radio" more generally, as outlined here:

https://github.com/iiab/iiab/tree/master/roles/azuracast#readme