iiab/iiab #1555: calibre-web play does not install ImageMagick

tim-moody, 2019-03-08:
On both a UB 18.04.2 VM and rpi the cal-web log says ImageMagick not installed when I import a pdf

tim-moody, 2019-03-08:
on rpi import just hangs with no message
on VM finishes without alerting to the error (probably due to policy mentioned in janeczku/calibre-web#827)

tim-moody, 2019-03-08:
actually the VM above was a NUC

tim-moody, 2019-03-08:
Seems the play also needs to patch the imagemagick policy.xml file

/etc/ImageMagick-6/policy.xml and change

`<policy domain="coder" rights="none" pattern="PDF" />
`
to

`<policy domain="coder" rights="read" pattern="PDF" />`

tim-moody, 2019-03-08:
but these lines do not exist on rpi

holta, 2019-03-12:
@tim-moody any chance you can spin up a PR?

_(Even if a solution for RPi is regrettably not yet obvious!)_

holta, 2019-03-13:
Excerpt from Calibre-Web's FAQ:
_(that we should feel free to further revise)_

> What to do if cover pictures are not extracted from PDF files
>
> Symptom: Calibre-Web logs a message like: `WARNING - PDF extraction forbidden by ImageMagick policy: not authorized '/tmp/calibre_web/xxxxxxxxxxxxxxxxxx' @ error/constitute.c/ReadImage/412` during upload of PDF files. ImageMagick recently changed their security policy, they now deny per default PDF operations, so covers can't be extracted anymore, you have to allow it in the policy.xml file of ImageMagick (located in /etc/Imagemagick-6) by changing line:
`<policy domain="coder" rights="none" pattern="PDF" />`
to
`<policy domain="coder" rights="read" pattern="PDF" />`
>
> Calibre-Web has to be restarted after changing the policy.xml file. Covers should be extracted without error from uploaded PDF files.

https://github.com/janeczku/calibre-web/wiki/FAQ#what-to-do-if-cover-pictures-are-not-extracted-from-pdf-files

holta, 2019-03-13:
I'm assuming this is now fixed, by PR #1565

holta, 2019-03-13:
PR #1567 looks good.

Please continue testing that thumbnail preview / cover pages appear for PDF's at http://box/books on Raspbian, Debian 9/10 and Ubuntu.

closed 2019-03-13 by holta