iiab/iiab #1556: Gitea supporting materials

aidan-fitz, 2019-03-08:
We need to add a few materials to ensure that Gitea is usable and useful:

- [ ] Documentation
  - [ ] Gitea documentation
    - This can be built from the Gitea repository using [Hugo] and then rendered statically.
  - [ ] IIAB-specific documentation
    - Anything that a sysadmin should know about installing and configuring Gitea.
- [ ] Optional [customizations]
  - Note: All languages are included with the Gitea binary by default. Please contribute translations upstream at [Crowdin] if you can!
- [ ] Developer tools!
  - [ ] Text editors - see [this collection]
  - [ ] SDKs for common languages, e.g. Python, Node.js, Processing
- [ ] Computer science and software development courses

[customizations]: https://docs.gitea.io/en-us/customizing-gitea/
[Hugo]: https://gohugo.io/
[this collection]: https://github.com/collections/text-editors
[Crowdin]: https://crowdin.com/project/gitea

holta, 2019-03-08:
Ref PRs: #1242 #1553 #1554

aidan-fitz, 2019-03-09:
Some possibilities for cloud text editors:

- [code-server] - VS Code on a web server
  - Not sure about the system requirements. It runs in Docker; we could easily translate the [Dockerfile] into an Ansible role, but code-server expects a container so that developers can run their code in an Ubuntu-like environment.
  - Also very new and might not be stable.
- [Theia] - cloud and desktop IDE written in TypeScript
- [Eclipse Che] - Java-based cloud IDE
  - Have heard it is bloated

[code-server]: https://github.com/codercom/code-server
[Dockerfile]: https://github.com/codercom/code-server/blob/master/Dockerfile
[Theia]: https://github.com/theia-ide/theia
[Eclipse Che]: https://github.com/eclipse/che

aidan-fitz, 2019-03-09:
[CRI-O](https://cri-o.io/) is a lightweight container runtime (similar to Docker) that supports the [Open Container Initiative](https://www.opencontainers.org/) Image Specification. We might be able to use it as a drop-in for Docker with apps that need to be containerized.

tim-moody, 2019-03-17:
I would like to see a series of syntax references like
https://www.w3schools.com/tags/tag_div.asp
for

- html
-  javascript
- css
- python
- as many others as are practical

holta, 2019-03-17:
If nec @darkenvy & @floydianslips can help us compare:

#1550 "Learn UI Design: The Complete Online Video Course" by Erik Kennedy. Can IIAB include this or similar?"

aidan-fitz, 2019-03-17:
@tim-moody Good idea! [MDN] is available under CC-BY-SA 2.5 with code samples released under CC0. Maybe we can ask for a ZIM version of it.

We can also get the Python reference ([Python 2] and [Python 3]), Java documentation ([JDK 8] and [JDK 9]), and [PHP manual][PHP]. Note that Python 2 will be deprecated on January 1, 2020. Also note that Oracle's Java documentation is subject to [this license][Java license].

[MDN]: https://developer.mozilla.org/en-US/
[JDK 8]: https://docs.oracle.com/javase/8/docs/
[JDK 9]: https://docs.oracle.com/javase/9/
[Java license]: https://www.oracle.com/technetwork/java/redist-137594.html
[Python 2]: https://docs.python.org/2/
[Python 3]: https://docs.python.org/3/
[PHP]: http://php.net/manual/en/index.php

holta, 2019-03-17:
The software profession is changing now that "Google" is replacing reference manuals in many cases.  

Still: ~7 billion people do not have reliable/affordable Internet (of which ~4 billion do not have any Internet at all).

So their needs _should_ be addressed with a truly polished "OER" (open educational resource) drawing from the very best the planet has to offer &mdash; across CS/IT classes from any/all countries willing to contribute here.

People like @deldesir in Haiti and @m-anish in India can greatly inform this design dilemma &mdash; and help us uncover prior work above & beyond Khan Academy's video that IIAB already includes: https://www.khanacademy.org/computing/computer-science

aidan-fitz, 2019-03-17:
Git documentation will be essential. The Git website is [open source][git about] with portions available under MIT, GPL, and CC-BY-NC-SA.

Notice how [GitHub's help website][GitHub help] covers both Git and GitHub; we should consider creating a full help website for Gitea (possibly as a ZIM).

[git about]: https://git-scm.com/site
[GitHub help]: https://help.github.com/en

holta, 2019-03-17:
> uncover prior work above & beyond Khan Academy

I'm sure I'm missing one or two (?) but FYI Internet-in-a-Box currently offers these 10+ resources to teach and learn software/hardware etc:

1) Khan Academy CS: https://www.khanacademy.org/computing/computer-science
2) Games that teach basic coding skills: http://oer2go.org/viewmod/en-blockly-games
3) MIT CS Course from RACHEL: http://oer2go.org/viewmod/en-rachelcourses
4) Edison Robotics: http://oer2go.org/viewmod/en-edison
5) Basic IT skills: http://oer2go.org/viewmod/en-GCF2015
6) Sugarizer that includes a Scratch activity where students can save their work: http://sugarizer.org
7) Scratch: http://oer2go.org/viewmod/en-scratch
8) OpenStax / intro computing, electrical engineer, object-oriented Java, modular/structured C++, XML: http://oer2go.org/viewmod/en-openstax
9) Rasp Pi User Guide: http://oer2go.org/viewmod/en-rpi_guide
10) WikiHow offers almost 300 CS & electronics cartoons/comic strips via ZIM file or RACHEL module: https://www.wikihow.com/Category:Computers-and-Electronics
11) Here are 85 Stack Exchange ZIM files from http://download.kiwix.org/zim/stack_exchange/ that relate to computer science & software engineering:
    - 3dprinting.stackexchange.com_en_all_2019-01.zim (118M)
    - ai.stackexchange.com_en_all_2019-01.zim (146M)
    - android.stackexchange.com_en_all_2019-01.zim (1.3G)
    - apple.stackexchange.com_en_all_2019-01.zim (2.4G)
    - arduino.stackexchange.com_en_all_2019-01.zim (452M)
    - askubuntu.com_en_all_2019-01.zim (5,6G)
    - augur.stackexchange.com_en_all_2019-01.zim (9.5M, prediction market built on Ethereum)
    - avp.stackexchange.com_en_all_2019-01.zim (176M, audio & video production)
    - bioinformatics.stackexchange.com_en_all_2019-01.zim (35M)
    - bitcoin.stackexchange.com_en_all_2019-01.zim (417M)
    - blender.stackexchange.com_en_all_2019-01.zim (9.1G, Blender for 3D graphics, animations, games)
    - civicrm.stackexchange.com_en_all_2019-01.zim (94M)
    - codereview.stackexchange.com_en_all_2019-01.zim (1.2G)
    - cogsci.stackexchange.com_en_all_2019-01.zim (140M)
    - computergraphics.stackexchange.com_en_all_2019-01.zim (124M)
    - craftcms.stackexchange.com_en_all_2019-01.zim (84M)
    - crypto.stackexchange.com_en_all_2019-01.zim (324M)
    - cs.stackexchange.com_en_all_2019-01.zim (644M)
    - cseducators.stackexchange.com_en_all_2019-01.zim (48M)
    - cstheory.stackexchange.com_en_all_2019-01.zim (179M)
    - datascience.stackexchange.com_en_all_2019-01.zim (418M)
    - dba.stackexchange.com_en_all_2019-01.zim (1.1G)
    - devops.stackexchange.com_en_all_2019-01.zim (75M)
    - drupal.stackexchange.com_en_all_2019-01.zim (699M)
    - dsp.stackexchange.com_en_all_2019-01.zim (474M)
    - ebooks.stackexchange.com_en_all_2019-01.zim (58M)
    - electronics.stackexchange.com_en_all_2019-01.zim (3.2G)
    - elementaryos.stackexchange.com_en_all_2019-01.zim (163M, user-friendly Linux distribution)
    - ell.stackexchange.com_en_all_2019-01.zim (613M, English language learners)
    - emacs.stackexchange.com_en_all_2019-01.zim (224M)
    - engineering.stackexchange.com_en_all_2019-01.zim (274M)
    - eosio.stackexchange.com_en_all_2019-01.zim (22M, distributed apps built on blockchain OS EOS.IO)
    - ethereum.stackexchange.com_en_all_2019-01.zim (375M)
    - expressionengine.stackexchange.com_en_all_2019-01.zim (72M, CMS)
    - freelancing.stackexchange.com_en_all_2019-01.zim (100M)
    - gamedev.stackexchange.com_en_all_2019-01.zim (1.7G)
    - gaming.stackexchange.com_en_all_2019-01.zim (1.9G)
    - gis.stackexchange.com_en_all_2019-01.zim (2.3G)
    - graphicdesign.stackexchange.com_en_all_2019-01.zim (2.6G)
    - ham.stackexchange.com_en_all_2019-01.zim (125M)
    - hardwarerecs.stackexchange.com_en_all_2019-01.zim (84M)
    - hsm.stackexchange.com_en_all_2019-01.zim (58M, history of science and mathematics)
    - interpersonal.stackexchange.com_en_all_2019-01.zim (139M)
    - iot.stackexchange.com_en_all_2019-01.zim (56M)
    - iota.stackexchange.com_en_all_2019-01.zim (18M, IoT cryptocurrency that doesn't use a blockchain)
    - joomla.stackexchange.com_en_all_2019-01.zim (75M)
    - magento.stackexchange.com_en_all_2019-02.zim (794M, open-source/PHP e-commerce platform)
    - mathematica.stackexchange.com_en_all_2019-02.zim (2.7G)
    - moderators.stackexchange.com_en_all_2019-02.zim (24M, community managers, admins, mods)
    - monero.stackexchange.com_en_all_2019-02.zim (47M, crypocurrency)
    - networkengineering.stackexchange.com_en_all_2019-02.zim (305M)
    - opendata.stackexchange.com_en_all_2019-02.zim (120M)
    - opensource.stackexchange.com_en_all_2019-02.zim (75M)
    - patents.stackexchange.com_en_all_2019-02.zim (93M)
    - pm.stackexchange.com_en_all_2019-02.zim (144M, project management)
[... 5 lines cut ...]
    - retrocomputing.stackexchange.com_en_all_2019-02.zim (87M)
    - reverseengineering.stackexchange.com_en_all_2019-02.zim (181M)
    - robotics.stackexchange.com_en_all_2019-02.zim (206M)
    - salesforce.stackexchange.com_en_all_2019-02.zim (760M)
    - scicomp.stackexchange.com_en_all_2019-02.zim (170M, solving scientific probs w/ computers)
    - security.stackexchange.com_en_all_2019-02.zim (1.0G, system and network administrators)
    - serverfault.com_en_all_2019-02.zim (2.7G)
    - sharepoint.stackexchange.com_en_all_2019-02.zim (742)
    - sitecore.stackexchange.com_en_all_2019-02.zim (114M, CMS and multichannel marketing software)
[... cut ...]
    - tridion.stackexchange.com_en_all_2019-02.zim (54M, CMS)
    - unix.stackexchange.com_en_all_2019-02.zim(2.1G)
    - ux.stackexchange.com_en_all_2019-02.zim (1.3G)
    - vi.stackexchange.com_en_all_2019-02.zim (183M)
    - webmasters.stackexchange.com_en_all_2019-02.zim (487M)
    - windowsphone.stackexchange.com_en_all_2019-02.zim (105M)
    - wordpress.stackexchange.com_en_all_2018-04.zim (1.0G)
    - workplace.stackexchange.com_en_all_2019-02.zim (434M)
    - writers.stackexchange.com_en_all_2019-02.zim (181M)

mikkokotila, 2019-03-18:
EKA's [Numerical Computing is Fun](https://github.com/eka-foundation/numerical-computing-is-fun). I can say that in part it is definitely inspired IIAB (or back then it was still school-server).  There is an intention to keep publishing new "episodes" regularly so it is also not a static thing. The program will take a person from nothing to "cancer cure AI models" in a way that makes sense for any person, any age. That's the mission.

Previously it went by the name jupyter4kids and it focus on more on numerical (data science) aspect of computation, as well as being presented in the format in which the learning and work is done (notebook). It would be an honor to have it as part of IIAB.

aidan-fitz, 2019-03-18:
Unfortunately, CRI-O is designed for use with Kubernetes (or k3s) and not for standalone use.

deldesir, 2019-03-18:
>Unfortunately, CRI-O is designed for use with Kubernetes (or k3s) and not for standalone use.

... and Kubernetes involve clusters. A containerized environment like k3s/k8s on CRI-O would make this possible. Isolation is already made clear by the simple act of using ansible: 
1- Learning / Teaching resources: _Kiwix, Ka-Lite, Calibre, ..._
2- Other Resources: _Elgg, nextcloud, node-red, bit-torrent, gitea, lokole, Moodle..._

aidan-fitz, 2019-03-19:
@deldesir How does Ansible enforce isolation between apps?

deldesir, 2019-03-19:
I admit I misuse the word ''isolation" here. We're not talking about ansible tower administration. We're discussing kubernetes with CRI-O as container runtime. What I wanted to point out is install of apps inside IIAB are already well orchestrated by using ansible (see the different roles > tasks). So this make it easy to know beforehand which containers/apps would be inside your pods.

holta, 2019-03-24:
Code.org releasing offline lessons in multiple languages

https://medium.com/@codeorg/code-org-announces-unprecedented-global-expansion-c39e946bbb36

Thanks to @mapmeld.  Excerpt:

> Code.org’s CS Fundamentals course, geared toward primary school, will be translated into the 10 most-widely spoken languages of our international user base — Chinese (traditional and simplified), French, Italian, Japanese, Korean, Polish, Portuguese, Spanish and Turkish. Educators will also have access to a new offline version of CS Fundamentals, which will empower schools in low- and no-bandwidth environments to teach computer science to all students.

mapmeld, 2019-03-24:
just seeing this thread now, I'd second @holta's link to Blockly Games as a quick benefit. The apps supports dozens of languages - this year I worked with translators to add Hausa, Igbo, and Yoruba (most common languages of Nigeria, after English). https://blockly-games.appspot.com/maze

holta, 2019-03-24:
FWIW code.org materials are technically illegal in Haiti's private schools, as they exist under a Creative Commons Non Commercial license, and about 90% of Haiti's schools are privately run.

However there is some wiggle room at the bottom here, saying that "a fee to compensate for your own time or overhead fees like venue space or insurance" are Ok:

> Can I charge for using your curriculum at my private school, after-school club, or other organization?
>
> Josh S. March 03, 2019 20:29
>
> All of our materials are free to use under Creative Commons license for non-commercial purposes. As long as you aren't representing our curriculum as your own or trying to sell it, you are welcome to use our materials to teach students so long as there's no cost to the student. If you are charging a fee to compensate for your own time or overhead fees like venue space or insurance, that's fine.
> 
> Please see our Terms of Service for more details.

https://support.code.org/hc/en-us/articles/202518413-Can-I-charge-for-using-your-curriculum-at-my-private-school-after-school-club-or-other-organization-

aidan-fitz, 2019-03-25:
@holta "Non-commercial" in Creative Commons licenses means "not primarily intended for or directed towards commercial advantage or monetary compensation." [[1]](https://wiki.creativecommons.org/wiki/NonCommercial_interpretation)

> The definition is **intent-based** and **intentionally flexible** in recognition of the many possible factual situations and business models that may exist now or develop later. Clear-cut rules exist even though there may be gray areas, and debates have ensued over its interpretation. In practice, the number of actual conflicts between licensors and licensees over its meaning appear to be few.

My dreadfully sleepy non-lawyer opinion:

- Schools can teach classes with NC content as long as they're not charging students excessive fees to take these classes. _This applies to both public and private schools._ Code.org itself states that schools may charge "a fee to compensate for [their] own time or overhead fees like venue space or insurance."
- We can sell IIAB devices with NC content to schools on a cost-plus basis as long as the _primary intent_ is to have a sustainable business model for a charitable mission rather than to exploit a commercial opportunity.

Because of the flexibility in what *non-commercial* can mean, I've chosen to emphasize the words "primarily" and "intended" in its legal definition. However, I bet that it would be easy to secure a commercial use license from Code.org, as they would likely be disappointed if their course could not be taught in *90%* of Haitian schools just because of a legal technicality.

aidan-fitz, 2019-03-25:
Do we have a ticketing system for processing approvals to use "non-free" content, similar to Wikimedia's [OTRS](https://meta.wikimedia.org/wiki/OTRS)?

holta, 2019-03-25:
> Do we have a ticketing system for processing approvals to use "non-free" content, similar to Wikimedia's [OTRS](https://meta.wikimedia.org/wiki/OTRS)?

Not at this time.  A few of us respond to bugs @ iiab . io but very few emails arrive here as we do not prominently advertise this email address.

> Because of the flexibility in what non-commercial can mean, I've chosen to emphasize the words "primarily" and "intended" in its legal definition.

Very interesting, as we seek to enable many education opportunities for the ~4 billion who have no Internet and ~2billion others who can rarely afford Internet.

Similarly, several have tried to persuade Khan Academy to make exceptions to their NC licensing and not gotten very far.

Others claim CC licenses are showing their age and new legal models (may) need to emerge for the coming decade.

Much more needs to be discussed & understood here in coming years, thanks @aidan-fitz

aidan-fitz, 2019-04-02:
#1573 contains all the basic information necessary to install and run Gitea. We can add information about use cases in a future pull request.