iiab/iiab #1557: ./runrole options

aidan-fitz, 2019-03-09:
We should allow low-level access to Ansible's command-line options through `runrole`. Ideally, arguments would be passed through directly to the `ansible-playbook` command.

In particular, I think being able to use tags would be very helpful for redoing parts of the installation process during development (and even during production). For example:
```
./runrole gitea --tags pre-install
```
would pass the `--tags pre-install` arguments through to `ansible-playbook`, causing only tasks with the `pre-install` tag to be run.

m-anish, 2019-03-17:
Agree more generally, that the script may be extended to support such switches. 

Care to do a PR? :)

holta, 2019-12-27:
Ref: 

#2062 Runrole regression [overstrict checking of local_vars.yml]

aidan-fitz, 2019-12-28:
@m-anish Wow, I forgot about this. I'll try to work on this in the coming months.

holta, 2020-01-03:
> @m-anish Wow, I forgot about this. I'll try to work on this in the coming months.

@aidan-fitz let's coordinate with @jvonau who created most of the [./runrole](https://github.com/iiab/iiab/blob/master/runrole) command, ok?

(Likely as part of [IIAB 7.2](https://github.com/iiab/iiab/milestone/7) since [IIAB 7.1](https://github.com/iiab/iiab/milestone/6) will be released in not-so-many-weeks !)

holta, 2021-05-27:
This was essentially taken care of in November, by PR's #2617 #2622 #2623.

(Please re-open or create a new ticket as necessary!)

closed 2021-05-27 by holta