iiab/iiab #1560: Editing content menus not working on Chromium

mir123, 2019-03-12:
This seems like the opposite of #1395 . Editing content menus from the admin console in Chromium (65.0.3325.181) which is the default browser on Raspbian, won't work. Menus do not appear at all,

[image]

Just loading up the page gives

`Uncaught TypeError: window[$(...).attr(...)] is not a function`

Clicking on "Load menu":

`Uncaught ReferenceError: getContentMenuToEdit is not defined`

Choosing a language and clicking on "Show content":

`Uncaught ReferenceError: redrawAllMenuItemList is not defined`

It works ok from Firefox on a remote client.

holta, 2019-03-12:
Thanks @mir123.

Let's try to find someone to do this same test using Chromium browser on Ubuntu or Debian to confirm.

tim-moody, 2019-03-12:
do menu properties have values?
in js does currentJsMenuToEdit exist and does it have a list of menu items in menu_items_1?

tim-moody, 2019-03-12:
have any changes been make to /library/www/html/home/menu.json?
does the home page render when visiting the server?

tim-moody, 2019-03-12:
if firefox works, does it issue any warnings; maybe deprecated syntax?

holta, 2019-03-12:
@mir123 & @tim-moody if possible run `apt install chromium-browser` on Ubuntu/Debian/etc for easier testing?

holta, 2019-03-13:
@tim-moody is making progress here.  Thank you Tim.  It appears this might be solvable in Chromium by working thru this JavaScript incompatibility:

> In Firefox 58, when the exception is unused, the identifier can be omitted, as in the code below. This is also allowed in TypeScript 2.5 and later. However, at the time of this writing (August 2018), the proposal to support this has not yet been included into an ECMAScript standard, so this syntax may not work in other browsers.
>
> ff and chrome support, not chromium

tim-moody, 2019-03-13:
fixed in  iiab/iiab-admin-console#157

holta, 2019-03-13:
@mir123 can you retest after patching Admin Console as follows?

- cd /opt/iiab/iiab-admin-console
- sudo git pull
- sudo ./install

holta, 2019-03-13:
@mir123 plz close this issue if you've hopefully confirmed it's fixed, Thanks!

holta, 2019-03-14:
Works with Raspbian's Chromium version 65.0.3325.181 for me:

Admin Console [http://box.lan/admin] -> Content Menus -> Edit Content Menus -> Content Item List -> [drag-and-drop menu items]

closed 2019-03-14 by holta