iiab/iiab #1562: Don't hard code captive_portal url

m-anish, 2019-03-12:
[This](https://github.com/iiab/iiab/blob/d2b7bb78a776bd21cad88a39eb0e2407f32f737d/roles/captive-portal/templates/capture-wsgi.py#L188) line seems to want to hard code the captive portal url on first glance. Perhaps we should inherit this from a variable. 

I think this might throw up a bug in case somebody chooses a hostname other than box?

holta, 2019-03-12:
@georgejhunt or @m-anish can you propose a PR?

Or ask @tim-moody & @jvonau if you're unsure how?

m-anish, 2019-03-17:
@holta this is potentially a bug rather than an enhancement :)

holta, 2019-12-21:
@georgejhunt can you please update this tkt with the latest?

And close it ideally?

holta, 2020-01-03:
@georgejhunt can you confirm this tkt is now closeable — as a result of just-merged PR #2070 ?

georgejhunt, 2020-01-03:
There is now a variable captiveportal_splash_page initialized to '/'. The IIAB already lets the user redefine whether the '/' goes to /home or to wordpress.
Initialized to some other value than '/', captive portal will go there.

closed 2020-01-03 by georgejhunt

holta, 2020-01-03:
> There is now a variable captiveportal_splash_page initialized to '/'. The IIAB already lets the user redefine whether the '/' goes to /home or to wordpress.
> Initialized to some other value than '/', captive portal will go there.

Thanks.  This needs to be in local_vars.yml and http://FAQ.IIAB.IO — let's not forget prior to IIAB 7.1 release, around 2020-1-30.

holta, 2020-01-03:
> Thanks. This needs to be in local_vars.yml and http://FAQ.IIAB.IO — let's not forget prior to IIAB 7.1 release, around 2020-1-30.

Followup is happening here:

PR #2094 "How does one use vars iiab_home_url & captiveportal_splash_page ?"