iiab/iiab #1564: Add openwebrx - allow people to tune into SDR's connected to IIAB

m-anish, 2019-03-13:
[OpenWebRX](https://sdr.hu/openwebrx) allows users to control and tune into radio stations of their choice. One can buy $20 cheap RTL-SDR's (other more expensive models supported too) and connect them to the IIAB. Users can then listen in and control over a web interface. 

I believe this will open doors to other radio-based apps people can build on top of this.

m-anish, 2019-03-13:
Subset of [this](https://github.com/eka-foundation/home/issues/25)

m-anish, 2019-03-21:
Okay, I've made some progress on this:
1. I am able to install openwebrx + dependencies on my ubuntu 18.04 based linux mint
2. I am able to connect an RTL-SDR dongle to my laptop via its USB port. Model: RTL2832U + R820T2 v3 
3. The normal frequency range I can connect to is 24MHz-1.7GHz
4. This current dongle supports the direct sampling mode, where I can tune between 0-14.4MHz (In theory). 
5. In practice, I get decent reception above 4MHz and into the upper bounds (14.4MHz) in direct sampling mode. This is obviously dependant on the antenna. 
6. Multiple users are able to open a browser and tune within a 2MHz band around some center frequency to their desired station. 

So, 
1. I am going to initially, propose that we can do a "recommended" radio setup, which would focus on being low cost, and easy to setup (but not necessarily the best quality)
2. Then I am going to setup openwebrx as a PR on IIAB, so people can purchase and connect an RTL-SDR to their server.
3. I am going to allow setting the center frequency to be tuned at, so people can tune into their desired frequencies in the range `center +/- 1MHz`
4. Run the server on startup.
5. Profit!

holta, 2019-03-30:
@m-anish in how many countries can kids/communities participate, in your estimation?

e.g. Is there enough interesting material on the air that is findable in most all countries?

e.g. Is this primarily true in cities or also true in very rural parts?

e.g. Is there "interesting" radio material substantially beyond the AM and FM bands that you've been able to find in rural India, that you can mention to inspire others?

m-anish, 2019-03-30:
> @m-anish in how many countries can kids/communities participate, in your estimation?

Pretty much everywhere, I'd wager. I am able to capture radio waves cleanly as far as Kuwait from my location in Dharamsala, a distance of 3000km with the stock antenna that came with the dongle.

> e.g. Is there enough interesting material on the air that is findable in most all countries?

Depends on what one means by interesting. Tricky for me to answer. 

> e.g. Is this primarily true in cities or also true in very rural parts?

FM will be more common in cities. AM (Short-Wave) can go really long distances. 

> e.g. Is there "interesting" radio material substantially beyond the AM and FM bands that you've been able to find in rural India, that you can mention to inspire others?

I am probably not at liberty to discuss what is found outside off of the normal commercial/amateur bands. I havent found much activity (yet) on amateur radio bands but that is also due to my limited knowledge of ham radio. Lots of activity on commercial (usual) short-wave bands.