iiab/iiab #1569: dnsmasq sometimes fails to start on Raspbian (Desktop definitely, possibly also Lite)

holta, 2019-03-14:
Example, that took place several minutes after boot just to be sure:
```
root@box:~# ping mit.edu
ping: mit.edu: Temporary failure in name resolution

root@box:~# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled; vendor preset:
   Active: failed (Result: exit-code) since Thu 2019-03-14 00:10:15 EDT; 2min 17
  Process: 1059 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=
  Process: 821 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCC

Mar 14 00:10:14 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and ca
Mar 14 00:10:15 box.lan dnsmasq[821]: dnsmasq: syntax check OK.
Mar 14 00:10:15 box.lan dnsmasq[1059]: dnsmasq: unknown interface br0
Mar 14 00:10:15 box.lan systemd[1]: dnsmasq.service: Control process exited, cod
Mar 14 00:10:15 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP
Mar 14 00:10:15 box.lan systemd[1]: dnsmasq.service: Unit entered failed state.
Mar 14 00:10:15 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-co
```
@jvonau what log files might help you diagnose this pattern, on 10.8.0.46 = 141-rpi3bp-desk-feb24-mch8-BIG here?

That I have not rebooted in case that helps understand it.  FYI manually running "systemctl start dnsmasq" did work.

holta, 2019-03-14:
I still haven't rebooted &mdash; and this doesn't look good &mdash; but is hopefully revealing:

> root@box:~# systemctl status networking
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-03-14 00:10:13 EDT; 32min ago
     Docs: man:interfaces(5)
  Process: 412 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
  Process: 407 ExecStartPre=/bin/sh -c [ "$CONFIGURE_INTERFACES" != "no" ] && [ -n "$(ifquery --read-environment --list --exclude=lo)" ] && udevadm settle (code=exited,
 Main PID: 412 (code=exited, status=1/FAILURE)
>
> Mar 14 00:10:13 box.lan ifup[412]: Loading netfilter rules...run-parts: executing /usr/share/netfilter-persistent/plugins.d/15-ip4tables start
Mar 14 00:10:13 box.lan ifup[412]: run-parts: executing /usr/share/netfilter-persistent/plugins.d/25-ip6tables start
Mar 14 00:10:13 box.lan ifup[412]: done.
Mar 14 00:10:13 box.lan ifup[412]: device br0 already exists; can't create bridge with the same name
Mar 14 00:10:13 box.lan ifup[412]: run-parts: /etc/network/if-pre-up.d/bridge exited with return code 1
Mar 14 00:10:13 box.lan ifup[412]: ifup: failed to bring up br0
Mar 14 00:10:13 box.lan systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Mar 14 00:10:13 box.lan systemd[1]: Failed to start Raise network interfaces.
Mar 14 00:10:13 box.lan systemd[1]: networking.service: Unit entered failed state.
Mar 14 00:10:13 box.lan systemd[1]: networking.service: Failed with result 'exit-code'.

holta, 2019-03-14:
Seems unrelated to:

#1306 IIAB 6.7/master fails to install on latest Ubuntu 18.04: "Unable to start service dnsmasq"

tim-moody, 2019-03-17:
I experienced this on raspbian lite. For me so far it has only happened during a resizing boot on a large (128G) sdcard. boot time fsck might also affect this.

holta, 2019-05-08:
Is the just-merged dnsmasq fix PR #1636 for Tony Anderson & #1306 possibly be related?

holta, 2019-07-09:
@jvonau can you confirmed this is now solved by your recent dnsmasq PR's ?

closed 2019-07-09 by holta