iiab/iiab #1575: More automated reporting of networking/similar issues

holta, 2019-03-19:
I'm thinking we should automate reporting of networking [and other] bugs so we can be 10X more efficient during the initial phases.  Instead of spending days-upon-days and weeks-upon-weeks gathering context each time (whereupon we lose people often, permanently in many cases...)

Networking issues will never entirely go away, as we've discovered over 7 years now -- nor will other bugs of course -- as there will always Yet Another wacky networking topology thrown at us, Etc -- no matter what we tell implementers we simply cannot control them.

So it would seem wise that IIAB include a script to auto-publish the output of the 5-or-10 key log files & network configuration files, so initial reporting becomes an order of magnitude more efficient going forward.  Non-technical implementers (and there are many) will happily partake if they're leading themselves and all of us forward.  e.g. the best would be if 1 short command tested for Internet and dumped all these files to pastebin/GitHub or similar and also somehow emailed us at the same time.

A more mature version of this "iiab-diagnostics" command (or whatever we call it) could also dump everything into /opt/iiab/diagnostics (or wherever) with crisp instructions for (1) moving it to USB stick (2) publishing it to the web from Windows typically.

holta, 2019-03-19:
If folks can add to this ticket their recommendations of the 10-or-so-most-useful IIAB config & networking files that would be Extremely Useful!

Also: which such files might will be the most privacy-sensitive, and as such perhaps should not be published to GitHub etc?

holta, 2019-03-23:
What IIAB files are most useful for remote support / remote diagnostics?  _DRAFT:_
- /etc/iiab/iiab.env
- /etc/iiab/iiab.ini
- /opt/iiab/iiab/iiab-install.log
- /opt/iiab/iiab/iiab-network.log [if `./iiab-network` has been run]
- /opt/iiab/iiab/iiab-debug.log [if `./runrole <ROLE>` has been run]
- /library/www/html/home/menu.json
- output of `ip a`
- output of `brctl show`
- output of other networking commands?

mikkokotila, 2019-04-11:
I guess one thing that seems useful would be to have a single command (in path preferably) where the user just types something like `$ iiab-diag` which gathers all the various files and meaningful system specs, package it into zip and emails it. 

Then the next step could be to have some automated analysis tools to automate common processes, and finally implement those automations on the system with some prescriptions (e.g. run this command and see what happens). If that data could be collected, maybe an AI could be later trained that would be able to sniff the system and go through a process of trial and error to fix the system for the user.

georgejhunt, 2019-04-15:
- I was searching for something like pastebin for targz files. It looks like filebin works: https://filebin.ca/4dnRGP2bDOOk/raspbian-9-190415-ghunt.17244.tgz. 
- I discovered that vim is able to open tgz files and enable easy browsing of content.
- I was able to download this filebin url via chrome. But wget on macbook got an error 503 "service temporarily unavailable".
- I just tried to drop the tgz into this issue input box. The error message suggested that zip files are supported.
- But I'd rather keep the convenience of having all the information packaged together and browsable via vim, rather than to be forced to create a separate  directory to unzip the separate files into.

holta, 2019-04-15:
Huge thanks @mikkokotila & @georgejhunt for your designs here.

@jvonau has a lot of experience in tech support and may also have valuable insights e.g. things like which things are most indicative of implementer/operator _intentions_ *alongside* the technical issues (:

holta, 2019-06-23:
PRs #1757 & #1763 are now merged &mdash; so `/usr/bin/iiab-diagnostics` is now available in reasonably polished form for new implementers, as documented here:

https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md

Tangentially Related:
#1749 -> PR #1752 OpenVPN is very commonly needed when DNS is failing
#1759 Assess MediaWiki/Discourse/CRM/FB needs to scale up IIAB Community Support
#1762 IIAB Community Analytics above & beyond AWStats

holta, 2019-06-27:
The core is now implemented by @georgejhunt's PRs #1757 and my #1763.

Please re-open as nec, if/where further improvement is nec!

closed 2019-06-27 by holta