iiab/iiab #1576: Installing server onto IIAB

mitra42, 2019-03-19:
I'm trying to install a Internet Archive server so that it can potentially become one of the options available on IIAB. 

This is not just a new content module, so the places to do this are obviously widespread throughout IIAB, and I haven't so far found them collated anywhere.  I have a hand-installed version running and can repeat that process based on the instructions at https://github.com/internetarchive/dweb-mirror/blob/master/README-raspberrypi.md
 
For example ... 
* appearing on the menu of installable options
* Appearing in the control panel for any administration
* Appearing as an option for the user
* Fetching the files from github or npm and setting up the distribution.

I can probably make the appropriate changes, but I'm not going to be able to do this without some help in finding where the changes need to be made, and any other considerations to make it play nicely. 

This is, of course, a work in progress, and will probably change a fair amount once its used in the field for real deployments, what I'm wondering about is whether there is interest in this being part one of the levels of IIAB distribution and if so whether someone can help with this process.

holta, 2019-03-19:
@mitra42 I'm traveling and away for much of the coming 10 days but @tim-moody & others should be able to help you in the interim, also shout if you get stuck!

PS we can also arrange a live call if that's preferable?

tim-moody, 2019-03-20:
IIAB has a number of servers that render content to users via a browser. There are two main models, and I'll describe kiwix and ka-lite as examples of them.

Kiwix runs as a server on a soft coded port, which can also be proxied by apache, and serves content in the form of zim files that are downloaded from Kiwix. The Admin Console has screens which display lists of such zims with descriptions by language and allows the user to check the zim of interest and click a button to cause it to be downloaded and stored where kiwix expects it.

Ka-lite also runs as a server on a port and serves videos and exercise files stored on IIAB. Ka-lite has its own admin user interface to download videos and metadata for storage on IIAB, so the IIAB Admin Console does not handle this task.

We need to understand how the Internet Archive Decentralized Web works. It sounds like it is a server similar to both of these, perhaps serving warc files or I don't know what. Does it have its own admin user interface or would a panel need to be added to the Admin Console. If the latter, is there a catalog somewhere that can drive the selection and download of content?

mitra42, 2019-03-20:
Good questions .... and part of the admin side will have to evolve over time in order to work better with the rest of IIAB. 

The server has three effective modes of operation.   
- as a crawler it can crawl a configurable set of Internet Archive content and download them. (Note these are IA "Items" not the Wayback machine, so mostly videos, audio, and texts).  
- It can serve that locally stored content completely (once a few bugs are fixed) offline using the Internet Archive UI. 
- it can  run as a proxy - storing content locally as someone browses. 

Currently the server runs on its own port (http:....:4244) in theory that's movable though I'd love to keep it consistent. It should be able to be proxied, but this isn't fully tested yet, in which case I'd move its internal port and have it on https:....:4244 

Future (relatively soon) work will be to also have it serve videos via WebTorrent, so users can share each other's bandwidth rather than being limited to the RPi's constraints. That will run on a different port.

It will have its own Admin UI for managing downloads etc, but there might be one or two things that would be better at a level higher up (like updating the server, restarting it etc). Its configured in a .yaml file which has defaults that shouldnt need changing.

Content is stored in its own directory hierarchy, it looks for a top level "archiveorg" directory in a configurable list of places, which could include `/library` or other standard IIAB spot, and it also checks for any mounted USB drives that have `archiveorg` at the top level, it can serve from multiple locations at the same time, and there are also a set of options to the crawler that will make it build a new USB drive from already cached content. 

Installation is currently via `yarn install` which runs an install script on completion. BUT it does need to run a current version of node, that https://github.com/internetarchive/dweb-mirror/blob/master/README-raspberrypi.md has instructions for installing on top of IIAB (and for installing on raw NOOBS) 

If I'm  not answering the important questions let me know, or we can talk - I'm currently (till next Wednesday) in an Francisco. Then mostly offline for 10 days since I'm heading to Valencia for Internet Freedom Festival (where I'm presenting and will be mentioning this).

holta, 2019-03-20:
Refs: #1519 #1520 #1521

tim-moody, 2019-03-20:
Sounds like we don't need much besides an ansible role to start with. We tend to soft code the port, but set it to the author's convention unless there is a conflict. You also need <service>_install and <service>_enabled global ansible variables to control installation. There is a place in networking for logic to include ports if a service is installed and enabled. We use systemd to start and stop services.

We already install node for a couple of roles. Ansible can run scripts, but depending on complexity it might make sense to convert the yarn install to ansible.

Keep in mind that the target IIAB audience is functionally offline, so downloading of content is expected to be a provisioning step and not a regular user activity. But the ability to serve from USB is consistent with our strategy. Also allowing an implementer to browse Internet Archive for content and then have it stored to the server for offline use would be quite important.

mitra42, 2019-03-20:
Agreed on all points ... yes, I'm assuming that in most cases the IIAB is offline, badly online, or intermittently online, our system can also for example run on a laptop to pull content to a USB, then ship the USB to the IIAB. Or download to the SD during provisioning. I'm trying to cover all the likely use cases, but fully offline is a base-case that it has to work for. 

I've seen it, but I've never setup a Ansible script before, though I'm familiar with shell scripts, npm, yarm etc. Making that first time ansible on a system where I don't know what is required is my concern.  For example - we hit the problem doing the manual install where some ports weren't open and needed a manual edit of that file, and I don't know the full list of places that should be edited so that this works automatically.

mitra42, 2019-03-20:
An example ansible role for a similar service would be very useful.

tim-moody, 2019-03-20:
Though it may be more complicated than you need sugarizer is a role that uses nodejs. It also gets its source from git.

https://github.com/iiab/iiab/tree/master/roles/sugarizer

I don't know yarn, but apparently ansible supports it

https://docs.ansible.com/ansible/latest/modules/yarn_module.html

ansible is not really a scripting language. it is a set of declarations, written in yaml, about the desired state of the target, so we don't say install a package, we say the state of a package should be 'present'.

mitra42, 2019-03-28:
@georgejhunt - the link to the non-ansible instructions is there in the original issue. [https://github.com/internetarchive/dweb-mirror/blob/master/README-raspberrypi.md]

georgejhunt, 2019-03-28:
There's an ansible branch on my repo which trys to install dweb. The
ansible play runs to completion. But the server running on port 4244 does
not run under node. When I try to run the server under node by hand, the
error complains about wrtc. I added cmake with apt-get, and wrtc fails to
compile with the error message:

 /opt/iiab/archorg/node_modules/wrtc/src/binding.cc:10:0:

/opt/iiab/archorg/node_modules/wrtc/src/datachannel.h:20:45: fatal
error: webrtc/api/datachannelinterface.h:
No such file or directory

 #include "webrtc/api/datachannelinterface.h"
This is on a rpi3, which probably already had build-essentials -- so not a
clean test.
 I'm out of my depth.

Ansible branch is at
https://github.com/iiab/iiab/compare/master...georgejhunt:archorg?expand=1

mitra42, 2019-03-28:
wrtc has some known problems compiling on certain platforms, I've been getting error messages during install but not during operations because I test for the existence of wrtc .  I'll take a look at your branch.  (I'm on a plane tonight, then at IFF so it might not be till the end of next week depending on when I get a chance and internet connection to work on it. )

holta, 2019-04-04:
> wrtc has some known problems compiling on certain platforms, I've been getting error messages during install but not during operations because I test for the existence of wrtc . I'll take a look at [...]

@mitra42 no rush but let @georgejhunt & @m-anish know if there are next steps they can help with?

mitra42, 2019-04-05:
Sure- I'm at the IFF conference this week so no time to do any code hacking.

mitra42, 2019-04-19:
Hi @georgejhunt I'm back in Australia and catching up on the backlog .. that branch is no longer availabe, did you rename it ?

holta, 2019-04-19:
> Hi @georgejhunt I'm back in Australia and catching up on the backlog .. that branch is no longer available, did you rename it ?

fyi @georgejhunt is heads down working on offlining OpenStreetMap, but hopefully @m-anish might be able to help here too, with any needed Ansible/IIAB questions?

mitra42, 2019-04-19:
Sure - @m-anish , @georgejhunt had a first go at Ansible, posted it, but that link doesnt work anymore (I guess George renamed the branch or something), so until/unless its found we are back to the top ... starting from scratch.

georgejhunt, 2019-04-19:
Yes, renamed.

https://github.com/georgejhunt/iiab/tree/distweb

My apologies

mitra42, 2019-04-22:
@georgejhunt - had a good skype chat with @m-anish and @holta and tried the following based on the skype ... 
```
cd /opt/iiab
sudo mv iiab iiab-release # The old version
sudo git clone 'https://github.com/georgejhunt/iiab' --depth 1
cd /opt/iiab/iiab
sudo ./iiab-install —reinstall
```
It failed with following error.  Notably, there was no distweb specific stuff that I could see before this failure.
```
TASK [iiab-admin : Put an autostart line to check for default password in LXDE (raspbian)] ****************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /home/pi/.config/lxsession/LXDE-pi/autostart does not exist !", "rc": 257}
	to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry
```
I checke that directory and `/home/pi/.config/lxsession/LXDE-pi`exists but only has `desktop.conf` &  `lxde_ssh_warn.sh` in it.

mitra42, 2019-04-22:
Ok - my mistake I need to `sudo git clone 'https://github.com/georgejhunt/iiab' -b distweb --depth 1` but it still fails with :
```
TASK [network : Install clean copy of /usr/bin/iiab-gen-iptables from template] ***************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'archorg_port' is undefined"}
	to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry
```

mitra42, 2019-04-22:
PR created to fix that one ... and retrying

m-anish, 2019-04-22:
FWIW, the playbook fails with the log below on ubuntu/bionic64 VM

```
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["npm", "install", "--allow-root", "--unsafe-perm=true"], "delta": "0:01:03.047264", "end": "2019-04-22 09:36:16.645696", "msg": "non-zero return code", "rc": 1, "start": "2019-04-22 09:35:13.598432", "stderr": "npm WARN deprecated hapi@16.7.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated text-encoding@0.7.0: no longer maintained\nnpm WARN deprecated accept@2.1.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated ammo@2.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated boom@5.2.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated cryptiles@3.1.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated call@4.0.2: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated catbox-memory@2.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated catbox@7.1.5: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated items@2.1.2: This module has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version of hapi to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated joi@11.4.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated heavy@4.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated somever@1.0.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated mimos@3.0.3: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated statehood@5.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nnpm WARN deprecated subtext@5.0.1: This version has [... cut ...] e", "npm ERR! fatal: failed to stat '/root/.npm/_cacache/tmp/git-clone-705e0d26': Permission denied", "npm ERR! ", "", "npm ERR! A complete log of this run can be found in:", "npm ERR!     /root/.npm/_logs/2019-04-22T09_36_16_629Z-debug.log"], "stdout": "", "stdout_lines": []}
	to retry, use: --limit @/opt/iiab/iiab/run-one-role.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=29   changed=12   unreachable=0    failed=1  
```

mitra42, 2019-04-22:
We got past that one by running it with sudo. And past the variable issue. Its now on me to fix two problems stopping it from starting a) its in a different directory than it expects b) executable name and options are changed. Both "me" issues, so I hope to get to both of those this morning  if there aren't too many fires in email.

mitra42, 2019-04-24:
Those problems are fixed ... clean installs not working yet - here are what I believe would be clean install instructions. 
curl -o install.sh d.iiab.io/install.txt
vi install.sh and change repo to mitra42
Ran into some problems SSHing into the fresh box, so it will need to wait till I get home later today for next step.

holta, 2019-04-24:
> curl -o install.sh d.iiab.io/install.txt

Better yet:

curl -o /usr/sbin/iiab https://raw.githubusercontent.com/iiab/iiab-factory/master/iiab

> vi install.sh and change repo to mitra42

vi /usr/sbin/iiab

...changing Line 179 (https://github.com/iiab/iiab-factory/blob/master/iiab#L179) from...

git clone https://github.com/iiab/iiab  --depth 1

...to...

git clone https://github.com/mitra42/iiab -b distweb

mitra42, 2019-04-24:
Updated instructions - with this, it seems to work. 
```
sudo bash
curl -o /usr/sbin/iiab https://raw.githubusercontent.com/iiab/iiab-factory/master/iiab
vi /usr/sbin/iiab
```
...changing Line 179 (https://github.com/iiab/iiab-factory/blob/master/iiab#L179) from...
`git clone https://github.com/iiab/iiab --depth 1`
...to...
`git clone https://github.com/mitra42/iiab -b distweb`
```
chmod +x /usr/sbin/iiab
iiab
```
And the service appears on http://box.local:4244

(http://box:4244 or http://box.lan:4244 is what it says in the docs but seems to be a myth - see #1583

mitra42, 2019-04-24:
Next steps - once someone else confirms this clean install works are I think
[ ] how to link it into the UI that appear 
[ ] revisit proxy, what address(es) to proxy. 
[ ] which level (if any) of distribution should it be included in Small/Medium/Large and what the instructions are for enabling it in the other distributions
And of course a bunch of work i'm still doing on the server itself, but they will all show up with reinstalls/updates.

georgejhunt, 2019-04-25:
Ok, then I'll merge it into my repo, and do a clean install with
internetarchive_(install/enabled) set to true.

mitra42, 2019-04-25:
Thinking about the apache proxying, it looks to me like you are trying to do is make urls within our server turn up on the main server via Apache, at the moment I think that is a mistake, because the server is doing things like pushing history URLs and adding its own server/port to make urls that would go to archive.org stay local. Instead I think it just needs a redirect so that /xxx/yyy goes to http://localhost:4244/yyy

The question is what xxx should be .... some options
`/archiveorg/yyy`  ->.  `localhost:4244/yyy`   BUT then we'd have URLS like `/archiveorg/arc/archive.org/details/foo`   which has unneccessary duplication iit. 
All of the public facing urls start `/arc/` and most start `/arc/archive.org` so it would make sense to me to redirect  
`/arc/yyy` ->` localhost:4244/arc/yyy`

Note the reason for the `/arc/archive.org` pattern is that in future the dweb-mirror server will add support for URLs like /ipfs and /webtorrent because even though most may not work in IIAB they are part of a trend of distributed web urls starting with the name of the service.  Allowing the reverse to also happen (e.g. for IPFS to forward back an url starting /arc). We also have tentative plans for other urls under /arc so I'm trying to setup for future consistency here.

mitra42, 2019-04-25:
Hmmm - I'm not sure this is even possible anyway, it doesn't appear possible in Apache to redirect to another port on the same host, unlike in express (the one I'm using in Node):
RedirectMatch doesn't have access to the %{SERVER_HOST} 
RewriteRule cant do a redirect, 
ProxyPass mucks up any root-relative urls in the code

I see a number of answers in Google/StackOverflow etc, but they all assume the ServerHost is known, and since box.lan seems to be a myth (#1583 ) I can't just use box.lan:4244 and assume it will work.

georgejhunt, 2019-04-25:
If just using "box.lan:4244" would work for you, that may be an option. Ansible assembles "box.lan" from ansible variables {{ iiab_hostname }} and {{ iiab_domain }} -- which in turn is known at install time.

mitra42, 2019-04-25:
The problem is that box.lan:4244 seems to be a myth that I've posted in other issues esp #1583 , none of the RPIs I've installed IIAB on show up at box.lan, they've all shown up at box.local, so redirecting to box.lan:4244 isn't going to work.

georgejhunt, 2019-04-25:
Yes, in any local network, there can be only one box.lan. Network admins
would need to guarantee uniqueness -- which I guess does not help for
discovery.

georgejhunt, 2019-04-25:
Looking in more detail at the "mythical" name resolution problem:

On the clean install just completed, initially I tried <local ip add>:4244,
which appeared to work.
I associated to 'internet in a box', but got a browser not found message.
But then within 30 seconds, I got a IIAB home page.

I have too many IIAB devices with wifi, and can never tell who it is that
is I am talking to. So I changed the SSID of the unit under test to
"archiveorg". Then I associated and went to box.lan, which my mac
immediately rewrote to https://google...box.lan ... and returned a not
found error message..
But when I specifically went to http://box.lan:4244, I got the newly
installed application. So we need to communicate the conditions more
explicitly, comparing how it works for me, and fails for you.

Adam has seen lots of instances where old Android machines need to be
treated differently. But windows machines seem pretty reliable -- have
repeatable results, consistent across hardware clients. (and my mac seems
solid).

mitra42, 2019-04-26:
I only have one of the RPI's on my Lan. I've *never* seen it come up as box.lan, it shows up as box.local on both Mac and iPhone.  Same happened on the lan at the office, and same happens whether I connect via my wiFi router (that the RPI is plugged into) or if I associate to Internet In A Box. 

OK - so the instructions could say try box, box.lan and box.local which sort-of works, BUT won't work for a redirection where you have to know what hostname the client is seeing.

tim-moody, 2019-04-26:
The term 'lan' is ambiguous. It can refer to the network on which IIAB is an appliance in which case name resolution is done by something else and box.lan will often not work. Or it can refer to the lan controlled by IIAB. box.lan usually works there. Your mac and iphone probably support mdns so box.local works in the first case, but other clients don't support that. Any reference to box.lan assumes you are connected to the IIAB hotspot or to the IIAB lan by any means.

georgejhunt, 2019-04-26:
Let's drill down a little.
After you associate with the rpi with your client, does
/etc/resolv.conf on your client indicate that the dns resolver  is
set to "nameserver 172.18.96.1"?
if you ping box.lan, does it resolve to 172.18.96.1?
Do you get a ping response?

mitra42, 2019-04-26:
@tim-moody - even connected to the WiFi "internet in a box" it fails.
@georgejhunt - I'll run thru those steps when I get home.

tim-moody, 2019-04-26:
http://box.lan fails (? returns 404) when the client is connected to the Internet in a Box hotspot?

mitra42, 2019-04-26:
No, http://box.lan fails UNRESOLVED when connected to the Internet in a Box hotspot, every install I've done (with or without my code) and on every device and every network I've tried.  I'll try George's suggestions when I get home to see if we can figure out why.

mitra42, 2019-04-26:
@georgejhunt 
> Let's drill down a little. After you associate with the rpi with your client,
Successfully associated with “Internet in a Box"
> does /etc/resolv.conf on your client indicate that the dns resolver  is set to "nameserver 172.18.96.1”?
IP: 172.18.113.110; gateway: 172.18.96.1 DNS: 8.8.8.8
> if you ping box.lan, does it resolve to 172.18.96.1? Do you get a ping response?
ping box.lan fails (Unknown host); ping box.local works; 

Cleared 8.8.8.8 and NS is now 172.18.96.1 and ping works 

Old iPhone without a SIM
box.lan fails - goes to Google; http://box.lan works

Newer iPhone - with SiM
box.lan fails - goes to Google; http://box.lan also fails, I’m guessing its trying the DNS via the SIM card first; box.local works. 

The 8.8.8.8 DNS is very common, its Googles. I must have switched my phone to it at some point when I was on a WiFi where DNS wasn’t configured correctly in DHCP. 

So looks like if box.lan is to work then instructions need to be clear. 
won’t work if any of following are true
a) connected via a router
b) manually using a DNS (e.g. 8.8.8.8)
c) also have a GSM connection on the phone. 

Of course, this doesn’t solve the redirect problem since in any of these cases, the user may be getting to the RPI via IP address or box.local in which case redirecting to http://box.lan:4244 is going to fail. 

Note every attempt I've done up to now has had at least one, if not all three of these failure-causes happening,

georgejhunt, 2019-04-26:
So a client comes to the rpi on http://box.lan/arc, or http;//box.local/arc.
Could a redirect to http://172.18.96.1:4244 work? Needs to be tried

mitra42, 2019-04-26:
Redirecting to 172.18.96.1:4244 is going to probably work *if* its on the local LAN, but not if the PI is attached to a router (for example that's the recipe I'm seeing proposed on mesh networks). 

I'm not an Apache expert, its evolved, and I've been using nginx (or express in Javascript) which doesnt have this problem,  but I think part of the problem is that mod_rewrite doesnt seem to be working, possibly because there is no <Directory> directive, and from my read of the docs, mod_rewrite (unlike RedirectMatch) has access to the http variables.

mitra42, 2019-05-17:
@holta is something holding up the release at this point ? 

Mostly I'm wondering when to change directions on my main README to tell them to use the main IIAB release rather than my fork.

holta, 2019-05-17:
Sorry @mitra42 nothing's holding it up, I'll work on the 2-line description (that students see) in coming days, so that it hopefully gets merged then.

holta, 2019-05-25:
@mitra42 
- I just merged your PR #1617 after sprucing up several bits of community documentation, both in the PR and in places like
  - http://FAQ.IIAB.IO Item 9 _"What services (IIAB apps) are suggested during installation?"_)
  - the ["What's New?" section of IIAB 7.0's DRAFT release notes](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes#whats-new)

  Which hopefully will makes this more approachable to IIAB implementers in many different corners of the world after IIAB 7.0 is released in June.
- I'll add a [MenuDef](https://github.com/iiab/iiab/wiki/IIAB-Menuing) with the Internet Archive logo later this weekend, so that an appropriate educator-oriented 2-sentence blurb appears on offline IIAB home pages, whenever your Ansible role/playbook is installed.
- This needs more testing on RPi 3 B+ and I'm hoping @mikerjensen and others you know might be able to help us here in coming days, now that this is part of IIAB's mainline http://download.iiab.io installer?
- It looks like there's a problem with Apache's [/etc/apache2/sites-available/internetarchive.conf](https://github.com/iiab/iiab/blob/master/roles/internetarchive/templates/internetarchive.conf) &mdash; http://IP-OR-BOX-NAME:4244 works but http://IP-OR-BOX-NAME/archive does not yet work — can you look into that during your next/current release cycle?
- Feel free to further revise https://github.com/iiab/iiab/tree/master/roles/internetarchive#browsing as we get these URLs tighter[*] and others like myself and @m-anish in India will help!

[*] ideally http://box/archive will become the main/mnemonic URL, as teachers want something very short!

mitra42, 2019-05-25:
Thanks -Adam,  I've been mostly offline for a couple of days, I saw and am just going through the changes you made to the PR prior to merging. 

Most seem to be variations between the sample install that @georgejhunt put together and what you want and I don't have an opinion on those, as my Pull Request was just trying to follow George's patterns. 

Proxypass in apache wont work, the URIs are rewritten on our server which has to be at the top level of whichever port its on.  It will need to be a redirect, the question raised in earlier discussion was what to redirect to, and the problem with Apache  not enabling mod_rewrite, and redirect (unlike on NGINX, or Express). not being able to read what "server" it was accessed from (i.e. whether the user is coming bia localhost or box or box.lan or box.local),  I've tried to figure it out from apache docs, but it needs someone who knows Apache and IIAB better, and can figure out where to look on IIAB's system and how to change it without breaking something else in IIAB.

mitra42, 2019-05-25:
I'm working on a USING.md document - current version is at https://github.com/internetarchive/dweb-mirror/blob/master/USING.md it won't be public until some of the issues noted in it are resolved (some involving significant code in dweb-mirror), but it will also give people more of a guide in particular to managing their offline connections.   My timeline on this is about a month, but it will get incrementally better before then.

mikerjensen, 2019-05-26:
I just ran the standard IIAB curl install command (curl d.iiab.io/install.txt | sudo bash)  on my existing IIAB installation. It completed, but box.local:4244  is not recognised...

holta, 2019-05-26:
> I just ran the standard IIAB curl install command (curl d.iiab.io/install.txt | sudo bash) on my existing IIAB installation. It completed, but box.local:4244 is not recognised... (box.local/home works fine though)

@mikerjensen: the above "IIAB 1-line installer" is really for fresh/clean installs (but great suggestion to possibly have it invoke upgrades in future!)

(1) For now, force an IIAB upgrade by running these 4 commands:

- sudo mv /opt/iiab/iiab /opt/iiab/iiab.old
- sudo mv /opt/iiab/iiab-admin-console /opt/iiab/iiab-admin-console.old
- sudo mv /etc/iiab/iiab.env /etc/iiab/iiab.env.old
- sudo mv /etc/iiab/local_vars.yml /etc/iiab/local_vars.yml.old

(2) And then reinstall & invoke the new /usr/sbin/iiab using IIAB's "1-line installer" as you originally indicated:

- curl d.iiab.io/install.txt | sudo bash

Soon after this begins, it will asks you to run "nano /etc/iiab/local_vars.yml" -- at that point please verify that your /etc/iiab/local_vars.yml is properly customized to your needs, making sure it contains these 2 lines especially:
```
internetarchive_install: True
internetarchive_enabled: True
```
( As seen here: https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L316-L319 )

mikerjensen, 2019-05-28:
Ok, thanks, I tried your instructions, and it seems to mostly work. 
Samba stopped working again, mainly because the iptables rules reverted to the default (i guess i forgot to edit the config file before continuing with the install - any way to do this through the iiab admin interface?).  i also notice that using the iiab admin interface to set a static IP address didn't work (IP address remained assigned by the router) - or should i be just running 'install configured options' again after the install (the address i requested remains in the form field).

Going to box.local:4244 redirects to:
http://box.local:4244/arc/archive.org/details/local?transport=HTTP&mirror=box.local:4244  
but this page sits permanently saying "Resources downloaded to http://box.local:4244
Loading ..." with nothing filling the screen below the header.. see:
[image]

holta, 2019-05-28:
> Samba stopped working, mainly because the iptables rules reverted to the default (i guess i forgot to edit the config file before continuing with the install - any way to do this through the iiab admin interface?).

Not at this time.

> BTW, I seem to need to restart samba (/etc/init.d/samba restart) after the first boot to get samba going.

Samba starts on subsequent reboots presumably?

> i also notice that using the iiab admin interface to set a static IP address didn't work (IP address remained assigned by the router) - or should i be just running 'install configured options' again after the install (the address i requested remains in the form field).

Have you followed the instructions at http://FAQ.IIAB.IO Item 21 ("How do I set a static IP Address ?")  

This may work better than the Admin Console possibly?  (Confirm if you can, Thanks!)

> Going to box.local:4244 redirects to:
> http://box.local:4244/arc/archive.org/details/local?transport=HTTP&mirror=box.local:4244
> but this page sits permanently saying "Resources downloaded to http://box.local:4244
> Loading ..." with nothing filling the screen below the header.. see: [image]

Please ask @mitra42

holta, 2019-05-28:
@mikerjensen if possible can you give us a ring on Skype-or-similar to talk with some of our networking guys, just so we understand the main things you're trying to accomplish?

PS can you post the 3 URL's below to give them more visibility here, after running `apt install pastebinit` ?

- pastebinit < /etc/iiab/iiab.env
- pastebinit < /etc/iiab/iiab.ini
- pastebinit < /etc/iiab/local_vars.yml

mitra42, 2019-05-29:
@mikerjensen That redirect looks good - which means the server is answering. BUT not sure why its not filling the screen.. . 
Two options 
A: it might be an odd test case with a machine first installed on the previous version- in which case clicking the Archive button should take you to the IA home page, on the local machine (that link should appear in the "Resources" page, but possibly its only setup during a fresh install. 
B: or maybe a bug in which csae there should be some error showing up in the Javascript log - can you check. 

This is super helpful, I'm doing rapid iteration on the UI for managing stuff, to get something working smoothly by the Dweb camp, but certainly haven't had opportunity to test in all scenarios yet.

mitra42, 2019-05-29:
@mikerjensen I'm able to duplicate your problem and will work on it. I had a disk crash today, so a bit slowed down.

mitra42, 2019-05-29:
I've posted a new issue at https://github.com/internetarchive/dweb-mirror/issues/144 since its not IIAB specific and will work on it there and report back.

mikerjensen, 2019-05-29:
Thanks,  yes, the instructions from the FAQ for setting the static IP work, while the setting in the admin interface doesn't seem to.

I tested the samba/boot problem more thoroughly, it's not that i have to 'restart' samba after a boot, but that samba isn't started at all on boot...  i had assumed that installing samba would set this up, but no..  and there doesn't seem to be a way of setting it to start in the IIAB admin interface.

The two things i am trying to accomplish is to run the RPI as an IIAB/Internet Archive test installation for the networks I work with, and to use it as a file server for the machines on my LAN.
Here are the pastebinit results:

http://paste.debian.net/1084280/
http://paste.debian.net/1084281/
http://paste.debian.net/1084282/

btw, one sporadic problem i've noticed is that dns doesn't seem to always work. I frequently (but not always have to change the /etc/resolv.conf entry:
nameserver 127.0.0.1
to:
nameserver 8.8.8.8

holta, 2019-05-29:
@mikerjensen 

1) IIAB doesn't manage your own Samba installation, but here are 2 commands you might want to consider:

systemctl is-enabled smbd
systemctl enable smbd

2) Reminder / please open your iptables firewall by changing variable ports_externally_visible from level "3" to level "4" within /etc/iiab/local_vars.yml to enable WAN-side access to Samba

3A) Within http://box.lan/admin > Configure > Network Parameters ...

- click the radio button next to "Gateway - You need two or more network adapters, one of which must be connected to the Internet or a larger Network (WAN)" under "Server's Role in the Network"
- continue below...

3B) To set "Static WAN IP Address" in  > Override IP Addresses (also within Admin Console) please now:

- click the checkbox "Check to use a static WAN IP Address instead of DHCP"
- pastebinit < /opt/iiab/iiab/config_vars.yml
- click "Save Configuration"
- pastebinit < /opt/iiab/iiab/config_vars.yml
- click "Install Configured Options" &mdash; or much faster, run these at the command-line:
  - cd /opt/iiab/iiab
  - sudo ./iiab-network

Thanks & talk soon!

mitra42, 2019-05-30:
The issue Mike had with the startup screen is fixed (pending confirmation by Mike) at  internetarchive/dweb-mirror#144

mikerjensen, 2019-05-30:
Great, yes it mainly works now! perhaps one thing i'd suggest is some sort of progress indicator on the page load? the text "Loading - note this can take a while if no-one else has accessed this item yet" is helpful, but even if i go to a page that has been accessed, it takes quite a long time, even the home page - just using the left arrow/back button on the browser to return to the home page takes about 30 seconds before anything else on the screen shows up. Similar times to go to the Community Audio page...

Going to the Community Video page is faster, but only loads the top part of the menu, doesn't show any data...like this: 
[image]

Clicking on the 'Collections' menu option option on this page doesn't do anything... the page remains the same.

I tried reloading / refreshing the page, and it got worse - the broken image icon replaced the small green graphic of people using a video camera.

Then after going to the 'American Libraries' page, and returning to the home page (via the back button), the 'American Libraries' title to the browser window remained there.  See graphic below. Also shown in the graphic is that there is a green ellipse in the white circle icons for the Community Video and Community Texts icons, but not for the others, i'm not sure of the significance of this. 

[image]

Finally, this morning i had to edit resolv.conf to nameserver 8.8.8.8 as 127.0.0.1 didn't work....

holta, 2019-05-30:
> (3A) Within http://box.lan/admin > Configure > Network Parameters ...
> 
> * click the radio button next to "Gateway - You need two or more network adapters, one of which must be connected to the Internet or a larger Network (WAN)" under "Server's Role in the Network"
> * continue below...
> 
> (3B) To set "Static WAN IP Address" in the "Override IP Addresses" sections (also within Admin Console > Configure > Network Parameters) please now:
> 
> * click the checkbox "Check to use a static WAN IP Address instead of DHCP"
> * pastebinit < /opt/iiab/iiab/config_vars.yml     # to publish a copy beforehand
> * click "Save Configuration"
> * pastebinit < /opt/iiab/iiab/config_vars.yml     # to publish a copy afterwards
> * click "Install Configured Options" — or much faster, run these at the command-line:
>   
>   * cd /opt/iiab/iiab
>   * sudo ./iiab-network

@mikerjensen can you post the 2 pastebins above, if poss?

[9 more replies]

closed 2019-06-17 by holta