iiab/iiab #1582: How do you know IIAB is successfully installed

mitra42, 2019-03-27:
### Expected behavior and actual behavior.
Some indication of success

### Steps to reproduce the problem.
Install IIAB from the script, run `sudo iiab` each time it fails. How do you know when its succeeded, becuase it appears both failure and success lead to a reboot with no message. 

I have just run "sudo iiab" three times, probably unnecessarily (and unfortunately things like en.zip get downloaded every time ) trying to catch whether it succeeded or failed, but it always jumps to reboot sometime when I'm not looking at the screen. 

### Specifications like the version of the IIAB, operating system version, or hardware details.

I would suggest that on success it should say something like .... 
"IIAB installation successfull, press return to reboot and then open .... in your browser"

holta, 2019-03-27:
> I would suggest that on success it should say something like ....
"IIAB installation successfull, press return to reboot and then open .... in your browser"

I agree, as IIAB is increasingly installed by new classes of eager implementers who don't have school server context, and don't understand or even remember that they've read these 2 lines: (if they read them at all!)
```
# 7. IIAB AUTO-REBOOTS WHEN DONE (typically 1-to-2 hours later) which sets
#    hostname, improves RTC + memory mgmt, and starts BitTorrents if necessary
```
i.e. https://github.com/iiab/iiab-factory/blob/master/iiab#L32-L33

What is now needed is indeed much more loud (literally in-your-face) messaging announcing that IIAB software was successfully installed, and outlining possible next steps.

FYI almost exactly this was suggested during a Feb 28 design call with @jvonau, @georgejhunt & @m-anish in the context of ppl trying to install IIAB onto esoteric networking HW, where IIAB's installer naturally/regrettably does always not complete.

Conclusion: I'll prototype this and roll this out seeking community feedback in April, if not possibly earlier.

Ref: #1519 #1520 #1521

mitra42, 2019-03-27:
Even if you remember that it will AUTO-REBOOT WHEN DONE, you don't know whether it auto-rebooted on successful completion or because it failed. (Auto reboot being a likely response to failure).

holta, 2019-03-27:
> Even if you remember that it will AUTO-REBOOT WHEN DONE, you don't know whether it auto-rebooted on successful completion or because it failed. (Auto reboot being a likely response to failure).

Indeed.  As in the past IIAB was installed by < 100 people around the world most all of whom were in direct contact with us in the OLPC Community School Server project (formerly known as XSCE) so this was less of an issue.

But it's now time to make it far more self-explanatory / self-service...for folks who are not at all connected to the One Laptop Per Child tradition.

holta, 2019-03-28:
Done, see: https://github.com/iiab/iiab-factory/blob/master/iiab#L248-L278

closed 2019-03-28 by holta