iiab/iiab #1583: How to connect to a new server

mitra42, 2019-03-27:
### Expected behavior and actual behavior.

As the instructions at http://d.iiab.io/install.txt say ...
```
8. EXPLORE IIAB: http://box or http://box.lan or http://172.18.96.1 THEN
 ADD CONTENT at http://box.lan/admin -- passwords are at http://FAQ.IIAB.IO
```
### Steps to reproduce the problem.
Try and access any of these addresses, none of these three work.
 
a) it only comes up as 172.18.96.1 if you are connected to it by wireless, but connecting to "Internet in a Box" wireless isn't in the instructions.
b) It doesn't appear to come up on http://box.lan it appears to come up on http://box.local
c) http://box goes to box.com at least on Firefox.

mitra42, 2019-03-27:
And ... http://box.local seems to take a while to propogate, I tried immediately after the reboot and it failed, a while later it worked which I think (but could be wrong) means that MDNS is taking its time to announce.

tim-moody, 2019-03-27:
All of the above access urls only apply if you are connected to the IIAB wireless hotspot. Otherwise it depends on name resolution in the host network. box.local is avahi and not all client OSes can consume it.

holta, 2019-03-27:
Thanks @mitra42 !

@tim-moody is absolutely correct but this deserves a more complete explanation for newcomers, as part of the http://d.iiab.io install process.

I'll work on it and seek community feedback in April alongside your #1582 &mdash; it's now time to move past our OLPC legacy &mdash; where this kind of info was often communicated separately, over various community support channels.

holta, 2019-03-27:
> And ... http://box.local seems to take a while to propogate, I tried immediately after the reboot and it failed, a while later it worked which I think (but could be wrong) means that MDNS is taking its time to announce.

@jvonau do you know why this might be happening?  We can also ask @georgejhunt during tmrw's call?

mitra42, 2019-03-27:
An immediate start would be to suggest connect to the Internet In A Box WiFi in [http://d.iiab.io/install.txt], there is no reason when following that doc to assume its necessary, AND, given how buggy the install process is, there is every reason to think that failure to connect is because the installation failed, rather than because you aren't actually testing it right.

holta, 2019-03-27:
> given how buggy the install process is

Which bugs are you taking about?  We're seen more than 9 out of 10 succeed, with the failures generally (but not always) having to do with networking failures and upstream mirrors that Raspbian does not offer with high bandwidth or reliability.

mitra42, 2019-03-27:
See #1582 so its hard to tell if it succeeded, and I see red error messages going by (something to do with kalite being in another sharing circle) and it was previously failing for me when trying over Wifi. 

I've found it works if I resume with a `sudo iiab` but I haven't got it to work first time yet, and since I have no way to tell whether it succeeded my confidence in the install is pretty low (and I've done it more than once now) - so not being able to connect to the box as described in the instructions isn't going to improve that confidence.

holta, 2019-03-27:
> See #1582 so its hard to tell if it succeeded

I strongly agree.  Will be made crystal clear in coming weeks, starting with IIAB installs that occur over Ethernet, and then also those that occur over Wi-Fi.

> I see red error messages going by (something to do with kalite being in another sharing circle)

Can you send a copy of the exact error message from /opt/iiab/iiab/iiab-install.log if you still have it that IIAB machine kicking around?

> was previously failing for me when trying over Wifi.

#1520 & #1536 are extremely high-priority for IIAB 7.0 this spring, i.e. making IIAB very friendly to install IIAB over Wi-Fi.

> not being able to connect to the box as described in the instructions

Working on #1583 here as we speak!

mitra42, 2019-03-27:
Sorry @holta  the error message isn't in that log any more. I reran `sudo iiab` several times afterwards, trying to see if it had succeeded or failed.

holta, 2019-03-28:
> An immediate start would be to suggest connect to the Internet In A Box WiFi in [http://d.iiab.io/install.txt], there is no reason when following that doc to assume its necessary

Done.  See 8. and 9. within http://d.iiab.io/install.txt (https://github.com/iiab/iiab-factory/blob/master/install.txt#L35-L45) and also:

https://github.com/iiab/iiab-factory/blob/master/iiab#L260-L277

Not the final word of course (as many don't read at all!)  But it's a start until more rigorous and in-your-face explanations are presented during the IIAB's install process itself, as part of #1582.

holta, 2019-03-28:
> Sorry @holta the error message isn't in that log any more. I reran `sudo iiab` several times afterwards, trying to see if it had succeeded or failed.

/opt/iiab/iiab/iiab-install.log keeps records of all runs, so it might still be in there?

In any case, plz do forward the error message if this problem later recurs, Thanks @mitra42 !

tim-moody, 2019-03-28:
the log is usually cumulative, so rerunning should not remove old runs

holta, 2019-03-28:
Resolved by iiab/iiab-factory#69

closed 2019-03-28 by holta

mitra42, 2019-04-24:
Lets revisit the box.lan / box / box.local 

The instructions still say box.lan or box in a lot of places. Even on WiFi this seems to be a myth, I just did a clean install , and connecting to it on WiFi it came up as box.local. Neither box, nor box.lan worked.

I'm on a Mac, I don't know if the '.local' depends on the system you are on (determining .local means to  look at local MDNS etc) or if its something the RPI is putting out, but same thing happens on my iphone - its box.local  

I also notice https://github.com/iiab/iiab/issues/8 and https://github.com/iiab/iiab/issues/1347 dealing with problems of box and box.lan showing up but they might not be related.

tim-moody, 2019-04-26:
please see my comment to #1576 . You are blaming IIAB for some other device's inability to resolve box.lan.

mitra42, 2019-04-26:
Maybe ... its some other device - see George's response there which I'll try when I get home. 

But note the crucial thing is that even as a sophisticated user the instructions didn't work - even when connected to the WiFi as instructed, and from multiple other devices they failed, so its fine to ignore them from the point of this issue as long as people are prepared for the user-support when none of the suggested alternatives (box or box.lan) work, (#1576 relates to the specific problem of a redirect failing and a redirect  unfortuantely cannot just guess what domain name is working)