iiab/iiab #1586: Outstanding IIAB issues (network & Node-RED) on Ubuntu Server 19.04 Beta

holta, 2019-03-29:
@jvonau "A start job is running for Wait for Network to be Configured" holds up boot (for more than 2 min) every time, on BIG-sized IIAB install on Ubuntu Server 19.04 Beta, in a VM.

What are the best ways to diagnose this?

@m-anish "systemctl start nodered" fails every time on Ubuntu 19.04 Beta:

> root@box:~# systemctl status nodered
● nodered.service - Node-RED
Loaded: loaded (/etc/systemd/system/nodered.service; enabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Fri 2019-03-29 00:49:53 PDT; 29min ago
Process: 1986 ExecStart=/usr/bin/node-red-pi -v (code=exited, status=203/EXEC)
Main PID: 1986 (code=exited, status=203/EXEC)
>
> Mar 29 00:49:53 box.lan systemd[1]: nodered.service: Service RestartSec=100ms expired, scheduling restart.
Mar 29 00:49:53 box.lan systemd[1]: nodered.service: Scheduled restart job, restart counter is at 5.
Mar 29 00:49:53 box.lan systemd[1]: Stopped Node-RED.
Mar 29 00:49:53 box.lan systemd[1]: nodered.service: Start request repeated too quickly.
Mar 29 00:49:53 box.lan systemd[1]: nodered.service: Failed with result 'exit-code'.
Mar 29 00:49:53 box.lan systemd[1]: Failed to start Node-RED.

Refs: #1580 PR #1585

m-anish, 2019-03-31:
Difficult to spend any cognitive energy on this. Check nodesource's readme where they themselves say they do not support 19.04. Its not even out yet! 

https://github.com/nodesource/distributions

Please test this "bug" maybe a month after 19.04 is out, and nodesource officially supports this.

holta, 2019-04-11:
Disco Dingo (Ubuntu 19.04) Final Freeze was a couple hours ago.

Final release expected in a week from today (April 18).

holta, 2019-04-18:
Ubuntu 19.04 was released earlier today:
https://wiki.ubuntu.com/DiscoDingo/ReleaseNotes
http://releases.ubuntu.com/19.04/

holta, 2019-04-18:
Tagging @deldesir & @floydianslips just in case either has time to test IIAB 7.0/master (http://d.iiab.io) on the newly released [Ubuntu 19.04](http://releases.ubuntu.com/19.04/) ?

holta, 2019-04-21:
@floydianslips will test http://d.iiab.io on Ubuntu Server 19.04

(We're looking for someone else to test on Ubuntu Desktop 19.04 if poss!)

holta, 2019-04-22:
@floydianslips installed BIG-sized http://d.iiab.io onto Ubuntu Server 19.04 and it appears to have worked flawlessly...

holta, 2019-04-22:
This ticket's issues (appear) resolved...but do re-open as nec.

holta, 2019-04-22:
@floydianslips clarified that the last remaining issue with BIG-sized http://d.iiab.io on Ubuntu Server 19.04 persists...the issue is:

> "A start job is running for Wait for Network to be Configured" holds up boot (for more than 2 min) every time, on BIG-sized IIAB install on Ubuntu Server 19.04 Beta, in a VM.
>
> Likely affects MIN-sized & MEDIUM-sized too?
>
> What are the best ways to diagnose this?

floydianslips, 2019-04-22:
Adding optional: true to /etc/netplan/50-cloud-init.yaml seems to solve the reboot issue. I don't know if it causes any issues down the chain but it reboots quickly and renders IIAB menu. 

network:
       ethernets:
              enp0s3: 
                     dhcp: true
                     _optional: true_
            version: 2

holta, 2019-04-22:
> Adding `optional: true` to /etc/netplan/50-cloud-init.yaml seems to solve the reboot issue. I don't know if it causes any issues down the chain

@georgejhunt @jvonau should this be wrapped within an Ansible condition?  If so, what condition (e.g. is_ubuntu_19 or similar) is appropriate?

floydianslips, 2019-04-23:
The `optional: true` allows the boot process to continue and then it brings up the network interface when it is ready. I have only tried this on a VM where the host is connected via ethernet.

holta, 2019-04-23:
> The `optional: true` allows the boot process to continue and then it brings up the network interface when it is ready. I have only tried this on a VM where the host is connected via ethernet.

I'm hoping @georgejhunt or @jvonau can assist you in creating a PR...or just feel free to throw one together for their review??

holta, 2019-05-03:
> The optional: true allows the boot process to continue and then it brings up the network interface when it is ready. I have only tried this on a VM where the host is connected via ethernet.

@floydianslips I've reconfirmed with a fresh install that Ubuntu 19.04 pauses for more than 2min during boot, with message `A start job is running for Wait for Network to be Configured`

_(Please go ahead and create the above-mentioned PR !)_

holta, 2019-05-04:
CLARIF: PR #1633 doesn't hurt, but it and then running ./iiab-network are NOT sufficient to overcome Ubuntu Server 19.04's stalling for 2+ minutes during each boot:
```
A start job is running for Wait for Network to be Configured
```

holta, 2019-05-04:
FYI **prior** to running IIAB's roles/network, Ubuntu 19.04's /etc/netplan contains one single file **50-cloud-init.yaml** here:
```
# This file is generated from information provided by
# the datasource.  Changes to it will not persist across an instance.
# To disable cloud-init's network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
network:
    ethernets:
        enp0s3:
            dhcp4: true
    version: 2
```
FYI **after** running IIAB's roles/network, Ubuntu 19.04's /etc/netplan contains one single file **01-iiab-config.yaml** here:
```
network:
  version: 2
  renderer: networkd
  ethernets:
    enp0s3:
      dhcp4: yes
#
```

holta, 2019-05-04:
Should @floydianslips perhaps consider adding the line he's found to work...
```
optional: true
```
...after `dhcp4: yes` in https://github.com/iiab/iiab/blob/master/roles/network/templates/network/netplan.j2#L18 ?  (i.e. this is the file that's used as a template to create /etc/netplan/01-iiab-config.yaml on all versions of Ubuntu 18.04 or higher)

jvonau, 2019-05-04:
Attach iiab.ini, need to see which network backend was detected

holta, 2019-05-04:
1) @jvonau called and helped figure out this telltale sign: `systemctl status systemd-networkd-wait-online` was failing (on this Ubuntu 19.04 VM) until I added `optional: true` within /etc/netplan/01-iiab-config.yaml &mdash; which got rid of the 2+ min delay on boot &mdash; and correspondingly allows the `systemd-networkd-wait-online` service to run properly.

> Attach iiab.ini, need to see which network backend was detected

2) Here are the 3 network sections from the Ubuntu 19.04 VM's /etc/iiab/iiab.ini
```
[detected_network]
has_ifcfg_gw = none
prior_gateway_(device_gw2) = "enp0s3"
dhcpcd_result =
network_manager_active = False
systemd_networkd_active = True
wan_in_interfaces = False
wireless_list_1(wifi1) = not found-1
wireless_list_2(wifi2) = not found-2
num_wifi_interfaces = 0
discovered_wireless_iface = none
discovered_wired_iface = none
num_lan_interfaces = 0
gui_static_wan = False
iiab_lan_iface = none
iiab_wan_iface = enp0s3
[computed_network]
iiab_wan_enabled = True
user_wan_iface = enp0s3
iiab_wan_iface = enp0s3
iiab_lan_enabled = True
user_lan_iface = auto
iiab_lan_iface = none
iiab_network_mode = Appliance
hostapd_enabled = True
host_ssid = u19-srv-may-2-iarchive
host_wifi_mode = g
host_channel = 6
[network]
dansguardian_enabled = False
squid_enabled = False
wondershaper_enabled = False
iiab_network_mode_applied = Appliance
dhcpd_enabled = False
dhcp_service2 = dnsmasq disabled
named_enabled = False
dnsmasq_enabled = False
no_net_restart = False
```

PR #1634 (2019-05-04): Revise template for /etc/netplan to mitigate 2+ min stall during Ubuntu 19.04 boot
Fixes #1586's more-than-2-minute delay ("A start job is running for Wait for Network to be Configured") on every boot of IIAB on Ubuntu 19.04

Thanks to @jvonau who agrees with this change suggested by @floydianslips [*] while hoping others test this across with Ubuntu 18.04, 19.04 and others across many different network conditions in coming months!

[*] originally suggested by https://askubuntu.com/questions/972215/a-start-job-is-running-for-wait-for-network-to-be-configured-ubuntu-server-17-1/1110474#1110474  [... cut]
closed 2019-05-04 by holta