iiab/iiab #1591: Better Kolibri performance with static file serve and wsgi

ivanistheone, 2019-04-07:
Hello @arky @holta 

I want to point you to some potential performance improvements for Kolibri on IIAB that my colleagues have recently been working on. As you are aware, the Kolibri backed is a standard Django server which means it can benefit from "collaborative" integration with the web server for static file serving and WSGI.

The work of my colleagues has focussed on `nginx` as the web server, but I believe you can adapt the code below to be used for `apache` just as well.

### 1. Static file serving
Consider adding a directive similar to this:
https://github.com/learningequality/cloud-kolibri-demo/blob/master/config/nginx_site.template.conf#L13-L15
with the appropriate path set based on the kolibri role parameters.
Static file serving will greatly improve the performance of video playback by multiple users.

### 2. WSGI app serving (optional)
By default, kolibri runs its own web server (cherrypi) to handle requests, but higher performance is possible by using Kolibri as a WSGI application (the standard protocol for serving Python web applications).

Running Kolibri via the WSGI bridge called from the web server introduces some complexity, but can lead to performance benefits (use all CPU cores available). 

Unfortunately our team's focus on config turning and testing has focused predominantly on  nginx setup so you'll have to adapt to the `apache` setup and configs to set the appropriate packages working. Here are the relevant links:

Setup wsgi upsteam: https://github.com/learningequality/kolibri-server/blob/master/nginx.conf#L27-L31

Start the Kolibri uWSGI worker: https://github.com/learningequality/kolibri-server/blob/master/debian/kolibri-server.init#L48-L68

uWSGI worker config: https://github.com/learningequality/kolibri-server/blob/master/uwsgi.ini

### 3 use .deb package (optional)
I wanted to flag for you that Kolibri can be installed on Ubuntu/Debian systems via the `.deb` package, see [here](https://kolibri.readthedocs.io/en/latest/install.html#id7). The exact same code will be installed, but you should consider trying out the alternative install-via-deb approach to simplify the kolibri recipe.

Note that I'm not the best person to consult for advice on the above matters ( @jredrejo @benjaoming are much more knowledgeable), but I thought I'd post some links to the best of my knowledge as partial response to https://github.com/learningequality/kolibri-docs/issues/79

holta, 2019-04-07:
Tagging @darkenvy so he too's aware, as efforts strengthen to scrape the best http://LibreTexts.org for offline education of many kinds!

jredrejo, 2019-04-08:
I think comments or issues related to this topic should be continued in https://github.com/learningequality/kolibri-server
There is where all the work to automatically setup kolibri using uwsgi & nginx is being done.
I suggest that people interested on it check the sources there, test the package using the [kolibri-proposed ppa](https://launchpad.net/~learningequality/+archive/ubuntu/kolibri-proposed/) , and fill any issue that can be found or recommendation to improve it.

holta, 2019-05-10:
See Also:

#1646 Migrate IIAB to kolibri-server for RPi performance

holta, 2019-07-07:
See also the Kolibri via dpkg / .deb discussion (for systemd unit file especially?) here:

#1832 Tracking the best systemd unit file(s) for Kolibri?