iiab/iiab #1599: "apt install hostapd" fails every time on Ubuntu 18.04 NUC without AP

holta, 2019-04-16:
This error occurs even after reboot, even after manually attempting `apt remove hostpad` :

> TASK [2-common : Install 22 common packages: acpid, bridge-utils, bzip2, curl, gawk, hostapd, htop, i2c-tools, logrotate, make, mlocate, netmask, net-tools, ntfs-3g, pandoc, rsync, sudo, tar, unzip, usbmount, usbutils, wget] ***
/etc/systemd/system/hostapd.servicefatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1555373153, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'hostapd'' failed: E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "rc": 100, "stderr": "E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr_lines": ["E: Sub-process /usr/bin/dpkg returned an error code (1)"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nhostapd is already the newest version (2:2.6-15ubuntu2.2).\nThe following packages were automatically installed and are no longer required:\n  gyp libfltk-images1.3 libfltk1.3 libhttp-parser2.7.1 libjs-async\n  libjs-inherits libjs-node-uuid liblua5.1-0 libuv1-dev nodejs-doc pandoc-data\n  qtdeclarative5-qtquick2-plugin\nUse 'apt autoremove' to remove them.\n[master 4633297] saving uncommitted changes in /etc prior to apt run\n 1 file changed, 1 insertion(+), 1 deletion(-)\n0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.\n1 not fully installed or removed.\nAfter this operation, 0 B of additional disk space will be used.\nSetting up hostapd (2:2.6-15ubuntu2.2) ...\r\nFailed to mask unit: File /etc/systemd/system/hostapd.service already exists.\r\ndpkg: error processing package hostapd (--configure):\r\n installed hostapd package post-installation script subprocess returned error exit status 1\r\nErrors were encountered while processing:\r\n hostapd\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "hostapd is already the newest version (2:2.6-15ubuntu2.2).", "The following packages were automatically installed and are no longer required:", "  gyp libfltk-images1.3 libfltk1.3 libhttp-parser2.7.1 libjs-async", "  libjs-inherits libjs-node-uuid liblua5.1-0 libuv1-dev nodejs-doc pandoc-data", "  qtdeclarative5-qtquick2-plugin", "Use 'apt autoremove' to remove them.", "[master 4633297] saving uncommitted changes in /etc prior to apt run", " 1 file changed, 1 insertion(+), 1 deletion(-)", "0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.", "1 not fully installed or removed.", "After this operation, 0 B of additional disk space will be used.", "Setting up hostapd (2:2.6-15ubuntu2.2) ...", "Failed to mask unit: File /etc/systemd/system/hostapd.service already exists.", "dpkg: error processing package hostapd (--configure):", " installed hostapd package post-installation script subprocess returned error exit status 1", "Errors were encountered while processing:", " hostapd"]}

Main error appears to be:
```
Failed to mask unit: File /etc/systemd/system/hostapd.service already exists.
```

While I could easily remove that file...I'd also like to understand why this keeps happening, so that `iiab-install --reinstall` becomes genuinely reliable for others...

holta, 2019-04-16:
Clarif: this error does not occur on Ubuntu 16.04 and Ubuntu 18.04 NUC's that contain Wi-Fi Access Points, nor does it occur within an RPi 3 B+ that likewise contains an AP.

holta, 2019-04-16:
Running "apt purge hostapd" and then rerunning "iiab-install --reinstall" finally seems to have worked..._but is that really something regular folks are supposed to be able to figure out?_

holta, 2019-04-16:
Per @jvonau's suggestion, here is everything from today (2019-04-15) as taken from the end of 10.8.0.22's /var/log/apt/term.log : http://paste.ubuntu.com/p/bj8QGYj8NY/

(This includes the results of "apt purge hostapd" near the bottom, which is what finally got "iiab-installer --reinstall" working again.)

holta, 2019-05-01:
This appears no longer to be an issue.

(Plz re-open as necessary.)

jvonau, 2019-08-15:
https://git.launchpad.net/ubuntu/+source/wpa/tree/debian/hostapd.NEWS
https://git.launchpad.net/ubuntu/+source/wpa/commit/?id=8269c9738670c454ce4cc48cc280e951a490f71f

jvonau, 2019-08-15:
[hostapd.postinst](https://git.launchpad.net/ubuntu/+source/wpa/tree/debian/hostapd.postinst) is what is being run 
> #!/bin/sh
set -e
if [ -d /run/systemd/system ] && [ "$1" = configure ]
then
    DAEMON_CONF=
    . /etc/default/hostapd
    if [ -z "$DAEMON_CONF" ] && [ ! -r /etc/hostapd/hostapd.conf ] && ! systemctl --quiet is-active hostapd.service
    then
	    systemctl mask hostapd.service
    fi
fi
#DEBHELPER#
exit 0

from .62 
> cat /etc/iiab/iiab.env
This is a configuration file for IIAB
It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=7.0
IIAB_REVISION=0
IIAB_GATEWAY_ENABLED=False
LAN_IP=172.18.96.1
OS=ubuntu
OS_VER=ubuntu-18
WWWROOT=/library/www/html
STAGE=9
FQDN=box.lan
IIAB_USB_LIB_SHOW_ALL=True
IIAB_WAN_DEVICE="enp0s3"
IIAB_LAN_DEVICE="none"
HOSTAPD_ENABLED=True

That is a bit misleading as it is a single interface machine
> systemctl is-enabled hostapd
disabled

holta, 2019-08-15:
@jvonau I've re-opened this in case you have further suggestions.

holta, 2019-08-15:
@jvonau can you confirm this #1599 should be closed now that PR #1929 has been merged?

holta, 2019-08-16:
Let's declare victory.

@jvonau please re-open if/as nec-

closed 2019-08-16 by holta