iiab/iiab #1605: New OSM Vector Maps need clean UX + clean instructions ("pick a continent or region")

holta, 2019-04-18:
- [ ] I'll revise https://github.com/iiab/iiab-admin-console/pull/162 today, so implementers can at least survive the UX at http://box.lan/admin -> Install Content -> Get Map Region !

  Critical for the masses at https://linuxfestnorthwest.org/conferences/2019 next week, as well as the general public after that.

  Specifically: Right now it's far too confusing as people think that 2 or 3 continents can be downloaded and will all display together.  Basically everyone is falling in this costly UX trap &mdash; even OSM experts &mdash; leading to erroneous expectations and a massively expensive / time-consuming community support apparatus that's just not realistic.

  **Conclusion:** punting is not appropriate, i.e. pushing implementer frustrations & support burdens onto others, which only makes folks more aggravated when they discover that simultaneous display of 2 or 3 continents' detail is not realistically solvable.  **Remember:** IIAB installation & customation *always* happens during those very few days/hours/minutes that educators / implementers have as they're racing prior to their deployment/flight.

  So we need to cut this off at the pass, with a UX that's genuinely honestly setting reasonable expecatations.  i.e. making clear that a single continent/region is the only thing that's realistic for 99.9 % of all implementers.

  **Possibly the only OSM Mass Customization that's truly realistic & supportable & documentable in the end is individual downloads from https://openmaptiles.com ???**

- [ ] Also let's get organized in coming weeks around larger documentation needing overhaul in places like:

  - [ ] http://FAQ.IIAB.IO Item 33 "[How do I add zoomable maps for my region?](http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_add_zoomable_maps_for_my_region.3F)"
  - [ ] https://github.com/iiab/maps (top of page, and bottom of page which is...)
  - [ ] https://github.com/iiab/maps/blob/master/docs/README.md
  - [ ] https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/README.md
  - [ ] Etc!

holta, 2019-04-24:
@georgejhunt please review both PRs:

https://github.com/iiab/iiab-admin-console/pull/162
https://github.com/iiab/maps/pull/2

holta, 2019-06-17:
There's been a huge amount of progress here during Q2 2019, and it should advance further towards IIAB 7.0's release in coming days as @georgejhunt & I work on:

#1710 Living Docs for IIAB 7.0 Maps [OSM]

Towards releasing this quite major advance that @georgejhunt & others have been working on for many months, which roughly speaking is:

https://github.com/iiab/iiab/issues/877#issuecomment-405935272 Package up vector-based OSM maps

holta, 2019-06-27:
Good Enough for IIAB 7.0 !

For IIAB 7.1, let's please not lose these 2 remaining (possible!) priorities:

- [ ] Individual country/city downloads from https://openmaptiles.com
- [ ] Video/Sceencast Walk-Thru of Map Pack installation & (especially!) common usage tips, _that many teachers/students never figure out unless shown!_

holta, 2020-05-05:
Let's consider closing this now, for the release of IIAB 7.1, as documentation gets cleaned up here:

#2387 "How It Works: IIAB 7.0 -> 7.1 Maps Doc"

holta, 2020-05-07:
There are some thoughtful ideas above towards catalyzing IIAB Maps / adoptions, so let's keep this ticket open for the IIAB 7.2 release cycle...

holta, 2020-09-12:
As discussed Tuesday (2020-09-08) wording will be greatly clarified here...
https://github.com/iiab/maps/blob/master/osm-source/pages/installer/index.html
...to help new IIAB implementers understand how they can download-and-install OSM Maps Packs + Satellite Photo "squares".

(Integrating this into Admin Console can presumably then follow soon after that.)

holta, 2020-09-16:
@avni Tim wanted to draw your attention to:

"Mods for new Maps"
https://github.com/iiab/iiab-admin-console/issues/340

tim-moody, 2020-09-16:
>To bring this to the masses, here are 3 areas where rapid improvement is possible in coming weeks,

well not that rapid.

holta, 2020-09-27:
Hopefully IIAB 7.2 RC2 can be released within ~24h, including some version of PR iiab/maps#29 (IIAB Maps installer page explanations overhauled) based on @avni's feedback of 2020-09-09.

Before final release, we can (should!) also update https://github.com/iiab/iiab/wiki/IIAB-Maps with late-breaking options around Wikidata layers etc?

While making it all bit more clear for new folk (struggling to understand everything, when seeing this for the very 1st time!)