iiab/iiab #1606: Mode N not working in Raspberry pi 3 B+

KW4NP, 2019-04-20:
### Expected behavior and actual behavior.
It seems like selecting mode n in the Admin > Config > Internal WiFi AP > Internal WiFi Access Point Mode would make the wifi a bit faster over mode g, but instead it seems to kill the WiFi altogether.  Is this an IIAB bug?
### Steps to reproduce the problem.
Change from mode g to mode n and run install configured options.
### Specifications like the version of the IIAB, operating system version, or hardware details.
Raspberry pi 3 B+, IIAB 7.0

KW4NP, 2019-04-20:
Tested and confirmed that changing back to "g" seems to re-enable WiFi AP.  It seems like mode "ac" is not working either.

holta, 2019-04-20:
@georgejhunt @jvonau @tim-moody can you please look into this during April if possible?

georgejhunt, 2019-04-20:
https://unix.stackexchange.com/questions/184175/how-to-set-up-wifi-hotspot-with-802-11n-mode

KW4NP, 2019-04-20:
Adding "ieee80211n=1" restores proper function (ie. WiFi on), but I can't seem to confirm that it is actually in N mode.

holta, 2019-04-25:
@tim-moody mentions:

> https://github.com/raspberrypi/linux/issues/2619 for [ 802.11ac ]

Is this broadly desirable?

Which particular kinds of IIAB communities does this help if so?

KW4NP, 2019-04-29:
It could help anyone desiring to transfer large files (Kolibri, etc.) over wifi.  One of the participants in the discussion mentioned is using Rpi's for a school and wants AC mode.

holta, 2019-04-29:
> It could help anyone desiring to transfer large files (Kolibri, etc.) over wifi. One of the participants in the discussion mentioned is using Rpi's for a school and wants AC mode.

Tagging @m-anish in case he knows other use case where "faster offline broadband" is truly needed in the field?

jvonau, 2020-04-27:
https://github.com/raspberrypi/linux/issues/2619