iiab/iiab #1611: Some Rachel modules not displaying properly

KW4NP, 2019-04-22:
### Expected behavior and actual behavior.
Some Rachel (OER2GO) modules do display correctly when accessed from the main menu.  Blockly-Games is an example.  When the link on the home page is clicked, this page comes up.
[image]
However, apparently it should come up looking like this:
[image]
Which is what you get if you click on the blockly-games link in the index.
However, modules with an HTML index in the module root don't seem to have any problems.
Any suggestions on a possible workaround?
### Steps to reproduce the problem.
Install blockly-games from the admin console and add it to the home page.  There are multiple other modules affected as well.

### Specifications like the version of the IIAB, operating system version, or hardware details.
IIAB 7.0, Raspberry Pi 3 B+

holta, 2019-04-22:
> There are multiple other modules affected as well.

Can you name such others?

Such that these can hopefully be fixed upstream in the OER2Go / RACHEL source?

KW4NP, 2019-04-22:
There are quite a few which seem to be acting like this, but of course I have not tried all.  I am having problems with fr-hesperian, haiti-future, and I believe the electronic library of Quebec as well (I'm waiting for it to download so I can try it again).  I have only tried the French modules so far, but it seems to be linked somehow to IIAB as there are files that I assume should be providing the index (index.htmlf and rachel-index.php, linked together somehow) but for some reason they are not being used.  Is there a workaround for now?

KW4NP, 2019-04-22:
Could it be related to the menu item link?

holta, 2019-04-22:
> Could it be related to the menu item link?

Sure, we can also work around some of these locally within IIAB.

Do you want to submit a Pull Request addressing the most crucial ones?

Possibly emulating these 2 but translating from Spanish to English, French & Haitian Creole:

- https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/es-blockly-games.html
- https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/es-blockly-games.json

holta, 2019-04-22:
> * https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/es-blockly-games.json

See line 21 especially:
```
    "start_url": "blockly-games/es/index.html",

KW4NP, 2019-04-22:
Made a pull request with the files for English and French.  I'm not sure about all of the fields, but I went ahead and changed the language description fields.  Doesn't IIAB read the rachel-index.php file?  It seems like that would be the best way to provide the proper functionality for all of the modules.

holta, 2019-04-22:
@KW4NP should this issue be closed?

Or are there more OER2go/RACHEL modules you want to highlight and/or refine within https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/ ?

holta, 2019-04-22:
@KW4NP fyi if you installed using http://d.iiab.io (master branches, i.e. prerelease of IIAB 7.0 for now) then you can run the following in general / in future at any time, to grab the latest menu descriptions & logos that others might have contributed:
```
cd /opt/iiab/iiab-admin-console
git pull
./install
```
PS run the above as root, or preface each line with the `sudo` command.

KW4NP, 2019-04-22:
@holta, Thanks for your help.  There are a few more that I will probably fix as I get around to them, but there are also a few that I'm not exactly sure how to handle (fr-hesperian for one) as they only have the index.htmlf and rachel-index.php files to describe what is inside.  Maybe we could somehow process the rachel-index.php if there is no menu-def file?  Thanks also for your help with the updates.

holta, 2019-04-22:
> There are a few more that I will probably fix as I get around to them

Thank you @KW4NP

> there are also a few that I'm not exactly sure how to handle (fr-hesperian for one) as they only have the index.htmlf and rachel-index.php files to describe what is inside. Maybe we could somehow process the rachel-index.php if there is no menu-def file? Thanks also for your help with the updates.

Please take a look at the 3 Hesperian Guides near the bottom of http://iiab.me/dr (English, French, Haitian Creole) to see if that has what you and others want?

We could use your help encoding such things into [/opt/iiab/iiab-admin-console/roles/js-menu/files/menu-files/menu-defs/](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/) and will certainly assist you within our limited capacity, as Haiti and other quasi-French countries definitely need this!

KW4NP, 2019-04-22:
Yes, that is what I am looking for.  I'm not sure about others but it's certainly better than the default index.  Any chance we could import that into the master?

holta, 2019-04-22:
> Yes, that is what I am looking for. I'm not sure about others but it's certainly better than the default index. Any chance we could import that into the master?

Somehow I thought @tim-moody already done that in 2017 but I guess not, apologies.

Can you clarify which are most needed, such that this can hopefully happen in coming weeks &mdash; is it those 3 Hesperian descriptions (mini-menus) in English, French and Haitian Creole especially?

PS feel free also to scrape the output of http://iiab.me/dr or similar, creating your own PR(s) at anytime.

KW4NP, 2019-04-22:
All I really need is French, thanks.  If I have time I will work on what I can do here.

holta, 2019-04-23:
> All I really need is French, thanks. If I have time I will work on what I can do here.

Ok!  Can you write me privately, such that others working on Haiti & similar might collaborate with you?

My email address (and other ways to make contact with IIAB communities) is here:
http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F

holta, 2019-05-09:
Related:

PR https://github.com/iiab/iiab-admin-console/pull/196 to correct menu defs: en-blockly-games.json, fr-blockly-games.json, fr-haitifutur.json

KW4NP, 2019-05-14:
The box I have been working on is now deployed, so I probably won't be adding any more menu definitions in the near future.  Thanks for your help!

closed 2019-05-14 by KW4NP