iiab/iiab #1612: Re-running Admin Console's ./install should install the latest /library/www/osm-vector/maplist/assets/regions.json (from d.iiab.io if nec)

holta, 2019-04-24:
@georgejhunt https://github.com/iiab/maps/pull/2 is now merged, but the following does not work to install its revised http://d.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json to /library/www/osm-vector/maplist/assets/regions.json: 
```
cd /opt/iiab/iiab-admin-console
./install
```
Can you help fix this, such that `./install` always install the very latest version of regions.json, as this evolves over time?

(Or explain how this should work?!)

holta, 2019-04-25:
For now the answer appears to be to run:
```
cd /opt/iiab/iiab
./runrole osm-vector
```
A button to make this happen (i.e. get the latest regions.json catalog of possible downloadable map regions) could be incredibly useful within IIAB's Admin Console — just like we have such buttons to refresh to the latest catalog of downloadable ZIM files, and likewise for downloadable RACHEL/OER2Go modules...

georgejhunt, 2019-04-26:
I've made a second attempt to make bounding boxes -- currently at http://10.8.0.14/temp with handle of bboxes. I wont do any more until we have agreement. Pretty easy to change

holta, 2019-04-30:
> I've made a second attempt to make bounding boxes -- currently at http://10.8.0.14/temp with handle of bboxes. I wont do any more until we have agreement. Pretty easy to change

[1] Thanks @georgejhunt & @mir123 let's go ahead with these 1+9+1 bounding boxes you've posted to http://iiab.net/bboxes &mdash; as also specified here:

0.5 San Jose: West edge, South edge, East edge, North edge (rename it from Min)
1 North America: 179, -2, -2, 90
2 Central America: -119, -2, -44, 33
3 South America: 180, 0, -18, 25
4 Europe: -40, 26, 67, 90
5 Africa: -34, 0, 64, 46
6 Middle East: 13, 3, 72, 51 (create it)
7 North Asia: 19, 34, -160, 90 (rename it from Asia)
8 South Asia: 52, -11, -134, 56 (rename it from Southeast Asia)
9 Oceania: 52, 0, -82, 6 (rename it from Australia)
9.5 World: W, S, E, N (rename it from Detailed World)

[2] _Community Cocumentation is critically important and I will help you both wherever I can here_ &mdash; such that others can do this work of assembling the most popular bounding boxes in future years &dash; and even this year possibly, if strong communities arise needing particular bounding boxes that we did not anticipate!

e.g. ["How do I add zoomable maps for my region?"](http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_add_zoomable_maps_for_my_region.3F) within [FAQ.IIAB.IO](FAQ.IIAB.IO) (among others!)

Related:
#1605 New OSM Vector Maps need clean UX + clean instructions ("pick a continent or region")
#1618 iiab should not install menu-defs [OSM]
#1623 Cleanup of regions.json for IIAB 7.0 OSM "continents"
#1625 Move osm-vector materials from d.iiab.io to GitHub

holta, 2019-04-30:
Old and New OSM map region sizes thanks to @georgejhunt:

- San Jose: 623M -> 625M
- North America: 11.6G -> 16.8G
- Central America: 2.55G -> 2.98G
- South America: 3.11G -> 5.94G
- Europe: 19.8G -> 22.7G
- Africa: 4.84G -> 10.1G
- Middle East: 6.6G
- North Asia: 9.12G -> 13.9G
- South Asia: 6.32G -> 8.60G
- Oceania: 1.64G -> 3.60G
- World: 54.8G -> 54.8G

See bounding box visualizations @ http://iiab.net/bboxes (site temporarily down, check later!)

holta, 2019-05-05:
FYI the latest draft is usually at https://github.com/georgejhunt/maps/blob/new-bboxes/resources/regions.json as mentioned on #1623.

holta, 2019-05-31:
Map catalog updating is a hard problem that @tim-moody & @georgejhunt have been thinking about, that will inevitably be needed, but it might not land in IIAB 7.0 as its year-by-year design issues (more similar to ZIM catalogs than OER2Go catalogs?) are still being teased out.

As such http://FAQ.IIAB.IO should include concise documentation to keep IIAB implementers from getting stuck and frustrated:

1) How to workaround errors like "Add any MAPS that are installed to home/menu.js" @ #1707
2) How to clean out stale Map Packs from /library/www/osm-vector-maps/
3) How to manually grab a new /library/www/osm-vector/maplist/assets/regions.json when that becomes necessary from places like...
   - http://d.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json
   - https://github.com/georgejhunt/maps/blob/new-bboxes/resources/regions.json
4) Any other key aspects of OSM Vector Maps that will empower IIAB implementers to become understanding & independent?

holta, 2019-06-07:
Related stuff:

PR iiab/iiab-admin-console#211 Download if filename is changed

#1707 failed: "Add any MAPS that are installed to home/menu.js" [how to delete/upgrade OSM map packs?]

holta, 2019-06-20:
Let's declare this documented workaround sufficient for IIAB 7.0:
https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-upgrade-an-iiab-map-pack

closed 2019-06-20 by holta