iiab/iiab #1618: iiab should not install menu-defs

tim-moody, 2019-04-27:
/library/www/html/js-menu/menu-files/menu-defs/en-map_test.json should be installed with the other menu-defs in iiab-admin-console

tim-moody, 2019-04-27:
if it is not just for test purposes, in which case it should be removed

holta, 2019-05-02:
@georgejhunt is this resolved or on track to resolution?

Is @tim-moody's https://github.com/iiab/iiab-admin-console/issues/184 ("osm overwrites var getZimVersions in js-menu.js" -> "var getZimVersions in js-menu.js is not call definition for $.when as originally intended") related?

tim-moody, 2019-05-03:
not at all

georgejhunt, 2019-05-05:
The name collision between "osm" and "map" as I refine additions to admin-console are driving me crazy. I had taken the easy way out, and used the 3 char "osm" in code. But osm also refers to the osm role which Anish wants to keep. 
So I'm proposing to consolidate around using "map"  in my code, where osm is currently being used, and vector-maps for the collective reference. So the my role would be "vector-maps", and ansible variables would be "vector_maps_install"

georgejhunt, 2019-05-05:
is this the time to bite the bullet, and change "*_enabled" to "*_enable"?

tim-moody, 2019-05-05:
role and enable/install flags sound good. of course osm-vector-maps would be even better if a little long. the variable 'map' in your code depends on scope. if an ansible variable there could be conflict as they are all basically global. if javascript or backend, I think you are free to do as you chose.

tim-moody, 2019-05-05:
the change of 'enabled' to 'enable' is not a simple variable name change. the state needs to be tracked (we assume that if enabled is true is has been enabled, but that is not necessarily the case) and all should be done at once. also, in my view using iiab.env and sed to record state is about as ugly as it gets.

tim-moody, 2019-05-05:
what do any of the recent comments have to do with the heading of this issue?

georgejhunt, 2019-05-05:
The connection to heading is my current work flow on the objective of
moving menu stuff from iiab to admin-console (my understanding of the
heading).

Unfortunately, my mind is becoming more global all the time. Remembering
the context, and therefore the correct reference to find stuff, is more the
issue for me. I got tripped up on "intended_use" -- osm vs map. - confusion
I probably created for myself by picking one and then the other.

tim-moody, 2019-05-05:
where does the vector map stuff actually live? could it not just be a webroot type?

tim-moody, 2019-05-05:
the only reason that kiwix and kalite are separate types is that they have their own servers listening on their own ports. anything served by apache is either html (in modules) or webroot.

closed 2019-05-09 by holta

holta, 2019-05-09:
CLARIF: Resolved by PR #1639