iiab/iiab #1622: No web access

mikerjensen, 2019-04-28:
I just completed my first IIAB install on Raspbian (last night, so i guess it is ver 7.0). Running /usr/sbin/iiab  completes without errors, but there is nothing in the root directory of the web server - going to http://box.lan i just get the index of /home - with no files listed there, followed by: "Apache/2.4.25 (Raspbian) Server at box.local Port 80"

I'm not sure it is related, but nextcloud is running (http://box.lan/nextcloud loads), but doesn't recognise any login or password combination even although they are valid logins because they work when i ssh into the RPI

holta, 2019-04-28:
> (http://box.lan/nextcloud loads), but doesn't recognise any login or password combination even although they are valid logins because they work when i ssh into the RPI

ssh credentials have nothing to do with IIAB services / IIAB Apps in most cases.

Did you try the default username/password published at http://FAQ.IIAB.IO Item 13 ?

("What are the default passwords?")

holta, 2019-04-28:
> I just completed my first IIAB install on Raspbian (last night, so i guess it is ver 7.0). Running /usr/sbin/iiab completes without errors, but there is nothing in the root directory of the web server - going to http://box.lan i just get the index of /home - with no files listed there, followed by: "Apache/2.4.25 (Raspbian) Server at box.local Port 80"
>
> Also http://box.lan/admin gives a 404 error - url not found
>
> BTW, i'm using Mitra's fork - i.e i edited /usr/sbin/iiab to change the git clone line to:
git clone https://github.com/mitra42/iiab -b distweb --depth 1

How exactly did you install IIAB?

Does /opt/iiab/iiab-admin-console exist, as it should after installing using http://d.iiab.io ?

Please make sure your IIAB install is truly complete, including all steps such as:
https://github.com/iiab/iiab-factory/blob/master/iiab#L182-L186
https://github.com/iiab/iiab-factory/blob/master/iiab#L211-L213

holta, 2019-04-28:
@mikerjensen can you run the following to give us remote access to your RPi?
```
cd /opt/iiab/iiab
./install-support
```
(If so, please confirm that the above has run while your RPi is online, Thanks!)

mikerjensen, 2019-04-28:
Holta, thanks for the quick replies...
I installed IIAB by following Mitra's instructions here: 
https://github.com/internetarchive/dweb-mirror/blob/master/README-iiab-raspberry.md
1) I managed to get IIAB working after digging around the IIAB install docs, and discovered that if i ran  /opt/iiab/iiab/iiab-install  and  /opt/iiab/iiab-admin-console/install that made it work  - i guess they are not called from /usr/sbin/iiab ...

So all seems to work and i'm downloading content, although Mitra's addons at http://box.lan:4244  don't seem to load...

2) re nextcloud passwords,  oops, i didn't realise IIAB services would require different username/passwords - i tried the default ones from the FAQ - Admin/changeme and they worked fine, thanks.

holta, 2019-04-28:
> Holta, thanks for the quick replies...
> I installed IIAB by following Mitra's instructions here:
> https://github.com/internetarchive/dweb-mirror/blob/master/README-iiab-raspberry.md
>
> 1. I managed to [...]

FYI that is not true.  Both repos are definitely installed from /usr/sbin/iiab I outlined above 4h ago.  How you installed IIAB remains a mystery to many who have looked over this ticket.  We'll try to assist but at the same time, please understand that if you deviate from the official http://d.iiab.io installer (which creates /usr/sbin/iiab from https://github.com/iiab/iiab-factory/blob/master/iiab) this is of course at your own risk :-)

> So all seems to work and i'm downloading content, although Mitra's addons at http://box.lan:4244 don't seem to load...

Please ask @mitra42 ?

> 1. re nextcloud passwords,  oops, i didn't realise IIAB services would require different username/passwords - i tried the default ones from the FAQ - Admin/changeme and they worked fine, thanks.

Great!

> Anyway, i ran install-support it ran to completion, not sure what to do next though.

@georgejhunt can you login this machine remotely now, to help @mikerjensen consider next steps?

georgejhunt, 2019-04-28:
What symptoms am I being asked to help with. The one mentioned in the
call have been resolved.
I need to know the ip address connected with the tun0 device when you
type "ip a" at console.
How can I help you?  What symptoms?

mikerjensen, 2019-04-28:
Thanks George, I think not to worry at this point, as IIAB is now working.  I will follow up with @mitra42 regarding the issues with adding the Internet Archive to IIAB. Thanks again for all the quick responses everyone.

mitra42, 2019-04-28:
So ... the installer was based on the steps @holta had me go through, i.e. to rebase my ansible scripts onto IIAB's current release candidate at iiab/iiab#master which is mitra42/iiab#distweb.   Then as described in https://github.com/internetarchive/dweb-mirror/blob/master/README-iiab-raspberry.md to `curl -o /usr/sbin/iiab https://raw.githubusercontent.com/iiab/iiab-factory/master/iiab` and edit Line 179 to point to `git clone https://github.com/mitra42/iiab -b distweb --depth 1` instead of the master branch.

I'm pretty sure that @m-anish has done a clean install from that branch on his VM, and I've done an almost clean install (Only rerunning `runrole internetarchive` as I fixed each bug)

I notice a fix was posted yesterday to IIAB/IIAB#master to fix nextcloud, I merged it into mitra42/iiab#distweb last night, but that was after @mikerjensen 's first install failed. 

I think Mike then put it together by trial and error using some suggestions from me to `runrole internetarchive` and `runrole network`, but I'm guessing that the `/usr/sbin/iiab` script has never run to completion and that is why some things are still missing.   

@mikerjensen  think you were earlier hitting the problem of box.local showing up but not box.lan as documented in #1576 but if you have got around those and can now see box.lan then http://box.lan:4244 should work and bring up the IA interface. If not then catch me on whatsapp and we'll work through some debugging. 

My suggestion would be to do a fresh install on new Raspbian, but I know that can take hours, going to /opt/iiab/iiab and doing `git pull` then running /usr/sbin/iiab might solve the problem with nextcloud, and fix anything still missing, but I'm guessing its also going to skip over some steps it thinks are installed.  I think `/usr/sbin/iiab --reinstall` tries a bit harder, but I'm not sure of that.

holta, 2019-04-28:
> My suggestion would be to do a fresh install on new Raspbian,

Agreed.  Other steps might also have been accidentally omitted if /usr/sbin/iiab did not fully complete (@mitra42's explanation sounds very plausible to me!)

> but I know that can take hours, going to /opt/iiab/iiab and doing `git pull`

Then you can run `iiab-install --reinstall` (none of which is necessary now that http://box.lan/nextcloud works!)

Likewise the following is no longer necessary, if Admin Console (http://box.lan/admin) is now working:
```
cd /opt/iiab/iiab-admin-console
git pull
./install
```
(Above 3-line recipe brings improvements to Admin Console almost every week!)

mitra42, 2019-04-28:
Problems with box.lan:4244 seem to be due to the IA's main server having  had a problem, and the restore bringing back an old certificate and there being no pre-installed content for IA. Tracey is working on fixing hte cert so hopefully should have that quick.

holta, 2019-04-30:
Just FYI @mitra42 is using `journalctl -u internetarchive` to examine systemd log files, to refine this PR.

mitra42, 2019-04-30:
Actually ... there are no changes to the PR (since updating as @georgejhunt requested to disable internetarchive by default), all the changes have been in how dweb-mirror responds to failures of something or other (in this case the main gateway server) and have been pushed to that, or the dweb-archive NPM release and the PR automatically gets the current versions from NPM

holta, 2019-05-04:
@mikerjensen can this issue be closed now?

Separately: is #1629 on track towards a cleaner understanding (...or not there yet?)

holta, 2019-05-04:
@mikerjensen plz re-open if core issues remain unresolved?

closed 2019-05-04 by holta