iiab/iiab #1623: Cleanup of regions.json for IIAB 7.0

holta, 2019-04-29:
@georgejhunt

Each stanza contains 4 versions of the same "continent" name within https://github.com/iiab/maps/blob/master/osm-source/ukids/assets/regions.json

1) I've cleaned up "perma_ref": and "title": for all 1+9+1 regions — done!

2) Can you clean up the other 2 in coming days/weeks, in each of those 1+9+1 regions, such that each stanza's group of 4 names is self-consistent prior to IIAB 7.0's release?  These are the two where I'll need your help:

   a) header prior to the braces ( this affects the green mouseover region, displayed by http://box.lan/admin > Install Content > Add Map Region )

   b) "url": ( actual URL of map region, typically at archive.org )

PS I put the 1+9+1=11 "seq": numbers in order.

I could also use your help setting these other 9 fields (in each of those 11 cases!) when it's time:

      "west": "-122.24",
      "center_lon": "-122.0",
      "zoom": "10",
      "date": "2017-07-03",
      "north": "37.45",
      "east": "-121.60",
      "center_lat": "37.35",
      "south": "37.23",
      "size": "622747648"

Ref: #1619 and associated issues/PRs

PPS @georgejhunt did you or anyone experiment with Torrenting any of these earlier "continents" ?
```
San Jose: 608.15 MB
Central America: 2.43 GB
South America: 2.97 GB
North America: 11.03 GB
Africa: 4.62 GB
Europe: 18.90 GB
Asia: 8.69 GB
Southeast Asia: 6.03 GB
Australia: 1.56 GB
World: 52.24 GB

holta, 2019-04-30:
Huge thanks to @georgejhunt who is working on a new draft version here:
https://github.com/georgejhunt/maps/blob/new-bboxes/resources/regions.json

I hope he can consider ordering the 1+9+1 "continents" as follows:
```
5 San Jose
|
10 North America
20 Central America
30 South America
40 Europe
50 Africa
60 Middle East
70 North Asia
80 South Asia
90 Oceania
|
95 World

holta, 2019-05-02:
@georgejhunt can you change San Jose's seq number from 10 to 5 at https://github.com/georgejhunt/maps/blob/new-bboxes/resources/regions.json ?

Context: this is George's draft regions.json for IIAB 7.0, that will later appear in places like:
https://github.com/iiab/maps/blob/master/osm-source/ukids/assets/regions.json
And possibly also:
http://d.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json

georgejhunt, 2019-05-02:
Changing ukids regions.json would break the download function for anyone
who has current master, at this  point.
I could change http;//iiab.net/osm-vector to use the new bboxes, and
reflect the new sizes.

holta, 2019-05-02:
> Changing ukids regions.json would break the download function for anyone
who has current master, at this  point.
I could change http;//iiab.net/osm-vector to use the new bboxes, and
reflect the new sizes.

No doubt.

My question was: @georgejhunt can you change San Jose's seq number from 10 to 5 at https://github.com/georgejhunt/maps/blob/new-bboxes/resources/regions.json ?

(This appears to be a bug in your file...with both North America & San Jose having the same sequence number.)

holta, 2019-05-07:
Preview of IIAB 7.0's forthcoming 10 new downloadable map regions, that is easier to understand than http://iiab.net/bboxes for most people:

_Vector Maps Test Page -- Use The Admin Console to Download a Selected Region
http://iiab.net/osm-vector = http://iiab.net/osm-vector/maplist/_

Ref: #1612

georgejhunt, 2019-05-29:
The region generation pipeline has moved on beyond any of the suggestions in this issue. Let's start a new issue, if one is required, based upon current reality

closed 2019-05-29 by georgejhunt