iiab/iiab #1624: Calibre-Web install fails during pip on Raspbian

holta, 2019-04-29:
@tim-moody, @jvonau & @OzzieIsaacs do you understand why Calibre-Web is suddenly no longer installing on Raspbian?

Specifically, Internet-in-a-Box's standard Calibre-Web installer playbook appears to now fail every time at https://github.com/iiab/iiab/blob/master/roles/calibre-web/tasks/main.yml#L49-L54 with the following error:

> TASK [calibre-web : Download Calibre-Web dependencies (using pip) into virtual environment] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": ["/usr/local/calibre-web/bin/pip2", "install", "-r", "/usr/local/calibre-web/requirements.txt"], "msg": "stdout: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nRequirement already satisfied: Babel>=1.3 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 1)) (2.6.0)\nRequirement already satisfied: Flask-Babel>=0.11.1 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 2)) (0.12.2)\nRequirement already satisfied: Flask-Login>=0.3.2 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 3)) (0.4.1)\nRequirement already satisfied: Flask-Principal>=0.3.2 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 4)) (0.4.0)\nRequirement already satisfied: singledispatch>=3.4.0.0 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 5)) (3.4.0.3)\nRequirement already satisfied: backports_abc>=0.4 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 6)) (0.5)\nRequirement already satisfied: Flask>=0.11 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 7)) (1.0.2)\nRequirement already satisfied: iso-639>=0.4.5 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 8)) (0.4.5)\nRequirement already satisfied: PyPDF2==1.26.0 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 9)) (1.26.0)\nRequirement already satisfied: pytz>=2016.10 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 10)) (2018.9)\nRequirement already satisfied: requests>=2.11.1 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 11)) (2.21.0)\nRequirement already satisfied: SQLAlchemy>=1.1.0 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 12)) (1.2.18)\nRequirement already satisfied: tornado>=4.1 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 13)) (5.1.1)\nRequirement already satisfied: Wand>=0.4.4 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 14)) (0.5.1)\nRequirement already satisfied: unidecode>=0.04.19 in /usr/local/calibre-web/lib/python2.7/site-packages (from -r /usr/local/calibre-web/requirements.txt (line 15)) (1.0.23)\nCollecting Pillow>=5.4.0 (from -r /usr/local/calibre-web/requirements.txt (line 16))\n  Using cached https://files.pythonhosted.org/packages/81/1a/6b2971adc1bca55b9a53ed1efa372acff7e8b9913982a396f3fa046efaf8/Pillow-6.0.0.tar.gz\nRequirement already satisfied: Jinja2>=2.5 in /usr/local/calibre-web/lib/python2.7/site-packages (from Flask-Babel>=0.11.1->-r /usr/local/calibre-web/requirements.txt (line 2)) (2.10)\nRequirement already satisfied: blinker in /usr/local/calibre-web/lib/python2.7/site-packages (from Flask-Principal>=0.3.2->-r /usr/local/calibre-web/requirements.txt (line 4)) (1.4)\nRequirement already satisfied: six in /usr/local/calibre-web/lib/python2.7/site-packages (from singledispatch>=3.4.0.0->-r /usr/local/calibre-web/requirements.txt (line 5)) (1.12.0)\nRequirement already satisfied: click>=5.1 in /usr/local/calibre-web/lib/python2.7/site-packages (from Flask>=0.11->-r /usr/local/calibre-web/requirements.txt (line 7)) (7.0)\nRequirement already satisfied: Werkzeug>=0.14 in /usr/local/calibre-web/lib/python2.7/site-packages (from Flask>=0.11->-r /usr/local/calibre-web/requirements.txt (line 7)) (0.14.1)\nRequirement already satisfied: itsdangerous>=0.24 in /usr/local/calibre-web/lib/python2.7/site-packages (from Flask>=0.11->-r /usr/local/calibre-web/requirements.txt (line 7)) (1.1.0)\nRequirement already satisfied: urllib3<1.25,>=1.21.1 in /usr/local/calibre-web/lib/python2.7/site-packages (from requests>=2.11.1->-r /usr/local/calibre-web/requirements.txt (line 11)) (1.24.1)\nRequirement already satisfied: idna<2.9,>=2.5 in /usr/local/calibre-web/lib/python2.7/site-packages (from requests>=2.11.1->-r /usr/local/calibre-web/requirements.txt (line 11)) (2.8)\nRequirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/calibre-web/lib/python2.7/site-packages (from requests>=2.11.1->-r /usr/local/calibre-web/requirements.txt (line 11)) (3.0.4)\nRequirement already satisfied: certifi>=2017.4.17 in /usr/local/calibre-web/lib/python2.7/site-packages (from requests>=2.11.1->-r /usr/local/calibre-web/requirements.txt (line 11)) (2018.11.29)\nRequirement already satisfied: futures in /usr/local/calibre-web/lib/python2.7/site-packages (from tornado>=4.1->-r /usr/local/calibre-web/requirements.txt (line 13)) (3.2.0)\nRequirement already satisfied: MarkupSafe>=0.23 in /usr/local/calibre-web/lib/python2.7/site-packages (from Jinja2>=2.5->Flask-Babel>=0.11.1->-r /usr/local/calibre-web/requirements.txt (line 2)) (1.1.0)\nBuilding wheels for collected packages: Pillow\n  Building wheel for Pillow (setup.py): started\n  Building wheel for Pillow (setup.py): finished with status 'error'\n  Complete output from command /usr/local/calibre-web/bin/python2 -u -c \"import setuptools, tokenize;__file__='/tmp/pip-install-vLa0n7/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\\r\\n', '\\n');f.close();exec(compile(code, __file__, 'exec'))\" bdist_wheel -d /tmp/pip-wheel-KkEsol --python-tag cp27:\n  running bdist_wheel\n  running build\n  running build_py\n  creating build\n  creating build/lib.linux-armv7l-2.7\n  creating build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/_binary.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/Image.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageOps.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/MicImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/MspImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/TarIO.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageQt.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImtImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/features.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/ImageMode.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-2.7/PIL\n  copying src/PIL/GimpPaletteFile.py -> build/lib.linux-ar [... cut ...] 'open', open)(__file__);code=f.read().replace('\\r\\n', '\\n');f.close();exec(compile(code, __file__, 'exec'))\" install --record /tmp/pip-record-mNJ7SL/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/calibre-web/include/site/python2.7/Pillow\" failed with error code 1 in /tmp/pip-install-vLa0n7/Pillow/\nYou are using pip version 19.0.3, however version 19.1 is available.\nYou should consider upgrading via the 'pip install --upgrade pip' command.\n"}

OzzieIsaacs, 2019-04-29:
I also had this problem on raspian. I added pillow as dependency to extract non jpeg files from PDFs during upload. If you install pillow via pip it fails, the solution is to install pillow via apt.

holta, 2019-04-29:
> I also had this problem on raspian. I added pillow as dependency to extract non jpeg files from PDFs during upload. If you install pillow via pip it fails, the solution is to install pillow via apt.

Do you mean `apt install python-pil` ?

Or do you mean `apt install pillow` ?

PS can this be installed on Ubuntu 18.04+ & Debian 9+ too?

OzzieIsaacs, 2019-04-29:
I used the commando  `apt install python-imaging`, but this is indeed not working on Ubuntu 18.04.
I googled a bit and the package `python-pil` should be available in all 3 distros. So `apt install python-pil` should give the right result.

holta, 2019-04-29:
> package `python-pil` should be available in all 3 distros. So `apt install python-pil` should give the right result.

Amazing, Thanks You @OzzieIsaacs !

holta, 2019-04-29:
@OzzieIsaacs neither of these commands solved the problem of pip installing calibre-web

- `apt install python-pil` _[was already installed]_
- `apt install python-imaging` _[i've now installed this]_

I rebooted and tried again to be sure; still no luck.

Do you have other ideas for getting Pillow working within this venv (virtual environment) on Raspbian?  Is the below excerpt (from the longer error I [pasted above](https://github.com/iiab/iiab/issues/1624#issue-438134511)) relevant?

> Running setup.py clean for Pillow\nFailed to build Pillow\nInstalling collected packages: Pillow\n Running setup.py install for Pillow: started\n Running setup.py install for Pillow: finished with [...]
...

FYI this is a virtual environment (venv) attempting-and-failing to [pip install Calibre-Web's requirements](https://github.com/iiab/iiab/blob/master/roles/calibre-web/tasks/main.yml#L49-L54) according to this file:
```
root@box:~# more /usr/local/calibre-web/requirements.txt
Babel>=1.3
Flask-Babel>=0.11.1
Flask-Login>=0.3.2
Flask-Principal>=0.3.2
singledispatch>=3.4.0.0
backports_abc>=0.4
Flask>=0.11
iso-639>=0.4.5
PyPDF2==1.26.0
pytz>=2016.10
requests>=2.11.1
SQLAlchemy>=1.1.0
tornado>=4.1
Wand>=0.4.4
unidecode>=0.04.19
Pillow>=5.4.0

holta, 2019-04-30:
FYI the same error arises when I delete the virtual environment (/usr/local/calibre-web) and try from scratch:
```
cd /opt/iiab/iiab
./runrole calibre-web
```

OzzieIsaacs, 2019-04-30:
I think I see the problem, the version of pillow on raspian is 4.04 ( or similar), so it still tries to install a newer version. Could you please try to change the version in the last line of requirements.txt to 4.0.0 and try again?

holta, 2019-04-30:
> I think I see the problem, the version of pillow on raspian is 4.04 ( or similar), so it still tries to install a newer version. Could you please try to change the version in the last line of requirements.txt to 4.0.0 and try again?

Reverting the last line of requirements.txt to `Pillow>=4.0.0` did not work (even after I deleted the entire /usr/local/calibre-web virtual environemtn and recreated it to be sure).

FWIW python-pil 4.0.0 is installed on Raspbian on RPi 3B+ here:
```
root@box:~# apt -a list python-pil
Listing... Done
python-pil/stable,now 4.0.0-4 armhf [installed]
```

Any other tests you recommend I run?

holta, 2019-04-30:
Just in case this contains tips that might help us?

> How to setup Python Imaging Library, Pillow, on Raspbian Stretch Lite for processing images on your Raspberry Pi
https://www.techcoil.com/blog/how-to-setup-python-imaging-library-pillow-on-raspbian-stretch-lite-for-processing-images-on-your-raspberry-pi/

holta, 2019-05-02:
This Calibre-Web problem is broadly preventing people from installing MEDIUM-sized and BIG-sized Internet-in-a-Box on the most popular platform (Raspberry Pi).

So I will temporarily disable Calibre-Web now, just FYI.

(Until a fix is hopefully available in coming days...)

holta, 2019-05-05:
Has anyone had luck pip installing Calibre-Web _directly onto Raspbian on RPi 3_ in recent weeks?

e.g. with Calibre-Web's official instructions at https://github.com/janeczku/calibre-web#quick-start ?

(Just to be 100% sure the ongoing problem [#1624] is not an issue with IIAB or its `/usr/local/calibre-web` venv [virtual environment] designed to protect dependencies &mdash; much like python-pil / Pillow here!)

holta, 2019-05-16:
@OzzieIsaacs it looks like a miracle!

1) Out of curiosity how were you are to remove the Python Imaging Library requirement from [requirements.txt](https://github.com/janeczku/calibre-web/blob/master/requirements.txt) per https://github.com/janeczku/calibre-web/commit/55bb8d45903c6aa9e6dfbfb2d83ec3882a42d96f#diff-b4ef698db8ca845e5845c4618278f29a ?

2) Are you possibly able to release another Calibre-Web similar to 2019-04-20's v0.62 prior to IIAB 7.0's June 2019's final release in coming weeks?  _(Or should we continue to auto-install Calibre-Web's master branch [live, dynamically] as part of IIAB's install process just for now?)_

holta, 2019-05-16:
Confirmed this issue's resolved on Raspbian on RPi 3 B+ hence PR #1668 which is now merged.

(And hopefully @OzzieIsaacs can still answer the questions above when he has time!)

OzzieIsaacs, 2019-05-30:
I just created a new release V0.6.3 with the latest improvements.
1. The python-imaging lib was optionally in the code, so the code could run without if from the beginning. Without python-imaging you are loosing some features (extracting and uplaoding non jpg files as cover), which where specially requested by you guys.
2. I recommend not using the "live, dynamically branch", it tends to having more problems than the stable releases (I'm still practising in having non bugging releases)

holta, 2019-05-30:
> I just created a new release V0.6.3 with the latest improvements.

Thank you @OzzieIsaacs 

> 1. The python-imaging lib was optionally in the code, so the code could run without if from the beginning. Without python-imaging you are loosing some features (extracting and uplaoding non jpg files as cover), which where specially requested by you guys.

Teachers & others find preview covers are vitally important for PDF, JPG and all visual media files.  If some of these are missing in V0.6.3 on Raspbian (Raspberry Pi) let us know how we can test future versions to improve this situation!

> 2. I recommend not using the "live, dynamically branch", it tends to having more problems than the stable releases (I'm still practising in having non bugging releases.

Will do, thanks again!

holta, 2019-05-31:
@OzzieIsaacs [Calibre-Web 0.6.3](https://github.com/janeczku/calibre-web/releases/tag/0.6.3) is now part of IIAB 7.0 pre-releases, thanks to PR #1709 which was (very briefly!) tested on Ubuntu 18.04 & Raspbian.  3 quick questions, to help everyone manage their Internet-in-a-Boxes moving forward:

- Is there a way to find out Calibre-Web's version number within Calibre-Web's web user interface?
- Is there a way to find out Calibre-Web's version number from the command-line?
- Is there an email list of similar way that I (holt @ laptop.org) can updated every time that a new version of Calibre-Web is released?

Thanks much for considering some-or-all of these 3 suggestions/requests if you can!

OzzieIsaacs, 2019-05-31:
To anwer your questions:
1.) Admins can go to the admin section, on the bottom the current version number is listed, as long as you are on the stable update channel (default) 
[image]
2.) Not yet, but it's a good point, I will include it in the next release
3.) No currently there is no newsletter, twitter, facebook, slack or anything else. Maybe the "watch releases" function from Github could suit your needs (https://help.github.com/en/articles/watching-and-unwatching-releases-for-a-repository)

holta, 2019-06-02:
Many thanks @OzzieIsaacs. keep in touch !

closed 2019-06-02 by holta