iiab/iiab #1626: Kolibri videos not loading

KW4NP, 2019-04-29:
### Expected behavior and actual behavior.
Kolibri videos show a warning "The media could not be loaded, either because the server or network failed or because the format is not supported" when attempting to view them.  However, this is actually because the videos and pictures are returning a 404 error, apparently because of a problem either with Apache or the URL.
### Steps to reproduce the problem.
Install Kolibri and content and attempt to view.  French Touchable Earth and KA are both not working properly. 
### Specifications like the version of the IIAB, operating system version, or hardware details.
IIAB 7.0, Raspberry Pi 3 B+

holta, 2019-04-29:
@jredrejo & @benjaoming is this a bug you might possibly be familiar with?  Or have suggestions for tracking down?

The Kolibri installation experience is still not quite yet smooth on IIAB unfortunately, as briefly discussed with @jamalex and @ivanistheone last week:

#1545 Kolibri Bug Scrub for IIAB 7.0

jredrejo, 2019-04-29:
@holta if you are using `URL_PATH_PREFIX`  together with apache, the symptoms look like this issue that has already been solved https://github.com/learningequality/kolibri/issues/5414 but fix has not been released yet, it will be on next release, 0.12.3.

holta, 2019-04-29:
> @holta if you are using `URL_PATH_PREFIX` together with apache, the symptoms look like this issue that has already been solved [learningequality/kolibri#5414](https://github.com/learningequality/kolibri/issues/5414) but the fix has not been released yet, it will be on next release, 0.12.3.

Thanks @jredrejo !

Tagging @arky in case he too has ideas?

KW4NP, 2019-04-29:
Any option for a workaround?

KW4NP, 2019-04-29:
Confirmed that without the URL_PATH_PREFIX environment variable it works much better.  However, this is still probably broken for other users.

jredrejo, 2019-04-29:
> Any option for a workaround?

holta, 2019-05-10:
Kolibri 0.12.3 appears finalized yesterday, and should be pip installable (for testing with IIAB) in coming days off of: https://learningequality.org/download/

holta, 2019-05-18:
Closeable now that Kolibri 0.12.3 (#1608) is part of IIAB?!

holta, 2019-06-17:
@KW4NP I'm closing this now assuming 0.12.3, 0.12.4 (and 0.12.5 being released in a few days) have solved this?

(But feel free to re-open as nec!)

Ref: #1736

closed 2019-06-17 by holta