iiab/iiab #1629: Edit menu/add new language

mikerjensen, 2019-05-02:
Using IIAB 7.0, When i go to ContentMenus->Edit Menu Items->Select Menu Item->Edit 

I can't change any of the fields in the top half of the screen - the cursor becomes a stop-sign... - i've been asked to do this to change the names of the options for people who don't speak english - I thought perhaps these menu options are restricted, but i can't change the 'menu item language' drop-down either, although perhaps this is because i don't have other languages installed, however it is not clear if/ where to add new languages for this option anyway....a quick google around iiab site didn't reveal much about adding other languages but perhaps i missed it... The menu is the one below..

[image]

tim-moody, 2019-05-02:
As the header says, if you change any of the top set of attributes you have effectively created a new menu item definition. This functionality is coming, but I wanted to finish edit first.

There are a number of existing menu item definitions that are not English. Click Select Language to view them in Edit Content Menus.

If you give me a concrete example of the definition you want to create I could say more.

holta, 2019-05-02:
Hi @mikerjensen 

Thanks for writing up your experience!

Hearing perspectives like yours greatly helps everone, as we continue to refine the Internet-in-a-Box UX for implementers across dozens of countries.

_Small Clarification: you're using a [pre-release of IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes), rather than the final product, which will be available sometime in June 2019._

mikerjensen, 2019-05-02:
Thanks Tim, Holta. I've just distributed 12 RPIs preloaded with IIAB to our community networks here at a meeting in Nairobi - they will go back to India, Indonesia, Brasil, Colombia, Uganda, South Africa, Argentina, DRC, Myanmar and Malawi. I am still a complete noob w regard to IIAB, and haven't had much time to play with it before everyone leaves....Anyway, we thought to edit the text  on this page not with the aim of  creating a new menu item, but as a test to translate the text into a local language. I had missed the different language options on the Edit Content Menu page, but on that page  i'm unclear of the difference between the 'Select languages'  large blue button at the top, and the 'Select Default Language' drop-down menu lower in the page. I tried changing the language in the drop down, and then did 'Save Menu' but this didn't seem to result in any changes. 

Would also be good to know how to go about adding a language that is not in the list...

tim-moody, 2019-05-03:
We normally expect that en-oob, which is only there to provide context, will be removed from the menu and we also normally expect that menu items arise as content is added.

'Select languages' large blue button at the top will filter the Menu Item Definitions displayed in the list on the right. Always check to see if the definition you want exists before creating a new one.

The 'Select Default Language' drop-down is not used yet, but refers to the main language for all menu item defs.

A Menu Item Definition is generally a unique combination of intended use, language and target. So if you change one of these you have effectively created a new one.

Again, the functionality to create a new one is currently in development.

closed 2019-05-06 by mikerjensen