iiab/iiab #1631: ssid doesn't change

mikerjensen, 2019-05-02:
Using the Web interface to Change the Internal Wi-Fi AP name from the default, doesn't change the SSID as broadcast, it remains Internet in a Box, even after reboot.
Using 7.0 pre

holta, 2019-05-02:
Indeed: @floydianslips & @darkenvy observed exactly the same thing this past wkd

Are you guys submitting a patch/button as was mentioned over the wkd?

holta, 2019-05-06:
@tim-moody is it possible Admin Console's "SSID changer" code is stale as a result of it having been designed for Fedora / CentOS ~3 years ago?

(@floydianslips will poke at this with @darkenvy this week, and could use any tips you have...)

tim-moody, 2019-05-06:
@holta @floydianslips @darkenvyI probably makes sense to confirm the bug before you randomly cut code.
Using AdmCons current master on rpi 3 I did the following:
in Configure->Internal Wifi AP I changed SSID Internet in a Box to Different SSID
I clicked Save Configuration
I clicked Install Configured Options
I monitored progress in Display Job Status
When it succeeded, I clicked Control->Reboot
On my laptop I opened the wifi picker and saw Different SSID
I connected and could ssh in and view the home page

holta, 2019-05-07:
> @holta @floydianslips @darkenvyI probably makes sense to confirm the bug before you randomly cut code.

Last week 3 people independently watched the SSID fail to change (@floydianslips, @darkenvy & @mikerjensen) so is it possible that the UX needs to be explained better, within Admin Console's web interface itself?

holta, 2019-05-07:
FWIW I include serious usability bugs within the bug category, which appears (likely?) to be the situation here.  Either that of @floydianslips, @darkenvy & others have identified a systematic code failure on different OS's which they should fully clarify.

holta, 2019-05-07:
Recap: @floydianslips, @darkenvy & @mikerjensen please post details, so that @tim-moody and all of us can get on the same page here, understanding what you are experiencing on what exact OS and HW.

e.g. if in the end it turns out to be a case of back-end software that works but front-end buttons that people can't figure out how to use it -- that's just as important of a [UX/design] bug -- that we all need to understand better so we can fix it!

mikerjensen, 2019-05-08:
Thanks folks, you're quite right Tim, i just checked  - It is
      indeed possible to change the SSID through the web interface -
      after Save Configuration i hadn't followed that with 'Install
      Configured Options'  - I guess I assumed saving the configuration
      changes would also make them active after a reboot. It seems
      perhaps an unnecessary step to have a separate action for 'install
      configured options' - but if there is a good reason for this, then
      perhaps the dialog box that comes up after clicking the 'Save
      Configuration' button should also include a prompt/reminder to
      click the 'Install Configured Options' button as well..
    Mike
    
      p { margin-bottom: 0.25cm; line-height: 115%; }a.cjk:link { }a.ctl:link { }
    On 07/05/19 23:57, A Holt wrote:
    
    
      
      Recap: @floydianslips,
        @darkenvy
        & @mikerjensen
        please post details, so that @tim-moody
        and all of us can get on the same page here, understanding what
        you are experiencing on what exact OS and HW.
      e.g. if in the end it turns out to be a case of back-end
        software that works but front-end buttons that people can't
        figure out how to use it -- that's just as important of a
        [UX/design] bug -- that we all need to understand better so we
        can fix it!
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/iiab/iiab/issues/1631#issuecomment-490284233",
"url": "https://github.com/iiab/iiab/issues/1631#issuecomment-490284233",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-05-08 by mikerjensen