iiab/iiab #1632: Ansible 2.8.x's throws many new warnings

holta, 2019-05-02:
`apt` now installs Ansible 2.8.0rc2 automatically as part of IIAB installs, as a result of https://launchpad.net/~ansible/+archive/ubuntu/ansible

This is a surprise given Release Candidates were not pushed by apt in the past, and I had not expected we'd face Ansible 2.8.x until 2.8.0's official release date of 2019-05-16.

In any case, this new Ansible 2.8.x issues many warnings we need to address in due course, for example:

> [DEPRECATION WARNING]: Distribution Ubuntu 18.04 on host 127.0.0.1 should use
/usr/bin/python3, but is using /usr/bin/python for backward compatibility with
prior Ansible releases. A future Ansible release will default to using the
discovered platform python for this host. See https://docs.ansible.com/ansible/
2.8/reference_appendices/interpreter_discovery.html for more information. This
feature will be removed in version 2.12. Deprecation warnings can be disabled
by setting deprecation_warnings=False in ansible.cfg.

Whereas on Raspbian Ansible 2.8.x shows this instead:

> [WARNING]: Platform linux on host 127.0.0.1 is using the discovered Python interpreter at /usr/bin/python, but future installation of another Python interpreter could change this. See https://docs.ansible.com/ansible/2.8/reference_appendices/interpreter_discovery.html for more information.

Running a BIG-sized install of IIAB yielded 222 of these warnings:

> TASK [0-init : Set port 443 for Admin Console if adm_cons_force_ssl] *********** [DEPRECATION WARNING]: evaluating adm_cons_force_ssl as a bare variable, this behaviour will go away and you might need to add |bool to the expression in the future. Also see CONDITIONAL_BARE_VARS configuration toggle.. This feature will be removed in version 2.12. Deprecation warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.

As well as these 6 others:

> 2019-05-02 15:27:54,458 p=9952 u=root |  TASK [1-prep : Install uuid-runtime package (debuntu)] *************************
...
[WARNING]: Could not find aptitude. Using apt-get instead [UBUNTU 16.04]

> 2019-05-02 15:27:49,566 p=9952 u=root |  TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************
...
2019-05-02 15:27:53,853 p=9952 u=root |   [WARNING]: The value True (type bool) in a string field was converted to u'True' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.
...
2019-05-02 15:27:53,854 p=9952 u=root |   [WARNING]: The value False (type bool) in a string field was converted to u'False' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.

> TASK [2-common : sysctl] ******************************************************* 2019-05-02 15:28:35,577 p=9952 u=root |   [WARNING]: The value 1 (type int) in a string field was converted to u'1' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.

> TASK [2-common : sysctl] ******************************************************* 2019-05-02 15:28:36,124 p=9952 u=root |   [WARNING]: The value 0 (type int) in a string field was converted to u'0' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.

> TASK [2-common : sysctl] ******************************************************* 2019-05-02 15:28:37,204 p=9952 u=root |   [WARNING]: The value 268435456 (type int) in a string field was converted to u'268435456' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.

holta, 2019-05-03:
Curiously Ansible 2.8.0rc2 was removed from https://launchpad.net/~ansible/+archive/ubuntu/ansible after only a few hours.

Consequence: IIAB installer http://d.iiab.io is back to auto-installing Ansible 2.7.10 for now...

holta, 2019-05-03:
> Running a BIG-sized install of IIAB yielded 222 of these warnings:

More about these 2 recommendations, to solve these 222 issues:
https://github.com/ansible/ansible/issues/53428
https://docs.ansible.com/ansible/devel/reference_appendices/config.html#conditional-bare-vars

holta, 2019-05-10:
https://releases.ansible.com/ansible/ansible-2.8.0rc3.tar.gz is now available, with 2.8.0 itself expected in 6 days (Thursday May 16th).

holta, 2019-05-16:
[Ansible 2.8.0](https://docs.ansible.com/ansible/devel/porting_guides/porting_guide_2.8.html) is released and live in OS repos for {Ubuntu, Raspbian etc} and appears to at least work with IIAB 7.0 pre-releases...

holta, 2019-05-24:
> 2019-05-02 15:27:49,566 p=9952 u=root |  TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************
> ...
> 2019-05-02 15:27:53,853 p=9952 u=root |   [WARNING]: The value True [...]

The above 2 WARNINGS (thrown by 0-init/tasks/main.yml) are caused by vars `gw_active` and `is_rpi` rather mysteriously.

Whereas other booleans like `internet_available` and `first_run` do not throw this warning (!)

Possibly related to an Ansible 2.8.0 bug report https://github.com/ansible/ansible/issues/56788 ??

Ref: PR #1694

holta, 2019-05-25:
This is largely resolved now, thanks to PRs #1694 and #1696 &mdash; but let's leave this open hoping that the [1 python3 + 7 casting issues above](https://github.com/iiab/iiab/issues/1632#issue-439761755) can also be solved in coming weeks prior to [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes)'s release?!

tim-moody, 2019-05-25:
I really wish we had pushed back on this flawed design in ansible rather than just going along. It makes everything more complicated, unnecessarily.

holta, 2019-05-25:
> I really wish we had pushed back on this flawed design in ansible rather than just going along. It makes everything more complicated, unnecessarily.

The new Ansible syntax is 100% lunatic.  Perhaps there's a chance someone will raise a big stink and convince IBM/Red Hat/Ansible to throw out their years-old broken design and redefine the spec before Ansible 2.12 makes `when: var` illegal later in 2020 ?

Barring  that, there I don't see realistic options going forward (aside from dumping Ansible for Docker possibly ;)

tim-moody, 2019-05-25:
https://github.com/ansible/ansible/blob/stable-2.8/test/units/config/test_manager.py shows an array with strings and types. I wonder if there was a way to define True/False as Boolean once to avoid the ugly cast.

holta, 2019-05-28:
> > 2019-05-02 15:27:49,566 p=9952 u=root |  TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************
> > ...
> > 2019-05-02 15:27:53,853 p=9952 u=root |   [WARNING]: The value True (type bool) in a string field was converted to u'True' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.
> > ...
> > 2019-05-02 15:27:53,854 p=9952 u=root |   [WARNING]: The value False (type bool) in a string field was converted to u'False' (type string). If this does not look like what you expect, quote the entire value to ensure it does not change.
> 
> The above 2 WARNINGS (thrown by 0-init/tasks/main.yml) are caused by vars `gw_active` and `is_rpi` rather mysteriously, when they are being written to `/etc/iiab/iiab.ini`
> 
> Whereas other booleans like `internet_available` and `first_run` do not throw this warning (!)

Relevant: https://github.com/ansible/ansible/issues/56788#issuecomment-496640899

holta, 2019-11-15:
Ansible 2.9.1 is now well-tested, and yes a few colorful warnings might still be addressed, but let's open a new ticket for those where nec.

holta, 2020-01-06:
Recap on Ansible bare variables and | bool i.e. casting to boolean — from https://github.com/iiab/iiab/commit/6b95a6c50f7d73bd2f6d659a6f930bbd87bddfed#r36049005

>> for my information, is the "| bool" no longer required?
>
>Documentation for CONDITIONAL_BARE_VARS outlines that when "| bool" is needed, these warnings have a new default in the upcoming Ansible 2.10, and warnings can no longer be suppressed in Ansible 2.12:
>
>https://docs.ansible.com/ansible/latest/reference_appendices/config.html
>
>However https://github.com/ansible/ansible/pull/57691/files indicates "| bool" is only needed with non-boolean variables:
>
>>just make sure to apply the |bool filter to non boolean variables (ex: string variables with content like 'yes', 'on', '1', 'true')

holta, 2020-01-06:
See Also:

"Bare variables in conditionals"
https://docs.ansible.com/ansible/latest/porting_guides/porting_guide_2.8.html#bare-variables-in-conditionals

"Updating your playbooks"
https://docs.ansible.com/ansible/latest/porting_guides/porting_guide_2.8.html#updating-your-playbooks

jvonau, 2020-01-06:
Try this change, replace the top highlighted line with the second line:

>  diff --git a/roles/0-init/tasks/main.yml b/roles/0-init/tasks/main.yml
index e72f3ba..d847159 100644
--- a/roles/0-init/tasks/main.yml
+++ b/roles/0-init/tasks/main.yml
@@ -226,11 +226,11 @@
     - option: product_id
       value: "{{ ansible_product_uuid }}"
     - option: gw_active
-      value: "{{ gw_active }}"
+      value: "{% if gw_active %}True{% else %}False{% endif %}"
     - option: internet_available
       value: "{{ internet_available }}"
     - option: is_rpi
-      value: "{{ is_rpi }}"
+      value: "{% if is_rpi %}True{% else %}False{% endif %}"
     - option: first_run
       value: "{{ first_run }}"
     - option: local_tz

jvonau, 2020-01-06:
Looks like any quoted facts are strings while a bare variable without quotes is a bool...

jvonau, 2020-01-06:
>

Those 2 variable are booleans by virtue of not being quoted, the warning says as much but we are try to record them in iiab.ini throwing the warning noted..

holta, 2020-01-21:
I think https://docs.ansible.com/ansible/latest/porting_guides/porting_guide_2.8.html#bare-variables-in-conditionals is the best explanation generally &mdash; but @jvonau reminds us that this longer story below (from 8-10 months) ago is very insightful to see how Ansible got in this mess (:

> More about these 2 recommendations, to solve these 222 issues:
> [ansible/ansible#53428](https://github.com/ansible/ansible/issues/53428)

jvonau, 2020-01-21:
> Looks like any quoted facts are strings while a bare variable without quotes is a bool...

From above url:
"Alternatively, you can re-define your variables as boolean values (without quotation marks) instead of strings:" shows the variable defined without quotes and without | bool tacked on the when clause. Without quotes is how default_vars/local_vars are, in the format above.

holta, 2020-01-22:
This annoying situation is at least largely under control now for IIAB 7.1's February release, so I'm moving this tkt to the [IIAB 7.2 Milestone](https://github.com/iiab/iiab/milestone/7) &mdash; in preparation for Ansible 2.10 which is likewise coming in Q2 2020.

PR #2576 (2020-10-17): Remove "| bool" 1.5yrs later, as Ansible 2.10 changed conditional_bare_variables default
Resolves #1632.

The ` | bool` is no longer necessary on the right side of `when:` conditions now that Ansible 2.10 makes this more official:

- https://docs.ansible.com/ansible/latest/porting_guides/porting_guide_2.8.html#bare-variables-in-conditionals
- https://github.com/ansible/ansible/issues/70682

So I ran `grep -r '\s*|\s\+bool' roles | xargs sed -i 's/\s*|\s\+bool//'` in /opt/iiab/iiab while careful not to modify these 3 files:

```
root@box:/opt/iiab/iiab# grep -rl '\s*|\s*bool'

.git/objects/pack [... cut]
closed 2020-10-17 by holta