iiab/iiab #1635: KA Lite language pack

tim-moody, 2019-05-05:
Why is registration and installation of en language pack no longer part of the kalite role?

tim-moody, 2019-05-05:
I someone tries to install kalite with runrole after an initial install they will not get installed.

holta, 2019-05-05:
> Why is registration and installation of en language pack no longer part of the kalite role?

As far as I know KA Lite registration & installation of its mandatory en language pack were never part of https://github.com/iiab/iiab/tree/master/roles/kalite

KA Lite's Ansible role/playbook (above) could certainly be improved as such.  Likewise @arky's Transmission (BitTorrent) role takes the next step of downloading KA Lite learning content.

Auto-provisioning of "ubiquitous" content for Kolibri is also requested by a growing many, that we should work with Learning Equality to address soon: https://github.com/iiab/iiab/issues/1545

holta, 2019-05-05:
On a related note, IIAB's 1-line installer (http://d.iiab.io) does basic provisioning of KA Lite as follows:

https://github.com/iiab/iiab-factory/blob/master/iiab#L22-L24
https://github.com/iiab/iiab-factory/blob/master/iiab#L215-L240

holta, 2019-05-10:
If this is a feature request to rework the KA Lite role, it should moved to the IIAB 7.1 milestone, so that IIAB 7.0 can converge in coming weeks.

(Or if it's just a question, it should probably be closed now?)

jvonau, 2021-08-29:
#2739 #2928

closed 2025-03-11 by tim-moody