iiab/iiab #1638: Install errors out on Lokole install

floydianslips, 2019-05-08:
### Specifications like the version of the IIAB, operating system version, or hardware details.
Raspberry Pi 3 B+
Raspbian (April 2019)
IIAB 7.0

[image]

floydianslips, 2019-05-08:
`$ sudo pip install opwen_email_client
Collecting opwen_email_client
  Using cached https://files.pythonhosted.org/packages/1b/00/0e4c46c1d419d7e59ef8ef5fe91999cecd16b4b46ed2e950d8c01df2bb70/opwen_email_client-0.1.41.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-REAkkZ/opwen-email-client/setup.py", line 4, in <module>
        with open('requirements.txt', encoding='utf-8') as fobj:
    TypeError: 'encoding' is an invalid keyword argument for this function
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-REAkkZ/opwen-email-client/`

holta, 2019-05-08:
Does the same error occur if you change [/opt/iiab/iiab/roles/lokole/defaults/main.yml#L8](https://github.com/iiab/iiab/blob/master/roles/lokole/defaults/main.yml#L8) from 0.1.41 to 0.1.38 as per IIAB 6.7, and then run `cd /opt/iiab/iab` follwed by `./runrole lokole` ?

floydianslips, 2019-05-08:
Changing the version to 0.1.38 allowed the ./runrole to complete without an issue.

floydianslips, 2019-05-08:
Installed 0.1.38 and then in order .39, .40 and .41 without issue.

holta, 2019-05-08:
> Installed 0.1.38 and then in order .39, .40 and .41 without issue.

Many thanks @floydianslips: this confirms the regression in Lokole occurs somewhere after 0.1.38 &mdash; regardless whether installing on Raspbian Lite or Raspbian Desktop.

floydianslips, 2019-05-08:
Initial install of 0.1.38 and 0.1.39 seem to install without issue. 0.1.40 and 0.1.41 have the same error as above.

holta, 2019-05-08:
> Initial install of 0.1.38 and 0.1.39 seem to install without issue. 0.1.40 and 0.1.41 have the same error as above.

@c-w and @aidan-fitz see above &mdash; the regression was not part of 0.1.39 &mdash; but definitely appears to affect Lokole 0.1.40+

We're hopeful you can take the investigation from here, so the most recent versions of Lokole might once again be installable on Raspberry Pi before June, when IIAB 7.0's final release is scheduled?

PS I just reverted IIAB's master branch from 0.1.41 to 0.1.39 so that [BIG-sized] IIAB installs once again now work!

holta, 2019-05-10:
I had repeated problems pip installing Lokole 0.1.39 on Raspbian Desktop on RPi 3 B+ earlier today.

Reverting to a fresh install of 0.1.38 seems to have worked.  I don't get it, as I thought the issue was isolated to fresh installs of 0.1.49+

Thanks all for explanations, experiences, and possible workarounds...

c-w, 2019-05-13:
I believe this is a bug in `setup.py` that was introduced in https://github.com/ascoderu/opwen-webapp/commit/d0159dde193504a22a96b6ba6d4c5fb87c03acb2 and fixed in https://github.com/ascoderu/opwen-webapp/commit/c9aebc03c159530153189925462bbd01c201ff97.

holta, 2019-05-16:
> I believe this is a bug in `setup.py` that was introduced in [ascoderu/opwen-webapp@d0159dd](https://github.com/ascoderu/opwen-webapp/commit/d0159dde193504a22a96b6ba6d4c5fb87c03acb2) and fixed in [ascoderu/opwen-webapp@c9aebc0](https://github.com/ascoderu/opwen-webapp/commit/c9aebc03c159530153189925462bbd01c201ff97). A fix was published in [0.1.42](https://pypi.org/project/opwen-email-client/0.1.42/). Apologies for the inconvenience and thanks for your patience for the fix!

Thanks @c-w !

Any reason that IIAB 7.0 (final release expected in a few weeks, in June) should not consider Lokole 0.4.0 from https://github.com/ascoderu/opwen-webapp/releases ?

(On Raspbian / ARM, Ubuntu 18.04, Debian 9 and ideally Debian 10 too :)

c-w, 2019-05-16:
@holta Latest version `0.4.0` should be fine. There were some breaking changes to the sync setup and cronjob setup, but none of those features are exercised in the IIAB mode.

holta, 2019-05-16:
Appears fixed by PR #1666 &mdash; thanks everybody for [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) pre-release testing as we approach release in coming weeks!

closed 2019-05-16 by holta