iiab/iiab #1644: microSD card pricing has crashed, so OSM maps packs could contain continent-wide satellite data target for IIAB 7.1

holta, 2019-05-09:
Building these map packs will be a huge job, guided by George Hunt's design & workflows, that he has very courageously begun documenting here:

http://github.com/georgejhunt/maps/blob/simple/generate-regions/readme.md

Caveat: @georgejhunt believes that satellite data (from [Sentinel](https://sentinel.esa.int/), up to layer 13) will approximately double the size of these 10 map packs (currently most "continents" are about 5-to-20GB) but we'll have to see the exact sizes when the rubber meets the road.  Here are lat/long outlines of the 10 osm-vector-maps packs expected to be released alongside IIAB 7.0 in not so many weeks, in June 2019:

- http://iiab.net/osm-vector -> http://iiab.net/osm-vector/maplist/
- http://iiab.me/map
- http://iiab.net/bboxes

Context for @m-anish, @mir123, @mapmeld and others: [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) should include satellite data up to layer 7 as this photographic data is only 80MB total (a  judicious choice for now, thanks to @georgejhunt).  i.e. generating larger map packs that include satellite data up to layer 13 is highly unrealistic prior to IIAB 7.0's June 2019 release, when our larger focus on QA is more important at this time.

After IIAB 7.0 is released however: we should begin to act on this rather sudden and newly transformed market, i.e. that microSD cards pricing has crashed since November 2018, effectively (more than doubling the size that each NGO/implementer now has to work with at any particular price point.

_Something we could not possible have known &mdash; as microSD pricing had remained quite stable over the 2 prior years since 2016, and will greatly affect the viability of ~80GB English Wikipedia ZIM and other assests too!_

Refs: #877 #1605 #1612 #1639 #1643

holta, 2019-05-16:
@georgejhunt has made good progress investigating whether such pre-packaged satellite+OSM map packs are realistic (or not at all!)

If it happens at all it would likely land as part of IIAB 7.1+ ?

Summary from our weekly IIAB community call (http://minutes.iiab.io) posted to: https://github.com/iiab/iiab/issues/1036#issuecomment-493197113

georgejhunt, 2019-05-20:
## Size of Tiles of whole world on disk
Record data for future reference:
OSM data to zoom 9 occupies 580MB
SAT images to zoom 9 takes 755MB

holta, 2019-05-20:
> Size of Tiles of whole world on disk
...
> OSM data to zoom 9 occupies 580MB
> SAT images to zoom 9 takes 755MB

Thanks @georgejhunt!

FYI Kids On Computers appears interested in Uganda satellite imagery as part of Internet-in-a-Box in the coming month (by around mid-June 2019) just in case that turns out to be possible?

On the bright side, Uganda is only ~400 km wide and ~500km tall.  However that's a very substantial amount of imagery to pull together for a "prototype" that would not be broadly used around Uganda (just yet!)

Still: if something can come togethe in the coming month to help shape any kind of "Mass Customization" satellite photo solution for IIAB 7.1 &mdash; teasing out the implementer UX & download sanity issues that *everyone* around the planet will later benefit from &mdash; then this could assist everyone not just Ugandans?

FYI I'm told Kids On Computers' 3 schools are in 3 different places in Uganda...I'll talk with @avni and @randytate to try to find out a bit more!

avni, 2019-05-20:
If I am understanding the notes correctly, the 3 schools are: St. Paul’s School, Kampala (outskirts), and Nkugulutale. @randytate will have more detailed/up-to-date info.

holta, 2019-05-21:
> If I am understanding the notes correctly, the 3 schools are: St. Paul’s School,

Appears to be ~50 km E of Kampala: https://mwebaza.org/our-partner-schools/st-paul-infant-primary-school-and-coyote-ridge-elementary-school/more-about-st-paul-school/

> Kampala (outskirts)

Is a school name and lat/long possible?

> Nkugulutale

Appears to be ~10 km SW of Kampala

PS can @randytate clarify if these schools want their names & locations published?

randytate, 2019-05-22:
@holta You've got about as much info on school locations as I do. Here are the 3 schools, and the info lifted from https://mwebaza.org:

* Mwebaza School
  * [Kyengera, Uganda](https://www.google.com/maps/place/Kyengera,+Uganda) – approximately 10 miles southeast of the capital city of Kampala and 27 miles north of Lake Victoria; tropical terrain; suburban
* Mwebaza Annex (a.k.a. "MWAN")
  * [Nkugulutale, Uganda](https://www.google.com/maps/place/Nkugulutale,+Uganda) - approximately 5 miles from Mwebaza School
* St. Paul
  * [Nkokonjeru, Uganda](https://www.google.com/maps/place/Nkokonjeru,+Uganda) – approximately 30 miles east of the capital city of Kampala; tropical terrain; remote

holta, 2019-05-25:
@georgejhunt I did not realize that a Nigerian group is also building their RPi Internet-in-a-Box 7.0 pre-release boxes this weekend and in coming days.

As such the "continents" most highly demanded (at this time) are definitely:

1. Africa
2. Central America
3. South Asia

(Which is not to say that other "continents" do not matter eventually, but I would focus on the UX for the 3 above "continents" first and foremost for now &mdash; if there's any choice in the matter in coming days/weeks &dash; whether or not satellite photos for Zoom Levels 0-9 are yet included ;)

holta, 2019-10-21:
Delivered for IIAB 7.0 &mdash; and under improvement for IIAB 7.1 e.g. here:

#1958 "3 new OSM map packs covering {Spanish, French, Arabic}-speaking regions of the world"

closed 2019-10-21 by holta