iiab/iiab #1646: Migrate to kolibri-server for RPi performance

holta, 2019-05-10:
Appears unlikely for IIAB 7.0 in coming weeks?

Refs: #977 #1545

holta, 2019-05-10:
@jredrejo is there a roadmap for https://github.com/learningequality/kolibri-server/releases where Internet-in-a-Box developers can learn more about its target specs & possible dates over the coming year?

Thanks if poss!

jredrejo, 2019-05-10:
Hi @holta 
The `kolibri-server` package is considered finished and stable. We haven't announced it yet because there's an issue due to the fact `kolibri` didn't work properly in a multi-process scenario. It's a kolibri problem that has been solved in the develop branch of kolibri. After the new kolibri becomes stable we'll announce it.

In any case, the problem is quite trivial: sometimes when importing channels you have to refresh the page and click again on import. Other than that, everything works fine in all our tests. If your installation has all the channels imported, you can work normally with kolibri-server. Different installations have been using it successfully in the last months.

holta, 2019-05-10:
@arky please see Learning Equality's update on `kolibri-server` just above!

holta, 2019-05-11:
> The `kolibri-server` package is considered finished and stable. We haven't announced it yet because there's an issue due to the fact `kolibri` didn't work properly in a multi-process scenario. It's [...]

Thanks Much @jredrejo.  A few tiny questions if you can:

1) When you say "the new kolibri" are you talking about 0.12.3?  (Or some other version such as 0.13?)

2) Internet-in-a-Box communities need several of the Kolibri fixes that have been merged into 0.12.3 (for example #1626 and https://github.com/learningequality/kolibri/pull/5480) so the high-level question (affecting us in many ways ;) is &mdash; how is kolibri-server likely to track (or perhaps ship as a part of) Kolibri's regular releases, e.g. 0.13.x and 0.14.x etc?

3) Should `kolibri-server` work cleanly with Apache now or in future?  (Or is it really intended for NGINX?)

4) What's the preferred way to install kolibri-server?  (Where is the best documentation on kolibri-server if possible?)

jredrejo, 2019-05-12:
> Thanks Much @jredrejo. A few tiny questions if you can:
>
> 1. When you say "the new kolibri" are you talking about 0.12.3?  (Or some other version such as 0.13?)
It's not decided yet. The code is in develop but it's not merged into 0.12.3. I doubt it will be.

> 2. Internet-in-a-Box communities need several of the Kolibri fixes that have been merged into 0.12.3 (for example #1626 and [learningequality/kolibri#5480](https://github.com/learningequality/kolibri/pull/5480)) so the high-level question (affecting us in many ways ;) is — how is kolibri-server likely to track (or perhaps ship as a part of) Kolibri's regular releases, e.g. 0.13.x and 0.14.x etc?

`kolibri-server` is just a a set of configurations plus dependencies to use kolibri with nginx & uwsgi. It does not include any kolibri code. All the new features and fixes will be done on `kolibri`. 

> 3. Should `kolibri-server` work cleanly with Apache now or in future?  (Or is it really intended for NGINX?)

The setup is thought to be used with nginx. There are no plans to do an apache version and I don't think it's a good idea to maintain multiple ways of getting the same purpose.

> 4. What's the preferred way to install kolibri-server?  (Where is the best documentation on kolibri-server if possible?)

Currently it's available as a ppa package in the `kolibri-proposed` repo. The plan is moving it to the `kolibri`repo. Until that movement is done, the recommended way to install it is using the instructions available in https://kolibri.readthedocs.io/en/latest/install.html#debian-ubuntu , replacing `kolibri`by `kolibri-proposed`in the repo name and doing `sudo apt-get install kolibri-server` to install it.

jvonau, 2019-09-07:
Given that nginx is a dependency, moving to kolibri-server is not in the card right now.

benjaoming, 2020-03-26:
@jvonau - can you elaborate this part? Just wanna understand how the Nginx dependency blocks adaptation  of kolibri-server for you.

holta, 2020-03-26:
@benjaoming IIAB now uses NGINX (see https://github.com/iiab/iiab/blob/master/roles/nginx/README.md) but we would need help getting an NGINX config file for kolibri-server, that doesn't assume it's the only thing running on the machine (!)

If you can propose such an /etc/nginx/conf.d/kolibri-nginx.conf for starters?  As you can see we are currently using: https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri-nginx.conf.j2

jredrejo, 2020-03-26:
hello @holta 
`kolibri-server` does not assume it's the only thing running in the machine. Just the opposite, it uses `/etc/nginx/conf.d` and you can drop there any other nging config you want.

the only thing it assumes is that, **after** you decide which port you're using for kolibri (among three different options) that port will be used only by kolibri. But you can server other apps using nginx in the same server with their own configs in different ports without any conflict with kolibri.

In case you want to show to the world everything under the same port, you can also set a proxy configuration in nginx to serve all the applications under the same port without conflicting with the current kolibri-server nginx configuration

benjaoming, 2020-03-26:
@jredrejo it looks like (from the link shared by @holta) that IIAB already has a nice little `proxy_pass` configuration for Kolibri.

Opened an upstream issue to make kolibri-server easier to accommodate for.