iiab/iiab #1647: Question: does map_functions.js belong in both iiab and iiab-admin-console?

tim-moody, 2019-05-10:


tim-moody, 2019-05-10:
also, if I change code in the iiab-admin-console version do I risk breaking something somewhere else?

georgejhunt, 2019-05-10:
yes, I don't know how to do the splash page without  javascript, and I
don't want two start from scratch creating a new one when the one I have is
good enough.
I want there to be some function in IIAB without requiring the console.
Then there's the forwar dependency issue. If I rely on a javascript file in
the console, which is not yet installed, it's more complicated in ansible

tim-moody, 2019-05-10:
This seems similar to the dilemma I faced between js-menu.js and js-menu-functions.js. In the end I found there was much less in common than I originally thought and I let them diverge.

tim-moody, 2019-05-10:
In terms of dependencies, I agree that all roles in iiab should function without iiab-admin-console having been installed. But then they will function without js-menu, which means they don't need any of the assets that js-menu needs, like menu defs or zim_version_idx.json, etc. 

As far as I know the only thing required is library.xml. So iiab_update_menus, etc. and even the catalogs are only really needed by iiab-admin-console.

tim-moody, 2019-05-10:
So far we have

https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/templates/iiab-update-map
https://github.com/iiab/iiab-admin-console/blob/master/roles/console/templates/iiab-update-map
https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/templates/iiab-update-map.py

I'd like to settle on the 2nd of these as the location; not yet sure about which has the right the functionality.

tim-moody, 2019-05-10:
found another one
https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/iiab_update_menus.py

tim-moody, 2019-05-10:
Actually I think the 4th location (cmdsrv/scripts) is best. Any idea which is the right code between cmdsrv/templates/scripts/iiab_update_menus.py and js-menu/templates/iiab-update-map.py? Both are being installed and the js-menu version wins but has syntax errors.

tim-moody, 2019-05-11:
looks like the one installed by iiab works and the others are not used.

tim-moody, 2019-05-12:
done

closed 2019-05-12 by tim-moody