iiab/iiab #1648: Kolibri Not starting After Reboot

kananigit, 2019-05-11:
I installed Kolibri 0.12.2  by running the latest IIAB 7.0 on a Raspberry Pi Model 3B. After the first reboot upon successful IIAB intall, i used Kolibri and even downloaded a couple channels to it. I later shut down the server and when i started it the next day kolibri did not stat. here are some of the screenshots.

[image]

**/usr/local/kolibri/bin/kolibri --start**
[image]

**Contents of /library/kolibri/server.log**
[image]

holta, 2019-05-11:
@arky @jredrejo @KW4NP are you familiar with the kolibri service failing to start as above?

The only real difference (I know of) between @kananigit's Kolibri failure and my own 7.0 ("systemctl status kolibri" works fine on my pre-release of IIAB on RPi 3 B+) is that he downloaded a few Kolibri channels &mdash; whereas I did not.

KW4NP, 2019-05-11:
I did run into this issue when I had a problem with a malformed content database.  Deleting the database seemed to be successful, but Kolibri would not start with systemd.  I started it manually and let it run through the migrations and stopped it again, but on the next boot it did the same thing.  The solution I found was to start it manually, make sure everything was set up (facility, etc.), and reboot the machine.  I don't remember it just not starting with no prior problem, however.

KW4NP, 2019-05-11:
This one seems to be running an update routine for a new version - does Kolibri automatically update?

holta, 2019-05-11:
> This one seems to be running an update routine for a new version - does Kolibri automatically update?

Not that I know of!

(What evidence of a Kolibri "update routine for a new version" are you seeing?)

KW4NP, 2019-05-11:
In the second picture
```
INFO   Running update routines for new version
```

holta, 2019-05-11:
What kind of (presumably offline?) update Kolibri is running on start is unclear.  Googling "Running update routines for new version" does not give an immediate vanswer as to what this means.

Hopefully @jredrejo can shed more light...
...and on the larger question of how to diagnose `kolibri.service` when it won't start?

kananigit, 2019-05-12:
Thanks for the quick reply Tim. Id like to know what commands you ran to
start kolibri manually . I ran systemctl start kolibri but that did not
help as it timed out. Check the screenshot where I did systemctl status
kolibri

KW4NP, 2019-05-12:
I found the instructions here https://github.com/iiab/iiab/blob/master/roles/kolibri/README.rst   
Note that this should be run as root - not using sudo.

jredrejo, 2019-05-12:
> This one seems to be running an update routine for a new version - does Kolibri automatically update?

No, it does not. The routine you mention is to update the database models after a kolibri update.

holta, 2019-05-13:
@kananigit can you please try @benjaoming's suggestion at https://github.com/learningequality/kolibri-installer-debian/issues/66#issuecomment-491629055 ?

Specifically, can you install a smaller-than-MIN-sized IIAB 7.0 pre-release — but first set most all apps' variables to _install: False and _enabled: False in /etc/iiab/local_vars.yml — prior to launching `sudo iiab` ?

(Or if you prefer `./runrole kolibri` all alone?)

Either way please let us know whether Kolibri's service continues to fail or not, Thanks !

holta, 2019-12-28:
@kananigit have you experienced this in the last 7 months?

Thanks for clarifying please &mdash; whether your experiences are in Kenya, UK, or USA or wherever!