iiab/iiab #1655: How to delete Kolibri Facility?

KW4NP, 2019-05-11:
### Expected behavior and actual behavior.
No way to delete unwanted Kolibri facilities that are apparently created by running iiab-install if the default facility doesn't already exist.
@arky , do you have any insight on this?

### Steps to reproduce the problem.
Make a custom Kolibri facility, run ./iiab-install and then try to delete the new default facility that is apparently created if the default facility doesn't exist.

### Specifications like the version of the IIAB, operating system version, or hardware details.
IIAB 7.0, Rpi 3 B+

holta, 2019-05-11:
@jredrejo @benjaoming do you know the answer, or how we should proceed?

jredrejo, 2019-05-12:
There's no command to delete a facility. The closer option is a command called deprovision, with this purpose:
> Delete all facility user data from the local database, and put it back to a clean state (but leaving content as-is).

KW4NP, 2019-05-14:
Thanks for your help.  For anyone else using Kolibri on IIAB, you must run ``` export KOLIBRI_HOME=/library/kolibri ``` or it will not actually deprovision the IIAB Kolibri instance but will try to deprovision the default Kolibri home folder.  It would be nice to be able to take care of it from the admin console - a button to delete the facility for facility admins would be more user-friendly.

tim-moody, 2019-05-14:
iiab installs ka-lite in a venv that has wrappers for commands to do the export KA-LITE_HOME part. Not sure if that was done for kolibri. Since both ka-lite and kolibri have their own admin interface we have not added anything to the iiab admin console.

KW4NP, 2019-05-14:
I'm pretty sure that the venv work has not been done for Kolibri so far.  Running any commands without exporting the variable seems to act on a default .kolibri directory in home.

benjaoming, 2019-05-14:
It's not necessary to run Kolibri in a venv, we never made a distribution of Kolibri with dependencies, so it will run on any recent Debian/Ubuntu system library and not alter it.

holta, 2019-05-14:
Huge thanks @KW4NP @benjaoming @jredrejo I'll start adding the most requested Kolibri command-line survival tricks like this one to http://FAQ.IIAB.IO when I get back in front of a computer in coming days.

(Analogous to http://FAQ.IIAB.IO #32 which is our longstanding "tips & tricks survival guide" for KA Lite...exactly these kinds of tips & tricks are now essential for Kolibri, as implementers begin to consider the transition!)

tim-moody, 2019-05-14:
probably we should add wrappers for kolibri that set the env even without venv