iiab/iiab #1656: --non-interactive flag for "sudo iiab" installer

holta, 2019-05-12:
@floydianslips requests that IIAB installer https://github.com/iiab/iiab-factory/blob/master/iiab be enhanced with a --non-interactive command-line flag.

He's well aware that "sudo iiab" has interactive UX aspects that are critical to keep newbie installers on track &mdash; so they don't get completely stuck &mdash; and/or worse, badly shoot themselves in the foot.

These bits of interactive guidance have proved urgently necessary, greating streamlining the IIAB installation over the past year, in an era when few people read instructions.

All that being true, @floydianslips is also very hopeful that the same script could be modified for IIAB "test automation" regimes and similar, where {apt updates, reboots, loud warnings, etc} are all _intentionally_ avoided.

This should be doable early in the IIAB 7.1 release cycle ideally?

jvonau, 2019-08-25:
Think I have something in the works without resorting to --non-interactive, what if the "read" waiting for $ans had a timeout value in iiab? Should that not just wait for input and proceed anyway after waiting? Given if you prepopulate local_vars.yml most of the questions are skipped and the same if the system was recently updated, what else can be streamlined?

jvonau, 2019-08-25:
Think this --non-interactive is addressed in https://github.com/iiab/iiab-factory/pull/73, I was testing other code and didn't want to wait for a reboot because an update that just showed up... :)

jvonau, 2020-09-19:
iiab/iiab-factory#143

holta, 2020-11-01:
I had a short conversation with @floydianslips in person last week on why I don't agree with this proposal.

In short:

- Ppl who use a proper password won't be nagged and never have been
- Ppl who continuously use a published password should pay a price (being nudged, to stop putting themselves and their community at risk)
- Ppl who truly want bad security can hit [Enter] today if they want the published password — a single keystroke is not worth crying over.
- Likewise implenterd can and always been able to pre-position /etc/Iiab/local_vars.yml — i.e. people who are organized don't have to face inappropriate reminders and never have
- CTRL-C works to exit the 1-line-installer, and that's a really important skill for everyone to learn (in life) if they want to pre-position `/etc/Iiab/local_vars.yml` or fix their password etc

holta, 2021-08-22:
@floydianslips it took 2 years, but two new options are now available:

- For Clueful Testing People:
  `curl d.iiab.io/fast.txt | sudo bash`
  If interrupted, continue with... `sudo iiab -f` or `--fast`
- For DareDevs:
  `curl d.iiab.io/risky.txt | sudo bash`
  If interrupted, continue with... `sudo iiab -F` or `--risky`

Details:

- PR #iiab/iiab-factory#182

closed 2021-08-22 by holta