iiab/iiab #1658: Validate Moodle 3.5.7 LTS on release "2019-07-08" ?

holta, 2019-05-13:
Ref: #1529

holta, 2019-07-01:
Draft release notes for Moodle 3.5.7 should appear here in coming days, prior to its release:

https://docs.moodle.org/dev/Moodle_3.5.7_release_notes

holta, 2019-07-08:
> Draft release notes for Moodle 3.5.7 should appear here in coming days, prior to its release:
> 
> https://docs.moodle.org/dev/Moodle_3.5.7_release_notes

Release/Confirmed!

Please all who can test Moodle 3.5.7 going forward.

georgejhunt, 2019-07-10:
When I bring up the install page, (rpi,buster,big), I get at the top of the validation page:
PHP version 7.3 is not supported.
Reading moodle docs at: https://docs.moodle.org/dev/Moodle_3.5_release_notes

<quite> 
PHP version: minimum PHP 7.0.0 Note: minimum PHP version has increased since Moodle 3.3. PHP 7.1.x and 7.2.x are supported too. PHP 7.x could have some engine limitations.
</quote>
and 
https://docs.moodle.org/dev/Moodle_and_PHP7
"PHP 7.3 can be used with Moodle 3.6.4, Moodle 3.7 and later releases."

holta, 2019-07-10:
Thank @georgejhunt for catching this.

Can we consider dumping Moodle 3.5.x LTS in favor of 3.7.x ?

georgejhunt, 2019-07-10:
Or perhaps just dropping support for moodle entirely.

Do you know of any moodle users?

holta, 2019-07-10:
> Or perhaps just dropping support for moodle entirely. Do you know of any moodle users?

Many implementers try it.

Few follow through to use it long-term.

It seems that its main purpose is to rapidly trial the Moodle experience.

Interestingly this is the main purpose of BIG-sized IIAB too: to provide a live working environment to help educate implementers learn the ropes, establishing what kinds of LMS-like features their communities can rapidly learn & take advantage of...

holta, 2019-07-10:
I've upgraded [IIAB 7.0 DRAFT Release Notes](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) to Moodle 3.7 1 which solves the problem on Raspbian Buster on RPi 4, and hopefully other OS's too.

closed 2019-07-10 by holta

holta, 2019-07-11:
> > Do you know of any moodle users?
> 
> Many implementers try it.
> 
> Few follow through to use it long-term.

(One of) the reasons, is that we've never made it easy to adopt Moodle.

Specifically everyone that's tried to use Moodle with IIAB has struggled with its URL's.

As I believe IIAB's Moodle's URL's (like http://box/moodle) all force a redirect to http://box.lan/moodle) which is not what everyone wants...

Not that there's a magical, easy fix.  But we have a chicken+egg situation here:

#10 Moodle erroneously redirects to schoolserver.lan/moodle (bad enough on LAN, worse on WAN)
#551 How to mitigate campuswide failures for Nextcloud, WordPress, Moodle, Elgg etc?
#941 http://10.0.0.x/moodle must be customized on installation

Whereas with WordPress we've cleanly documented several solutions, that people use regularly:

http://FAQ.IIAB.IO -> ITEM 33 `"WordPress & Moodle Administration: What tips & tricks exist?"` -> `"How do I set the WordPress URL (like http://box/wordpress) so it works for everyone?"`