iiab/iiab #1664: IIAB breaks SMB

mikerjensen, 2019-05-14:
I have SMB (windows file sharing) working fine on the vanilla Raspbian distribution but after installing IIAB, the RPI stops serving SMB.  I tried twice. Same problem. 

The RPI is running IIAB pre 7.0 and is connected to wireless router via ethernet port. 

Only other clues are that a) after IIAB is installed, i can't ping the box anymore from the rest of the LAN. b) If i use the IIAB configured wifi hotspot to connect directly to the RPI then SMB file sharing works fine.

tim-moody, 2019-05-14:
b) sounds like samba_install: True and samba_enabled: True as set properly.
can you ping via the ip address? if iiab is connected to the router, then the router is doing name resolution and may or may not find box.lan
services_externally_visible: True must also be set for the smb (and other) ports to be open
try sharing by ip and not by host name.

mikerjensen, 2019-05-14:
On 14/05/19 15:35, Tim Moody wrote:
> b) sounds like samba_install: True and samba_enabled: True as set properly.

I assume so... i manually installed samba and edited smb.conf when i installed it on Raspbian, so am not sure where the two above settings are configured

> can you ping via the ip address? if iiab is connected to the router, then the router is doing name > > > > resolution and may or may not find box.lan

I can't ping via the IP address....i.e 'ping 192.168.1.218' results in no response, while 'ssh -l pi 192.168.1.218' works fine

> services_externally_visible: True must also be set for the smb (and other) ports to be open
> try sharing by ip and not by host name.
Not sure where this is set..

tim-moody, 2019-05-14:
OK. I assumed you had installed samba via iiab. So we need to backup.
Did you start with an image or install iiab from scratch?
If the latter, did you edit /etc/iiab/local_vars.yml or did you chose one of the min, medium, big local_vars configurations?

mikerjensen, 2019-05-14:
Sorry, I didn't make it clear - yes i just installed samba directly via apt...didn't occur to me that IIAB had an option. I had installed IIAB by grabbing the /usr/sbin/iiab script and running it with the min config - curl -o /usr/sbin/iiab https://raw.githubusercontent.com/iiab/iiab-factory/master/iiab

I just tried right now to edit local_vars.yml to enable samba. and rerunning /usr/sbin/iiab, rebooting, but this didn't seem to make any difference. Still can't ping the RPI from the network using the IP number, although ssh works fine...And as before, samba works fine when accessing the RPI via its built in wifi.

tim-moody, 2019-05-14:
and  services_externally_visible: True in local vars?

mikerjensen, 2019-05-14:
Yep.

tim-moody, 2019-05-14:
your samba is still there or just iiab's?
can you access the web server over the upstream network using ip address?
what if you
cd /opt/iiab/iiab
./runrole samba

tim-moody, 2019-05-14:
seems I was wrong and we do not open the samba port upstream even if services_externally_visible: True

probably we should

mikerjensen, 2019-05-14:
Well after running cd /opt/iiab/iiab ; ./runrole samba it seems that it is iiab's samba that is now running  - /etc/samba/smb.conf was replaced with IIAB's, unfortunately, even after rebooting, still no smb access via eth0. But i can access the web server over eth0 - both to the IP address and to http://box.lan/home

tim-moody, 2019-05-14:
here is where it should happen

https://github.com/iiab/iiab/blob/50e5e1067f82d3bbb834ea091e6b2aef251cf1d8/roles/network/templates/gateway/iiab-gen-iptables

mikerjensen, 2019-05-14:
Yes, I think you've found the problem - perhaps as you say, this needs to be changed to allow smb to be externally visible.
 # Don't forward from the outside to the inside.
$IPTABLES -A FORWARD -i $wan -o $lan -j DROP
$IPTABLES -A INPUT -i $wan -j DROP

tim-moody, 2019-05-15:
I created a PR #1665 if you want to test it.

holta, 2019-05-24:
@mikerjensen sorry for the delay: PRs #1677 #1691 #1692 are now merged so you can set `ports_externally_visible: 4` to open up WAN-side access to Samba/SMB during your next IIAB install.

To make this happen, please [comment out Line 50 and uncomment Line 51](https://github.com/iiab/iiab/blob/master/vars/local_vars_medium.yml#L50-L51) of [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) prior to your next IIAB install.

If there's any problem feel free to re-open this ticket, Thanks!

(And now onto completing https://github.com/iiab/iiab/pull/1617 at long last~)

closed 2019-05-24 by holta